libimobiledevice/usbmuxd #34: iPhone is not recognized after upgrade to Fedora 20

khomisha, 2015-01-21:
After upgrading to Fedora 20 the iphone was no longer found when plugin. In log:

янв 21 15:24:49 mk-hp usbmuxd[5719]: [15:24:49.938][3] usbmuxd v1.0.9 starting up
янв 21 15:24:49 mk-hp usbmuxd[5719]: [15:24:49.939][3] Successfully dropped privileges to 'usbmuxd'
янв 21 15:24:49 mk-hp usbmuxd[5719]: [15:24:49.942][3] Could not get old configuration descriptor for device 2-4: -5
янв 21 15:24:49 mk-hp usbmuxd[5719]: [15:24:49.979][3] Connecting to new device on location 0x20004 as ID 1
янв 21 15:24:49 mk-hp usbmuxd[5719]: [15:24:49.979][3] Initialization complete
янв 21 15:24:49 mk-hp usbmuxd[5719]: [15:24:49.980][3] Connected to v1.0 device 1 on location 0x20004 with serial number 18c6e131038e60b16a515187ec901304eca6edd4
янв 21 15:24:50 mk-hp usbmuxd[5719]: [15:24:50.044][1] preflight_worker_handle_device_add: Could not get ProductVersion string from device 18c6e131038e60b16a515187ec901304eca6e

khomisha, 2015-01-21:
I tried to plugin different apple devices and this is what I found:
Device	         iOS	        Comment
iPhone 1	        1.1.4	Not found in system
iPhone 3s 	5.1.1	Found after plugin, working fine
iPad  2	        8.1.2	During plugin tablet asks «Should i thrust this computer?». But afterwards it's not found in system
iPad	                6.0.1	Found after plugin, working fine
IPhone 4	        7.1.2	During plugin tablet asks «Should i thrust this computer?». Found after plugin, working fine

There is log
=============== iPhone 3s iOS 5.1.1
янв 21 16:51:18 mk-hp usbmuxd[3683]: [16:51:18.149][3] usbmuxd v1.0.9 starting up
янв 21 16:51:18 mk-hp usbmuxd[3683]: [16:51:18.150][3] Successfully dropped privileges to 'usbmuxd'
янв 21 16:51:18 mk-hp usbmuxd[3683]: [16:51:18.153][3] Could not get old configuration descriptor for device 2-4: -5
янв 21 16:51:18 mk-hp usbmuxd[3683]: [16:51:18.156][3] Connecting to new device on location 0x20004 as ID 1
янв 21 16:51:18 mk-hp usbmuxd[3683]: [16:51:18.156][3] Initialization complete
янв 21 16:51:18 mk-hp usbmuxd[3683]: [16:51:18.156][3] Connected to v1.0 device 1 on location 0x20004 with serial number f078ecbc8fe3c8fead2913d7eda6a956137e95e9
янв 21 16:52:06 mk-hp usbmuxd[3683]: [16:52:06.760][3] Removed device 1 on location 0x20004
янв 21 16:52:06 mk-hp usbmuxd[3683]: [16:52:06.871][3] usbmuxd shutting down
янв 21 16:52:06 mk-hp usbmuxd[3683]: [16:52:06.972][3] Shutdown complete
=============== iPad2 iOS 8.1.2
янв 21 16:53:07 mk-hp usbmuxd[3770]: [16:53:07.559][3] usbmuxd v1.0.9 starting up
янв 21 16:53:07 mk-hp usbmuxd[3770]: [16:53:07.560][3] Successfully dropped privileges to 'usbmuxd'
янв 21 16:53:07 mk-hp usbmuxd[3770]: [16:53:07.562][3] Could not get old configuration descriptor for device 2-5: -5
янв 21 16:53:07 mk-hp usbmuxd[3770]: [16:53:07.565][3] Connecting to new device on location 0x20005 as ID 1
янв 21 16:53:07 mk-hp usbmuxd[3770]: [16:53:07.565][3] Initialization complete
янв 21 16:53:07 mk-hp usbmuxd[3770]: [16:53:07.565][3] Connected to v1.0 device 1 on location 0x20005 with serial number b63691ed564fc58d098c81d5641c3c2d2b2eabfb
янв 21 16:53:08 mk-hp usbmuxd[3770]: [16:53:08.076][1] config_get_device_record: failed to read '/var/lib/lockdown/b63691ed564fc58d098c81d5641c3c2d2b2eabfb.plist': No such file
янв 21 16:53:08 mk-hp usbmuxd[3770]: [16:53:08.101][1] config_get_device_record: failed to read '/var/lib/lockdown/b63691ed564fc58d098c81d5641c3c2d2b2eabfb.plist': No such file
янв 21 16:53:08 mk-hp usbmuxd[3770]: [16:53:08.621][1] config_get_device_record: failed to read '/var/lib/lockdown/b63691ed564fc58d098c81d5641c3c2d2b2eabfb.plist': No such file
янв 21 16:53:57 mk-hp usbmuxd[3770]: [16:53:57.218][3] Removed device 1 on location 0x20005
янв 21 16:53:57 mk-hp usbmuxd[3770]: [16:53:57.454][3] usbmuxd shutting down
янв 21 16:53:57 mk-hp usbmuxd[3770]: [16:53:57.555][3] Shutdown complete
=============== iPad iOS 6.0.1 
янв 21 16:55:04 mk-hp usbmuxd[3804]: [16:55:04.044][3] usbmuxd v1.0.9 starting up
янв 21 16:55:04 mk-hp usbmuxd[3804]: [16:55:04.044][3] Successfully dropped privileges to 'usbmuxd'
янв 21 16:55:04 mk-hp usbmuxd[3804]: [16:55:04.046][3] Could not get old configuration descriptor for device 2-6: -5
янв 21 16:55:04 mk-hp usbmuxd[3804]: [16:55:04.048][3] Connecting to new device on location 0x20006 as ID 1
янв 21 16:55:04 mk-hp usbmuxd[3804]: [16:55:04.048][3] Initialization complete
янв 21 16:55:04 mk-hp usbmuxd[3804]: [16:55:04.049][3] Connected to v1.0 device 1 on location 0x20006 with serial number 567018368d87afcae130d33f085f89bdeca33409
янв 21 16:55:25 mk-hp usbmuxd[3804]: [16:55:25.585][3] Removed device 1 on location 0x20006
янв 21 16:55:25 mk-hp usbmuxd[3804]: [16:55:25.771][3] usbmuxd shutting down
янв 21 16:55:25 mk-hp usbmuxd[3804]: [16:55:25.872][3] Shutdown complete
============== iPhone 4 iOS 7.1.2
янв 21 17:31:51 mk-hp usbmuxd[4576]: [17:31:51.825][3] usbmuxd v1.0.9 starting up
янв 21 17:31:51 mk-hp usbmuxd[4576]: [17:31:51.826][3] Successfully dropped privileges to 'usbmuxd'
янв 21 17:31:51 mk-hp usbmuxd[4576]: [17:31:51.828][3] Could not get old configuration descriptor for device 2-7: -5
янв 21 17:31:53 mk-hp usbmuxd[4576]: [17:31:53.555][3] Connecting to new device on location 0x20007 as ID 1
янв 21 17:31:53 mk-hp usbmuxd[4576]: [17:31:53.555][3] Initialization complete
янв 21 17:31:53 mk-hp usbmuxd[4576]: [17:31:53.555][3] Connected to v1.0 device 1 on location 0x20007 with serial number 60d1c9bbd89d61ac04b3a97fb20bbe5b763d2309
янв 21 17:31:54 mk-hp usbmuxd[4576]: [17:31:54.713][1] config_get_device_record: failed to read '/var/lib/lockdown/60d1c9bbd89d61ac04b3a97fb20bbe5b763d2309.plist': No such file
янв 21 17:31:54 mk-hp usbmuxd[4576]: [17:31:54.817][1] config_get_device_record: failed to read '/var/lib/lockdown/60d1c9bbd89d61ac04b3a97fb20bbe5b763d2309.plist': No such file
янв 21 17:31:55 mk-hp usbmuxd[4576]: [17:31:55.069][1] config_get_device_record: failed to read '/var/lib/lockdown/60d1c9bbd89d61ac04b3a97fb20bbe5b763d2309.plist': No such file
янв 21 17:32:19 mk-hp usbmuxd[4576]: [17:32:19.172][3] Removed device 1 on location 0x20007
янв 21 17:32:19 mk-hp usbmuxd[4576]: [17:32:19.430][3] usbmuxd shutting down
янв 21 17:32:19 mk-hp usbmuxd[4576]: [17:32:19.531][3] Shutdown complete
янв 21 17:32:19 mk-hp usbmuxd[4618]: [17:32:19.735][3] usbmuxd v1.0.9 starting up
янв 21 17:32:19 mk-hp usbmuxd[4618]: [17:32:19.736][3] Successfully dropped privileges to 'usbmuxd'
[... cut ...]
янв 21 17:32:19 mk-hp usbmuxd[4618]: [17:32:19.798][3] Connecting to new device on location 0x20008 as ID 1
янв 21 17:32:19 mk-hp usbmuxd[4618]: [17:32:19.799][3] Initialization complete
янв 21 17:32:19 mk-hp usbmuxd[4618]: [17:32:19.799][3] Connected to v1.0 device 1 on location 0x20008 with serial number 60d1c9bbd89d61ac04b3a97fb20bbe5b763d2309
янв 21 17:32:49 mk-hp usbmuxd[4618]: [17:32:49.858][3] Removed device 1 on location 0x20008
янв 21 17:32:49 mk-hp usbmuxd[4618]: [17:32:49.895][3] usbmuxd shutting down
янв 21 17:32:49 mk-hp usbmuxd[4618]: [17:32:49.995][3] Shutdown complete

FunkyM, 2015-09-25:
Which iPhone do you mean?
Is it ```iPhone 1 1.1.4 Not found in system```?
Are you able to try the latest usbmuxd version?

khomisha, 2015-09-27:
Ok, I will try latest version