libimobiledevice/usbmuxd #35: make fails for usbmuxd

mwilliams42, 2015-01-29:
On Raspberry Pi, running rasbian.
After running ./autogen.sh

root@raspberrypi:/iphone_libs/usbmuxd# make
make  all-recursive
make[1]: Entering directory '/iphone_libs/usbmuxd'
Making all in src
make[2]: Entering directory '/iphone_libs/usbmuxd/src'
  CCLD   usbmuxd
/usr/bin/ld: usbmuxd-utils.o: undefined reference to symbol 'clock_gettime@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/librt.so.1: error adding symbols: DSO missing from command line
collect2: ld returned 1 exit status
Makefile:349: recipe for target 'usbmuxd' failed
make[2]: *** [usbmuxd] Error 1
make[2]: Leaving directory '/iphone_libs/usbmuxd/src'
Makefile:353: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/iphone_libs/usbmuxd'
Makefile:284: recipe for target 'all' failed
make: *** [all] Error 2
root@raspberrypi:/iphone_libs/usbmuxd#

help please?

mwilliams42, 2015-01-29:
If I use the 1.0.9 version the compile works.

FunkyM, 2015-01-29:
You probably have an older glibc that needs "-lrt" flags. What is your glibc version?

mwilliams42, 2015-01-29:
root@raspberrypi:/iphone_libs/ideviceinstaller-1.1.0# dpkg -s libc6
Package: libc6
Status: install ok installed
Priority: required
Section: libs
Installed-Size: 8705
Maintainer: GNU Libc Maintainers <debian-glibc@lists.debian.org>
Architecture: armhf
Multi-Arch: same
Source: eglibc
Version: 2.13-38+rpi2+deb7u7
Provides: glibc-2.13-1
Depends: libc-bin (= 2.13-38+rpi2+deb7u7), libgcc1
Suggests: glibc-doc, debconf | debconf-2.0, locales
Breaks: locales (<< 2.13), locales-all (<< 2.13), nscd (<< 2.13)
Conflicts: prelink (<= 0.0.20090311-1), tzdata (<< 2007k-1), tzdata-etch
Conffiles:
 /etc/ld.so.conf.d/arm-linux-gnueabihf.conf fcce838d54cc444500f01049db2aae8c
Description: Embedded GNU C Library: Shared libraries
 Contains the standard libraries that are used by nearly all programs on
 the system. This package includes shared versions of the standard C library
 and the standard math library, as well as many others.
Homepage: http://www.eglibc.org