libimobiledevice/usbmuxd #47: Static build fails

zgoda-mobica, 2015-12-02:
I have configured latest master code as follows:

```
$ ./configure --enable-static --disable-shared
```

Final configuration:

```
Configuration for usbmuxd 1.1.1:
-------------------------------------------

  install prefix ............: /usr/local
  preflight worker support ..: yes
  activation method .........: systemd
  systemd unit directory ....: /lib/systemd/system
  udev rules directory ......: /lib/udev/rules.d
  compiler ..................: gcc
```

Then make fails with:

```
# make
/usr/bin/make  all-recursive
make[1]: Entering directory '/root/src/usbmuxd-master'
Making all in src
make[2]: Entering directory '/root/src/usbmuxd-master/src'
  CC       usbmuxd-client.o
  CC       usbmuxd-device.o
  CC       usbmuxd-preflight.o
  CC       usbmuxd-log.o
  CC       usbmuxd-usb.o
  CC       usbmuxd-utils.o
  CC       usbmuxd-conf.o
  CC       usbmuxd-main.o
  CCLD     usbmuxd
/usr/local/lib/libimobiledevice.a(utils.o): In function `string_concat':
/root/src/libimobiledevice-master/common/utils.c:71: multiple definition of `string_concat'
usbmuxd-utils.o:/root/src/usbmuxd-master/src/utils.c:171: first defined here
/usr/local/lib/libimobiledevice.a(utils.o): In function `buffer_read_from_filename':
/root/src/libimobiledevice-master/common/utils.c:204: multiple definition of `buffer_read_from_filename'
usbmuxd-utils.o:/root/src/usbmuxd-master/src/utils.c:215: first defined here
/usr/local/lib/libimobiledevice.a(utils.o): In function `buffer_write_to_filename':
/root/src/libimobiledevice-master/common/utils.c:235: multiple definition of `buffer_write_to_filename'
usbmuxd-utils.o:/root/src/usbmuxd-master/src/utils.c:245: first defined here
/usr/local/lib/libimobiledevice.a(utils.o): In function `plist_read_from_filename':
/root/src/libimobiledevice-master/common/utils.c:246: multiple definition of `plist_read_from_filename'
usbmuxd-utils.o:/root/src/usbmuxd-master/src/utils.c:256: first defined here
/usr/local/lib/libimobiledevice.a(utils.o): In function `plist_write_to_filename':
/root/src/libimobiledevice-master/common/utils.c:271: multiple definition of `plist_write_to_filename'
usbmuxd-utils.o:/root/src/usbmuxd-master/src/utils.c:281: first defined here
collect2: error: ld returned 1 exit status
Makefile:407: recipe for target 'usbmuxd' failed
make[2]: *** [usbmuxd] Error 1
make[2]: Leaving directory '/root/src/usbmuxd-master/src'
Makefile:411: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/src/usbmuxd-master'
Makefile:342: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm building in armhf Debian Jessie chrooted.

nikias, 2018-04-04:
Why are you building static? This project and all of its dependencies are not supposed to be built statically.

closed 2018-04-04 by nikias

MarSoft, 2020-03-24:
Static building is sometimes useful, especially when you want to make an easily redistributable cross-platform tool.
Here is my current approach in achieving it. Did not yet test for runtime errors, but linking succeeds:
You'll need to make several changes to `libimobiledevice` sources:
1. In `common/thread.c`, remove all contents;
2. In `common/thread.h`, prepend each function definition with `extern ` keyword;
3. This will break building binary tools, so disable them: in `Makefile.am`, remove `tools` and `docs` from `SUBDIRS`, and remove `EXTRA_DIST` line.
4. Now run `./autogen.sh --enable-shared=no` and other flags you may need (probably you'll want to use custom `--prefix`).

voltagex, 2021-05-15:
@MarSoft are you able to provide a patch for this?

MarSoft, 2021-05-21:
@voltagex, here is the script I use on libimobiledevice/libimobiledevice@798601d to prepare it for static build:
```
	# remove code for thread funcs which is duplcated with usbmuxd
	cat /dev/null > common/thread.c
	# convert thread funcs to extern ones
	sed -i common/thread.h -e 's/^\([iv][no][ti]\)/extern \1/'
	# don't build tools (and docs), only lib itself
	sed -i Makefile.am -e 's/ tools docs$$//; s/^EXTRA_DIST =.*//'
	# apply fix from libplist#114
	sed -i src/idevice.c -e 's/WIN32/DLL_EXPORT/'
```
Newer commits don't work for me because of socket stuff change introduced in libimobiledevice/libimobiledevice@2c83cae88c50217f0fbbff5c3d2eab5f252f164e.