libimobiledevice/usbmuxd #49: Cheksum mismatch for usb

yv0ry, 2015-12-15:
Hello, 

Sorry, i've a problem with the install of usbmuxd,

Check that :

```
--->  Verifying checksums for usbmuxd
Error: Checksum (md5) mismatch for usbmuxd-1.0.7.tar.bz2
Error: Checksum (sha1) mismatch for usbmuxd-1.0.7.tar.bz2
Error: Checksum (rmd160) mismatch for usbmuxd-1.0.7.tar.bz2
Error: org.macports.checksum for port usbmuxd returned: Unable to verify file checksums
Please see the log file for port usbmuxd for details:
    /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_release_tarballs_ports_devel_usbmuxd/usbmuxd/main.log
To report a bug, follow the instructions in the guide:
    http://guide.macports.org/#project.tickets
Error: Processing of port usbmuxd failed
```
Do you have a solution for this problem ?

Thanks for advance, :)

Mark

yv0ry, 2015-12-15:
When i tape : sudo port -v checksum usbmuxd

I have : 

'''
The correct checksum line may be:
checksums           md5     9da6792773ac37015dceede466a2297e \
                    sha1    da40b4c28d355b2181b88ac8d33d6d47abff514a \
                    rmd160  c7483361e1239ffd673b52e3ce93c1be9aad8b8c
Error: org.macports.checksum for port usbmuxd returned: Unable to verify file checksums
Warning: targets not executed for usbmuxd: org.macports.checksum
Please see the log file for port usbmuxd for details:
    /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_release_tarballs_ports_devel_usbmuxd/usbmuxd/main.log
To report a bug, follow the instructions in the guide:
    http://guide.macports.org/#project.tickets
Error: Processing of port usbmuxd failed
'''

nikias, 2016-01-28:
Sorry, you need to report this issue to MacPorts.

closed 2016-01-28 by nikias