libimobiledevice/usbmuxd #52: Question is Here: I have connected an iPhone to Windows over USB. Is there a way to show my ideviceinfo on the remote Linux System machine whose network conected the same local ?

JasonHXiang, 2016-01-16:


milesrichardson, 2016-05-25:
@JasonHXiang did you ever figure this out? I'm thinking it could be accomplished with a SOCKS proxy and somehow instructing ideviceinfo to use the address of the SOCKS proxy instead of the local usbmuxd socket? But I don't know what I'm talking about...

rufushuang, 2016-07-08:
Got it done.
For example, using the python-client of usbmuxd.
Change HOST in `tcprelay.py`

    HOST = "localhost"

to the ip you want serve on lan.

The local ip on lan can be obtained by:

    import socket
    HOST = socket.gethostbyname(socket.gethostname())

FunkyM, 2016-07-12:
You can modify the sources of the server (usbmuxd, make sure to revert the "only listen locally" socket patch) and let the client connect to it remotely. Such functionality is ment for usbmuxd. This feature has to be added to usbmuxd in order for proper support of network connections. We have done some PoC of this before but integration was not done in usbmuxd yet.

closed 2016-07-12 by FunkyM