libimobiledevice/usbmuxd #53: Porting to Android

oct-test, 2016-03-30:
What are the steps to port this library to Android using NDK?

nickplee, 2016-08-23:
I accomplished this [here](https://blog.tendigi.com/finally-android-on-the-iphone-99b0c8b3adc) by building my own AOSP rom with `usbmuxd` integrated as a package.  Worked pretty well.

Tdogb, 2016-08-23:
OMG I didn't realize u were actually nick lee!! I am trying to do the exact same thing. I am having trouble making the iphone app. And I have no idea how to code my own operating system. I have experimented with usbmuxd between my computer and my phone however the connection (at least the vnc connection didn't seem very fast). are u using vnc? Is your project open source?

Tdogb, 2016-08-25:
I figured out how to port usbmuxd, all I need now is the app, can you give me some pointers other than the ones on your website.
Thanks!

MarcusXie, 2016-08-29:
How to port to Android? I need some help...

MarcusXie, 2016-08-29:
@Tdogb I can't open nick lee's link, could U give me some help?
Thanks!

pokmis, 2017-12-07:
Can anyone share the steps they took to get this up and running on Android?

XanaduAPI, 2020-12-05:
以下操作基于 Ubuntu / Windows 10 子系统 Ubuntu

~/.bashrc
[image]

sudo apt update
sudo apt upgrade
sudo apt install libtool
sudo apt install automake
sudo apt install autoconf
sudo apt install python
sudo apt install pkg-config
sudo apt install pkgconf

export ANDROID_ARM_64=$ANDROID_HOME/ARM_64
$NDK_HOME/build/tools/make-standalone-toolchain.sh --platform=android-24 --arch=arm64 --install-dir=$ANDROID_ARM_64

export PKG_CONFIG_PATH=$ANDROID_ARM_64/lib/pkgconfig
export PATH=$ANDROID_ARM_64:$PATH
export PATH=$PKG_CONFIG_PATH:$PATH
export PATH=$ANDROID_ARM_64/bin:$PATH

libplist
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 --without-cython CFLAGS=-static LDFLAGS=-static

libusbmuxd
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 CFLAGS=-static LDFLAGS=-static

openssl
./Configure --cross-compile-prefix=aarch64-linux-android- --prefix=$ANDROID_ARM_64 linux-armv4 no-asm no-ui no-shared enable-weak-ssl-ciphers CFLAGS=-static LDFLAGS=-static

libimobiledevice
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 --without-cython CFLAGS=-static LDFLAGS=-static

libusb
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 --disable-udev CFLAGS=-static LDFLAGS=-static

usbmuxd
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 CFLAGS=-static LDFLAGS=-static

tuanjnfor, 2020-12-18:
Dear @XanaduAPI 
Pls help to let me know how I can folder ARM_64 in ANDROID_HOME path.
It only has sdk folder.
Thank you so much!

cczhr, 2021-02-27:
If you need the executable files and library files transplanted from libimobiledevice open source project to Android platform, you can go to https://github.com/cczhr/libimobiledevice_android to view them. I compiled some of them .Hope to help you

XanaduAPI, 2021-03-01:
> If you need the executable files and library files transplanted from libimobiledevice open source project to Android platform, you can go to https://github.com/cczhr/libimobiledevice_android to view them. I compiled some of them .Hope to help you

I have compiled files too, but will yours work properly without jailbreaking? Or did you test it?

cczhr, 2021-03-02:
@XanaduAPI Android devices need root permission to run  .I made a small project based on this library. Here is the demo video link：https://www.bilibili.com/video/BV1MZ4y1P7K3

XanaduAPI, 2021-03-02:
> @XanaduAPI Android devices need root permission to run .I made a small project based on this library. Here is the demo video link：https://www.bilibili.com/video/BV1MZ4y1P7K3

If you need to jailbreak to run it, it's not necessary. Because I'm already running it on regular ARM(Linux).

tangwanchao, 2021-07-15:
@XanaduAPI  你是怎么做到的,方便微信联系吗?  有偿咨询
我的微信 twcjjww,
电话 18284228392