libimobiledevice/usbmuxd #70: usbmuxd does not work on macOS Sierra (10.12.5)

Jun-GitHub, 2017-06-09:
I build usbmuxd on macOS Sierra (10.12.5), everything is fine, but when I connect my iphone 7(iOS 10.2.1) it falls into neverending loop
**writing such logs:**
```
[08:32:55.406][3] usbmuxd v1.1.1 starting up
[08:32:55.407][4] Creating socket
[08:32:55.407][3] Successfully dropped privileges to 'hj'
[08:32:55.407][4] Initializing USB
[08:32:55.413][4] Registering for libusb hotplug events
[08:32:55.414][4] 0 devices detected
[08:32:55.414][3] Initialization complete
[08:33:10.215][4] New client on fd 7
[08:33:10.217][4] Client 7 connection closed
[08:33:10.217][4] Disconnecting client fd 7
[08:33:15.398][4] New client on fd 7
[08:33:15.398][2] Attempted to connect to nonexistent device 2
[08:33:15.398][4] Client 7 connection closed
[08:33:15.398][4] Disconnecting client fd 7
[08:33:15.399][4] New client on fd 7
[08:33:15.399][4] Client 7 connection closed
[08:33:15.399][4] Disconnecting client fd 7
[08:33:15.402][4] New client on fd 7
[08:33:15.402][2] Attempted to connect to nonexistent device 2
[08:33:15.402][4] Client 7 connection closed
[08:33:15.402][4] Disconnecting client fd 7
```

kamstrup, 2017-08-11:
i can't get usbmuxd to work on Sierra either. It runs fine, but is not able to detect the ipod that I plugged in (also tried re-plugging a few times). I can see in the daemon output that 'idevice_id -l' connects to the daemon alright - it looks like it just doesn't see the device.

Could it be usb-c / libusb related perhaps..?

I disabled the system usbmuxd with launchctl.

kamstrup, 2017-08-11:
Tested if it was related to https://github.com/libusb/libusb/issues/303, by upgrading libusb to HEAD with brew (to libusb 1fe3db7, which includes the fix). But am seeing the same behaviour - the device not being detected.

tianshanfz, 2017-11-12:
Have you found any solutions yet?  Devices are not detected on Mac OS 10.13 either.

kamstrup, 2017-11-13:
I ended up giving up and solved my problem using usbmuxd from Apple

tianshanfz, 2017-11-14:
@kamstrup  [https://github.com/libusb/libusb/issues/290](url). This worked for me.

nikias, 2018-04-04:
The problem is not usbmuxd but libusb.

closed 2018-04-04 by nikias