libimobiledevice/usbmuxd #81: No possible to perform operations on 2+ devices

michalzimniak, 2017-09-19:
I guess is a usbmux issue, but not sure. Please find attached files.
[device1.txt](https://github.com/libimobiledevice/usbmuxd/files/1313840/device1.txt)
[device2.txt](https://github.com/libimobiledevice/usbmuxd/files/1313842/device2.txt)
[usbmuxd.txt](https://github.com/libimobiledevice/usbmuxd/files/1313841/usbmuxd.txt)

mexmer, 2017-09-19:
your second log doesnt' have any useable information

No data to read
No data to read

that's all there is

there is not even hint of command issued or whatever.

keep in mind, when you have multiple devices attached you must always  specify UDID, there is no such thing as default device

michalzimniak, 2017-09-25:
Hi, I will send you more detailed logs.

I found strange behavior of usbmuxd if running as foreground task (# usbmuxd -f -d command):

if devices connected before usbmuxd started:

```
# usbmuxd -f -v
[01:59:27.884][3] usbmuxd v1.1.1 starting up
[01:59:27.885][4] Creating socket
[01:59:27.885][4] Initializing USB
[01:59:27.888][4] Registering for libusb hotplug events
[01:59:27.888][4] Found new device with v/p 05ac:12a8 at 1-55
[01:59:27.889][4] Found interface 1 with endpoints 04/85 for device 1-55
[01:59:27.889][4] Using wMaxPacketSize=512 for device 1-55
[01:59:27.889][4] USB Speed is 480 MBit/s for device 1-55
[01:59:27.889][3] Connecting to new device on location 0x10037 as ID 1
[01:59:27.889][4] Found new device with v/p 05ac:12a8 at 1-53
[01:59:27.890][4] Found interface 1 with endpoints 04/85 for device 1-53
[01:59:27.890][3] Connected to v2.0 device 1 on location 0x10037 with serial number 1cd53f1ea626f781627e7bc8fdc0437f4c3e596e
[01:59:27.890][4] preflight_worker_handle_device_add: Starting preflight on device 1cd53f1ea626f781627e7bc8fdc0437f4c3e596e...
[01:59:27.890][4] Using wMaxPacketSize=512 for device 1-53
[01:59:27.890][4] USB Speed is 480 MBit/s for device 1-53
[01:59:27.890][3] Connecting to new device on location 0x10035 as ID 2
[01:59:27.890][4] 2 devices detected
[01:59:27.890][3] Initialization complete
```

if devices connected when usbmuxd running:

```
[02:56:03.381][4] Device 1-55 RX aborted due to error or disconnect
[02:56:03.381][4] Device 1-55 RX aborted due to error or disconnect
[02:56:03.381][3] Removed device 1 on location 0x10037
[02:56:03.381][4] Device 1-55 RX aborted due to error or disconnect
[02:56:03.622][4] Caught SIGUSR1/2 but this instance was not started with "--enable-exit", ignoring.
[02:56:03.638][4] Caught SIGUSR1/2 but this instance was not started with "--enable-exit", ignoring.
[02:56:04.947][4] Device 1-53 RX aborted due to error or disconnect
[02:56:04.947][4] Device 1-53 RX aborted due to error or disconnect
[02:56:04.947][4] Device 1-53 RX aborted due to error or disconnect
[02:56:04.947][3] Removed device 2 on location 0x10035
[02:56:05.158][4] Caught SIGUSR1/2 but this instance was not started with "--enable-exit", ignoring.
[02:56:05.170][4] Caught SIGUSR1/2 but this instance was not started with "--enable-exit", ignoring.
[02:56:08.910][4] New client on fd 11 (pid 4657)
[02:56:08.910][4] Client 11 connection closed
[02:56:08.910][4] Disconnecting client fd 11
[02:56:08.910][4] Found new device with v/p 05ac:12a8 at 1-56
[02:56:08.910][3] Could not get old configuration descriptor for device 1-56: -5
[02:56:08.910][4] Setting configuration for device 1-56, from 0 to 4
[02:56:08.913][4] Found interface 1 with endpoints 04/85 for device 1-56
[02:56:08.913][2] Could not get serial number for device 1-56: -6
[02:56:10.002][4] New client on fd 11 (pid 1409)
[02:56:10.102][4] Client 11 connection closed
[02:56:10.102][4] Disconnecting client fd 11
[02:56:10.383][4] New client on fd 11 (pid 4687)
[02:56:10.383][4] Client 11 connection closed
[02:56:10.383][4] Disconnecting client fd 11
[02:56:10.383][4] Found new device with v/p 05ac:12a8 at 1-57
[02:56:10.383][3] Could not get old configuration descriptor for device 1-57: -5
[02:56:10.383][4] Setting configuration for device 1-57, from 0 to 4
[02:56:10.385][4] Found interface 1 with endpoints 04/85 for device 1-57
[02:56:10.385][2] Could not get serial number for device 1-57: -6
[02:56:11.001][4] New client on fd 11 (pid 1409)
[02:56:11.101][4] Client 11 connection closed
[02:56:11.102][4] Disconnecting client fd 11
^C[02:56:11.290][3] Caught signal 2, exiting
[02:56:11.291][4] Event processing interrupted
[02:56:11.291][3] usbmuxd shutting down
[02:56:11.391][4] Disconnecting client fd 14
```

The main problem is that I cannot restart usbmuxd if new device connection detected, because usbmuxd is used by idevicerestore to send data to devices, so usbmuxd restart will break the process (and brick the device).

Please also take a look at this one (one device connected only):

```
# idevicepair  pair
ERROR: Please accept the trust dialog on the screen of device 1cd53f1ea626f781627e7bc8fdc0437f4c3e596e, then attempt to pair again.
# idevicepair  pair
SUCCESS: Paired with device 1cd53f1ea626f781627e7bc8fdc0437f4c3e596e
# ideviceinfo
ERROR: Could not connect to lockdownd, error code -8
```