libimobiledevice/usbmuxd #83: Support using TCP sockets on Unix operating systems

qmfrederik, 2017-11-04:
I've done some work in my repositories to make `usbmuxd` listen on a TCP socket instead of a Unix socket on Linux and macOS if you specify a "--tcp" flag; there were some changes in libusbmuxd as well to support libusbmuxd connecting to a TCP socket.

The idea behind is that if you run usbmuxd in a Docker container, you'll probably want to expose a TCP socket rather than a Unix socket so that other docker containers can easily connect to that usbmuxd.

I'd be happy to create a PR for that but first I'd like to check whether there's any interest here in integrating that change :).

tianshanfz, 2017-11-14:
It's wonderful. I'm thinking about how to expose specific devices to different clients while unix socket makes it difficult.

qmfrederik, 2017-11-14:
@tianshanfz Meanwhile, you can always use tools like `socat` to create a TCP socket that just forwards all traffic to a Unix socket.

Dantee296, 2019-12-09:
@qmfrederik hi i would like to try this feature , where it is available to test ?

qmfrederik, 2019-12-18:
@Dantee296 Here: https://github.com/libimobiledevice-win32/usbmuxd/commit/e26087ca60ce626f0a2c170168e1d1b119561b2f

qmfrederik, 2021-01-20:
This was fixed via 6b3e64140419013834589dfcff25cbf9af1b6168

closed 2021-01-20 by qmfrederik