libimobiledevice/usbmuxd #85: usbmuxd fails to open occasionally

Blefish, 2017-12-28:
I am using usbmuxd with systemd and I have the udev rules set up to start systemd service.
Sometimes when device does a reboot, usbmuxd closes itself and fails to come back up again.

Dec 28 11:56:18 xx usbmuxd[2506]: [22:37:50.159][3] usbmuxd v1.1.1 starting up
Dec 28 11:56:18 xx usbmuxd[2506]: [22:37:50.159][0] Could not open lockfile

On other cases it simply shuts down without a failure.
This can be replicated easily by starting iPhone in recovery mode and doing firmware flash with idevicerestore.

My current workaround is to disable the udev remove event from 39-usbmuxd.rules. However I noticed that usbmuxd memory usage keeps increasing every time I plug the phone in, hence #84

Blefish, 2017-12-29:
It seems as if udev event is sent to signal shutdown if no phones are connected and soon afterwards another udev event for startup.

Blefish, 2017-12-29:
From the logs:

Dec 29 10:59:00 xx kernel: [  119.334963] usb 1-1.3.1.1: new high-speed USB device number 9 using ehci-pci
Dec 29 10:59:00 xx kernel: [  119.429886] usb 1-1.3.1.1: New USB device found, idVendor=05ac, idProduct=12a8
Dec 29 10:59:00 xx kernel: [  119.429890] usb 1-1.3.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 29 10:59:00 xx kernel: [  119.429891] usb 1-1.3.1.1: Product: iPhone
Dec 29 10:59:00 xx kernel: [  119.429893] usb 1-1.3.1.1: Manufacturer: Apple Inc.
Dec 29 10:59:00 xx kernel: [  119.429894] usb 1-1.3.1.1: SerialNumber: ffffffffffffffffffffffffffffffffffffffff
Dec 29 10:59:00 xx mtp-probe: checking bus 1, device 9: "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.1/1-1.3.1.1"
Dec 29 10:59:00 xx mtp-probe: bus: 1, device: 9 was not an MTP device
Dec 29 10:59:00 xx systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.963][3] usbmuxd v1.1.1 starting up
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.964][3] Successfully dropped privileges to 'usbmux'
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.967][3] Could not get old configuration descriptor for device 1-9: -5
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.969][3] Connecting to new device on location 0x10009 as ID 1
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.970][3] Initialization complete
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.970][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.970][3] Caught SIGUSR1, checking if we can terminate (no more devices attached)...
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:58.970][3] usbmuxd shutting down
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:59.019][3] Connected to v2.0 device 1 on location 0x10009 with serial number ffffffffffffffffffffffffffffffffffffffff
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:59.070][3] Removed device 1 on location 0x10009
Dec 29 10:59:00 xx usbmuxd[2986]: [03:01:59.071][3] Shutdown complete

FYI, inserting sleep(1) here seems to fix the issue:
https://github.com/libimobiledevice/usbmuxd/blob/master/src/main.c#L512

nikias, 2018-01-08:
So basically it seems like the newly started instance gets a SIGUSR1 even though it was just started, right?

If you don't mind, can you modify https://github.com/libimobiledevice/usbmuxd/blob/master/src/main.c#L501 to
```
usbmuxd_log(LL_FATAL, "Could not open lockfile: %s", strerror(errno));
```
So we can get the actual error why it fails.

Blefish, 2018-01-08:
Yes!

I'm not sure if lockfile is a completely different problem or related to this. I will add debugging and if I am able to reproduce I will let you know.

Blefish, 2018-01-23:
Seems like #86 together with my patch fixes this issue. Adding additional filter for ENV{DEVTYPE}=="usb_device" and now I get only one event for connection and one for disconnection

```
From 1df6cbbb3a0cdae43ab9cef00540a08769216471 Mon Sep 17 00:00:00 2001
From: Rudolf Tammekivi <rudolf.tammekivi@gmail.com>
Date: Mon, 22 Jan 2018 11:26:26 +0200
Subject: [PATCH] udev: Handle all events only once

This fixes spontaneous USB reconfiguration causing "remove" event to be sent to
the daemon, which causes it to close itself if it did not have enough time to
connect to phone.
---
 udev/39-usbmuxd.rules.in | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/udev/39-usbmuxd.rules.in b/udev/39-usbmuxd.rules.in
index 7d14d22..140869e 100644
--- a/udev/39-usbmuxd.rules.in
+++ b/udev/39-usbmuxd.rules.in
@@ -1,10 +1,10 @@
 # usbmuxd (Apple Mobile Device Muxer listening on /var/run/usbmuxd)
 
 # systemd should receive all events relating to device
-SUBSYSTEM=="usb", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", TAG+="systemd"
+SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", TAG+="systemd"
 
 # Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
-SUBSYSTEM=="usb", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
+SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
 
 # Exit usbmuxd when the last device is removed
-SUBSYSTEM=="usb", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="remove", RUN+="@sbindir@/usbmuxd -x"
+SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="remove", RUN+="@sbindir@/usbmuxd -x"
-- 
2.5.0

```

nikias, 2018-01-31:
@Blefish I merged #86 and your patch too.

Blefish, 2018-01-31:
Thanks, I will close this as it seems solved.

closed 2018-01-31 by Blefish