libimobiledevice/usbmuxd #88: usbmuxd does not work with iOs 11.1

MinhDucLe2412, 2018-01-29:
Hello,

I've just cloned the newest Version on Git and build everything and it works fine with ios 10. Then i tried with ios 11 and it says "ERROR: Could not connect to lockdownd, error code -3" although the Pairing process always succesful. 

Btw ideviceinfo gives the same error, but ideviceinfo -s works fine.

I'm using Raspbian

mexmer, 2018-01-29:
did you also recompile usbmuxd, not just libimobile?

MinhDucLe2412, 2018-01-29:
Thank you for the fast reply.

Yes, i compiled every single one of them(libimobiledevice, libusbmuxd, ifuse, libplist) and it worked like a charm with ios 10.3.3

But when i changed to ios 11.1.2 on iphone 7 it just stop working with those errors. The pairing process was successful, ideviceinfo -s worked but ideviceinfo and screenshot do not.

When i looked at the verbose part of usbmuxd i found out that the connection was reset from device somehow, and system log shoed me that iphnet-pair failed with error code 255.

If you want i can give you the log tomorrow, when i'm at work

MinhDucLe2412, 2018-01-30:
Here's what i did and the results:
pi@raspberrypi:/home $ idevicepair unpair
SUCCESS: Unpaired with device 8e2e8280b4172f8c322b1103f2b61041aa08f038
pi@raspberrypi:/home $ idevicepair pair
ERROR: Please accept the trust dialog on the screen of device 8e2e8280b4172f8c322b1103f2b61041aa08f038, then attempt to pair again.
pi@raspberrypi:/home $ idevicepair pair
SUCCESS: Paired with device 8e2e8280b4172f8c322b1103f2b61041aa08f038
pi@raspberrypi:/home $ 

And the syslog:

an 30 07:30:21 raspberrypi mtp-probe: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Jan 30 07:30:21 raspberrypi mtp-probe: bus: 1, device: 6 was not an MTP device
Jan 30 07:30:21 raspberrypi systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.637][3] usbmuxd v1.1.1 starting up
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.642][3] Successfully dropped privileges to 'usbmux'
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.658][3] Could not get old configuration descriptor for device 1-6: -5
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.663][3] Initialization complete
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.663][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.664][3] Connecting to new device on location 0x10006 as ID 1
Jan 30 07:30:21 raspberrypi usbmuxd[921]: [00:00:59.668][3] Connected to v2.0 device 1 on location 0x10006 with serial number 8e2e8280b4172f8c322b1103f2b61041aa08f038
Jan 30 07:30:21 raspberrypi kernel: [   59.693705] ipheth 1-1.2:4.2: Apple iPhone USB Ethernet device attached
Jan 30 07:30:21 raspberrypi kernel: [   59.694047] usbcore: registered new interface driver ipheth
Jan 30 07:30:21 raspberrypi systemd-udevd[926]: Process 'ipheth-pair' failed with exit code 255.
Jan 30 07:30:21 raspberrypi dhcpcd[347]: eth1: waiting for carrier
Jan 30 07:30:21 raspberrypi dhcpcd[347]: eth1: carrier acquired
Jan 30 07:30:21 raspberrypi kernel: [   59.782598] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
Jan 30 07:30:21 raspberrypi dhcpcd[347]: eth1: IAID f3:02:ad:d3
Jan 30 07:30:21 raspberrypi dhcpcd[347]: eth1: adding address fe80::705b:88d0:e616:2f94
Jan 30 07:30:21 raspberrypi dhcpcd[347]: eth1: carrier lost
Jan 30 07:30:21 raspberrypi dhcpcd[347]: eth1: deleting address fe80::705b:88d0:e616:2f94
Jan 30 07:30:22 raspberrypi gvfs-gphoto2-vo[715]: device (null) has no BUSNUM property, ignoring
Jan 30 07:30:22 raspberrypi gvfs-gphoto2-vo[715]: device (null) has no BUSNUM property, ignoring
Jan 30 07:30:22 raspberrypi gvfs-gphoto2-vo[715]: device (null) has no BUSNUM property, ignoring
Jan 30 07:30:22 raspberrypi gvfs-gphoto2-vo[715]: device (null) has no BUSNUM property, ignoring
Jan 30 07:30:22 raspberrypi gvfs-afc-volume-monitor[710]: creating volume for device uuid '8e2e8280b4172f8c322b1103f2b61041aa08f038'
Jan 30 07:30:22 raspberrypi gvfsd[615]: dbus_mount_reply: Error from org.gtk.vfs.Mountable.mount(): Unhandled Lockdown error (-3)

mexmer, 2018-01-30:
could you post ideviceinfo -d (need to have with debug compiled tho')
i did pair my ipad (11.2.2) and it's working without issues with usbmuxd and libimobile from current master - running latest debian without systemd.

MinhDucLe2412, 2018-01-30:
And how do i enable debug mode?

mexmer, 2018-01-30:
before compilation run configure --enable-debug-mode, then compile libimobile, all binaries will be compiled with debug mode enabled.

then you can use binaries with -d parametr do display debug output, for example ideviceinfo -d.
without debug mode, they will not produce much different output with our without -d parameter.

MinhDucLe2412, 2018-01-30:
is that --enable-debug-code or --enable-debug-mode?

According to https://github.com/libimobiledevice/libimobiledevice is code

MinhDucLe2412, 2018-01-30:
Here's the output:
Configuration for libimobiledevice 1.2.1:
-------------------------------------------

  Install prefix: .........: /usr/local
  Debug code ..............: yes
  Python bindings .........: no
  SSL support backend .....: OpenSSL

And output of ideviceinfo -d is still the same

mexmer, 2018-01-30:
right, it's debug-code, was not on linux, to confirm proper name.

you sure you running newly compiled ideviceinfo with -d (or --debug) parameter?

nikias, 2018-01-30:
maybe you are still using the old installed binaries instead of the new compiled ones? If you type
```which ideviceinfo```
does it show ```/usr/local/bin/ideviceinfo``` or something else?
That path is where you installed it to according to your latest comment, however system default might be in ```/usr/local/bin``` (in case you have any precompiled binaries installed).

MinhDucLe2412, 2018-01-30:
oh, you're right, i still run the old ones >_<

I found the new one and i worked! The issue can be closed.

Thank you very much

closed 2018-01-30 by nikias