libimobiledevice/usbmuxd #89: usbmuxd can not running

ubntpeter, 2018-02-26:
My environment is
openwrt 15.05 
kerenl 3.18.23
SoC QCA9531

1. I Add /etc/rc.d/S99usbmuxd which to execute usbmuxd when system startup.
But I can not find usbmuxd running in background.

2. I try to manually execute it by 
usbmuxd -v -f
[00:02:11.193][3] usbmuxd v1.1.1 starting up
[00:02:11.193][1] Another instance is already running (pid 0). exiting.

ls /var/run/
there is only usbmuxd.pid 
can not found usbmuxd

cat /var/run/usbmuxd.pid
no any data in it.

I try trace the code usbmuxd-1.1.1/src/main.c
It seem fail at  "if (lock.l_type != F_UNLCK) "  at line 511.

Could give me any suggestion ?
:)

mexmer, 2018-02-26:
either delete usbmuxd.pid or change usbmuxd.pid permission, so usbmuxd can write into it. i assume usbmuxd is running under own user/group on your system.

other posibility will be, that your filesystem or kernel doesn't support file locking, which i think is unlikely.

if you still have issues with it, you should ask on openwrt forums, since openwrt has officialy maintained package for usbmuxd.

ubntpeter, 2018-02-26:
Thanks Mexmer.

After I  add CONFIG_FILE_LOCKING=y in kernel config, the usbmuxd can  run when system startup.

closed 2018-02-26 by ubntpeter

nikias, 2018-02-26:
@ubntpeter Thanks for pointing this out. I think I need to add something for autoconf to detect if flock is actually available or not.

mexmer, 2018-02-27:
@nikias no need to do check for flock at runtime, just make wrapper around all parts, where you are using flock, then inside wrapper check for error EINVAL (invalid operation), which should pop on FS that doesn't support flock. in case of this error, just use alternate blocking method.

nikias, 2018-02-27:
@mexmer thanks