libimobiledevice/usbmuxd #95: Reciving lockdown error -8 (LOCKDOWN_E_MUX_ERROR) on latest push to master

guywithface, 2018-10-02:
Unable to connect to device after the latest push to master. Using an iOS 11 device.

Command: usbmuxd -f -v
Verbose output:
[17:10:02.658][3] usbmuxd v1.1.1 starting up
[17:10:02.658][4] Creating socket
[17:10:02.658][4] Initializing USB
[17:10:02.667][4] Registering for libusb hotplug events
[17:10:02.667][4] Found new device with v/p 05ac:12a8 at 3-13
[17:10:02.667][4] Found interface 1 with endpoints 04/85 for device 3-13
[17:10:02.668][4] Using wMaxPacketSize=512 for device 3-13
[17:10:02.668][4] USB Speed is 480 MBit/s for device 3-13
[17:10:02.668][4] 1 device detected
[17:10:02.668][3] Initialization complete
[17:10:02.668][4] Got lang ID 1033 for device 3-13
[17:10:02.668][4] Got serial '<MY_SERIAL>' for device 3-13
[17:10:02.668][3] Connecting to new device on location 0x3000d as ID 1
[17:10:02.668][3] Connected to v2.0 device 1 on location 0x3000d with serial number <MY_SERIAL>
[17:10:02.668][4] preflight_worker_handle_device_add: Starting preflight on device <MY_SERIAL>...
[17:10:02.668][4] New client on fd 13 (self)
[17:10:02.668][2] Attempted to connect to nonexistent device 0
**[17:10:02.668][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device faf6944a7ef4f4cf2571b36079131d7001cc972f, lockdown error -8**
[17:10:02.668][4] Client 13 connection closed
[17:10:02.668][4] Disconnecting client fd 13

Works again once reverting to the previous commit to master (tree commit d33402036d563667a76910dbfec50a37b622d3a5).

mexmer, 2018-10-02:
you need to update all important parts usbmuxd, libusmuxd, libimobile

nikias, 2018-10-02:
@guywithface you need to use latest libimobiledevice. I introduced a change that made the most recent commit necessary.

guywithface, 2018-10-03:
@nikias I went ahead and confirmed I'm pulling the latest from libplist, libusbmuxd, libimobiledevice, and usbmuxd. Here's part of my installation script:

cd ~; git clone https://github.com/libimobiledevice/libplist.git; \
	cd libplist; ./autogen.sh; make; make install

cd ~; git clone https://github.com/libimobiledevice/libusbmuxd.git; \
	cd libusbmuxd; ./autogen.sh; make; make install

cd ~; git clone https://github.com/libimobiledevice/libimobiledevice.git; \
	cd libimobiledevice; ./autogen.sh; make; make install

cd ~; git clone https://github.com/libimobiledevice/usbmuxd.git; \
        cd usbmuxd; ./autogen.sh; make; make install

Anything further I can do on my end to help debug this?

candoizo, 2018-11-11:
today noticed an update starting giving me the same error 

[2018-11-07 08:06] [ALPM] upgraded usbmuxd (1.1.0+28+g46bdf3e-1 -> 1.1.0+48+g1cc8b34-1)

I am on Arch Linux so I should have all the latest changes. I have tried different devices on iOS 10.2, iOS 11.0.2, and 11.2.2, but all of them also give me lockdown err -8

My versions:
libimobiledevice-1.2.0+66+g5a85432-2
libplist-2.0.0+11+gec9ba8b-2
libusbmuxd-1.0.10+13+gc724e70-1
usbmuxd-1.1.0+48+g1cc8b34-1

I have tried running things like `idevicepair pair && idevicepair unpair`, and restarting the daemon, restarting my computer

Here is output from `sudo usbmuxd -vv -f`
```
 [08:26:48.324][3] usbmuxd v1.1.1 starting up
[08:26:48.324][4] Creating socket
[08:26:48.324][5] initialized config_dir to /var/lib/lockdown
[08:26:48.324][5] client_init
[08:26:48.324][5] device_init
[08:26:48.324][4] Initializing USB
[08:26:48.324][5] usb_init for linux / libusb 1.0
[08:26:48.326][4] Registering for libusb hotplug events
[08:26:48.326][4] Found new device with v/p 05ac:12a8 at 1-6
[08:26:48.326][4] Found interface 1 with endpoints 04/85 for device 1-6
[08:26:48.326][4] Using wMaxPacketSize=512 for device 1-6
[08:26:48.326][4] USB Speed is 480 MBit/s for device 1-6
[08:26:48.326][4] 1 device detected
[08:26:48.326][3] Initialization complete
[08:26:48.326][4] Got lang ID 1033 for device 1-6
[08:26:48.327][4] Got serial 'XXXXXXXXXXXXXXXXXXXXXXXXXXXX' for device 1-6
[08:26:48.327][3] Connecting to new device on location 0x10006 as ID 1
[08:26:48.327][5] All 3 RX loops started successfully
[08:26:48.327][3] Connected to v2.0 device 1 on location 0x10006 with serial number XXXXXXXXXXXXXXXXXXXXXXXXXXXX
[08:26:48.327][4] preflight_worker_handle_device_add: Starting preflight on device XXXXXXXXXXXXXXXXXXXXXXXXXXXX...
[08:26:48.327][4] New client on fd 13 (self)
[08:26:48.327][5] Client command in fd 13 len 592 ver 1 msg 8 tag 1
[08:26:48.327][5] Client 13 connection request to device 0 port 62078
[08:26:48.327][2] Attempted to connect to nonexistent device 0
[08:26:48.327][5] send_pkt fd 13 tag 1 msg 8 payload_length 278
[08:26:48.327][4] Client 13 connection closed
[08:26:48.327][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device XXXXXXXXXXXXXXXXXXXXXXXXXXXX, lockdown error -8
[08:26:48.327][4] Disconnecting client fd 13
[08:26:50.767][4] New client on fd 12 (pid 411)
[08:26:50.767][5] Client command in fd 12 len 504 ver 1 msg 8 tag 99
[08:26:50.767][5] send_pkt fd 12 tag 99 msg 8 payload_length 230
[08:26:50.767][4] Client 12 connection closed
[08:26:50.767][4] Disconnecting client fd 12
[08:26:55.770][4] New client on fd 12 (pid 411)
[08:26:55.770][5] Client command in fd 12 len 504 ver 1 msg 8 tag 100
```

Downgrading back to 1.1.0+28+g46bdf3e-1 results in everything properly again

TheAifam5, 2018-11-11:
@candoizo the same issue I have.

mexmer, 2018-11-12:
@candoizo Youhave usbmuxd 1.1 but libusbmux 1.0.10, both must have be 1.1. Like i saidbefore, best bet is to compile all from github sources

TheAifam5, 2018-11-13:
@mexmer are you sure about that? Maybe Arch package maintainer needs to update the packages only to the latest version (based on git) ?

TheAifam5, 2018-11-13:
@candoizo I reported it here: https://bugs.archlinux.org/task/60789

mexmer, 2018-11-13:
yes, they need to update all their packages, not just usbmuxd.

candoizo, 2018-11-13:
Thanks for having a look @mexmer, I can surely see how 1.1.x vs. 1.0.x might have unexpected consequences, although @guywithface did imply he was having those problems still when he finished with
> Anything further I can do on my end to help debug this?

Maybe he got it working for him soon after, hoping with @TheAifam5 's report arch maintainers can sort this out as well

mexmer, 2018-11-13:
most common issue is, that people have partially installed maintainer packages and maintainer development packages, then try to compile one or more parts from source. in that case maintainer developer package gets included into build, causing incompatibilities.

that's why i'm saying you need either all from source and nothing from maintainer or all from maintainer and nothing from source. 

what people usually do when something from maintainer package doesn't work, they delete only that nonworking part and build from source ... yet another incompatibility introduced into system ....

also another thing is, that you need kill previous service (usbmuxd) when installing new one, you also need to run ldconfig, when installing new version of libraries. this is usually done by package manager, because there is install script included that does this. people forget to this, when they compile and install own version from source.

guywithface, 2019-01-10:
@mexmer @candoizo Just seeing this now. Just wanted to report that no I am not using the latest version. Still using the older version (tree commit d334020) and have compiled everything from scratch. In addition, on my machine I never installed usbmuxd through a package and only from source.

@mexmer I'll try the ldconfig and report back!

nikias, 2019-01-31:
Again I see device id 0 here.... Let's try to figure this out.
It starts in device.c in device_version_input():
```Connected to v2.0 device 1 ...```
https://github.com/libimobiledevice/usbmuxd/blob/master/src/device.c#L580-L589
```c
usbmuxd_log(LL_NOTICE, "Connected to v%d.%d device %d on location 0x%x with serial number %s", dev->version, vh->minor, dev->id, usb_get_location(dev->usbdev), usb_get_serial(dev->usbdev));
dev->state = MUXDEV_ACTIVE;
collection_init(&dev->connections);
struct device_info info;
info.id = dev->id;
info.location = usb_get_location(dev->usbdev);
info.serial = usb_get_serial(dev->usbdev);
info.pid = usb_get_pid(dev->usbdev);
info.speed = usb_get_speed(dev->usbdev);
preflight_worker_device_add(&info);
```
here we have the correct id. This function calls preflight_worker_device_add() with a struct that also has the device id in it.
In preflight_worker_device_add(), a copy of the structure is made:
https://github.com/libimobiledevice/usbmuxd/blob/master/src/preflight.c#L347-L349
```c
struct device_info *infocopy = (struct device_info*)malloc(sizeof(struct device_info));

memcpy(infocopy, info, sizeof(struct device_info));
```
Then, a thread is created that will call the thread handler preflight_worker_handle_device_add() with that copy.
Now in preflight_worker_handle_device_add() a idevice_private struct is created and filled with the device info, among that is the device id:
https://github.com/libimobiledevice/usbmuxd/blob/master/src/preflight.c#L125-L131
```c
struct device_info *info = (struct device_info*)userdata;
struct idevice_private *_dev = (struct idevice_private*)malloc(sizeof(struct idevice_private));
_dev->udid = strdup(info->serial);
_dev->mux_id = info->id;
_dev->conn_type = CONNECTION_USBMUXD;
_dev->conn_data = NULL;
_dev->version = 0;
```
This is then used to create a lockdown connection. It has a log message there:
https://github.com/libimobiledevice/usbmuxd/blob/master/src/preflight.c#L141
```c
usbmuxd_log(LL_INFO, "%s: Starting preflight on device %s...", __func__, _dev->udid);
```
Maybe you can add the dev->id there, like this:
```c
usbmuxd_log(LL_INFO, "%s: Starting preflight on device %d %s...", __func__, _dev->mux_id, _dev->udid);
```
This would allow us to see if, at this point, it still has the right device id. To look a bit further, the device id (mux_id) is passed to usbmuxd_connect() in libimobildevice's idevice_connect():
https://github.com/libimobiledevice/libimobiledevice/blob/master/src/idevice.c#L293
```c
int sfd = usbmuxd_connect(device->mux_id, port);
```
and it libusbmuxd.c it is finally sent to usbmuxd here:
https://github.com/libimobiledevice/libusbmuxd/blob/master/src/libusbmuxd.c#L731
```c
plist_t plist = create_plist_message("Connect");
plist_dict_set_item(plist, "DeviceID", plist_new_uint(device_id));
plist_dict_set_item(plist, "PortNumber", plist_new_uint(htons(port)));

res = send_plist_packet(sfd, tag, plist);
```
In usbmuxd, it will land in client.c and this is where the id will end up being 0:
https://github.com/libimobiledevice/usbmuxd/blob/master/src/client.c#L677-L687
```c
node = plist_dict_get_item(dict, "DeviceID");
if (!node) {
	usbmuxd_log(LL_ERROR, "Received connect request without device_id!");
	plist_free(dict);
	if (send_result(client, hdr->tag, RESULT_BADDEV) < 0)
		return -1;
	return 0;
}
val = 0;
plist_get_uint_val(node, &val);
device_id = (uint32_t)val;
```
Somewhere in between the id gets lost. Maybe you can also add a printf to libusbmuxd's send_plist_packet() here:
https://github.com/libimobiledevice/libusbmuxd/blob/master/src/libusbmuxd.c#L541
Just add a line after this:
```c
plist_to_xml(message, &payload, &payload_size);
printf("%s\n", payload);
```
Then report back

pauldotknopf, 2019-01-31:
Here is the new output:

```
[13:51:09.658][3] usbmuxd v1.1.1 starting up
[13:51:09.658][4] Creating socket
[13:51:09.658][5] initialized config_dir to /var/lib/lockdown
[13:51:09.658][5] client_init
[13:51:09.658][5] device_init
[13:51:09.658][4] Initializing USB
[13:51:09.658][5] usb_init for linux / libusb 1.0
[13:51:09.660][4] Registering for libusb hotplug events
[13:51:09.660][4] 0 devices detected
[13:51:09.660][3] Initialization complete
[13:51:09.660][4] New client on fd 11 (pid 2758)
[13:51:09.660][5] Client command in fd 11 len 499 ver 1 msg 8 tag 19
[13:51:09.660][5] send_pkt fd 11 tag 19 msg 8 payload_length 278
[13:51:09.660][5] Client 11 now LISTENING
[13:51:16.864][4] New client on fd 12 (pid 2973)
[13:51:16.864][5] Client command in fd 12 len 504 ver 1 msg 8 tag 1
[13:51:16.864][5] send_pkt fd 12 tag 1 msg 8 payload_length 230
[13:51:16.864][4] Client 12 connection closed
[13:51:16.864][4] Disconnecting client fd 12
[13:51:16.865][4] Found new device with v/p 05ac:12ab at 1-10
[13:51:16.865][3] Could not get old configuration descriptor for device 1-10: -5
[13:51:16.865][4] Setting configuration for device 1-10, from 0 to 4
[13:51:16.870][4] Found interface 1 with endpoints 04/85 for device 1-10
[13:51:16.870][4] Using wMaxPacketSize=512 for device 1-10
[13:51:16.870][4] USB Speed is 480 MBit/s for device 1-10
[13:51:16.870][4] Got lang ID 1033 for device 1-10
[13:51:16.870][4] Got serial '963052f0100dfacf16f610ee9a9b8409d8311fbc' for device 1-10
[13:51:16.870][3] Connecting to new device on location 0x1000a as ID 1
[13:51:16.870][5] All 3 RX loops started successfully
[13:51:16.870][3] Connected to v2.0 device 1 on location 0x1000a with serial number 963052f0100dfacf16f610ee9a9b8409d8311fbc
[13:51:16.870][4] preflight_worker_handle_device_add: Starting preflight on device 1 963052f0100dfacf16f610ee9a9b8409d8311fbc...
[13:51:16.870][4] New client on fd 14 (self)
[13:51:16.870][5] Client command in fd 14 len 592 ver 1 msg 8 tag 1
[13:51:16.871][5] Client 14 connection request to device 0 port 62078
[13:51:16.871][2] Attempted to connect to nonexistent device 0
[13:51:16.871][5] send_pkt fd 14 tag 1 msg 8 payload_length 278
[13:51:16.871][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device 963052f0100dfacf16f610ee9a9b8409d8311fbc, lockdown error -8
[13:51:16.871][4] Client 14 connection closed
[13:51:16.871][4] Disconnecting client fd 14
[13:51:22.423][4] New client on fd 13 (pid 1759)
[13:51:22.423][5] Client command in fd 13 len 504 ver 1 msg 8 tag 56
[13:51:22.424][5] send_pkt fd 13 tag 56 msg 8 payload_length 230
[13:51:22.424][4] Client 13 connection closed
[13:51:22.424][4] Disconnecting client fd 13
^C[13:51:22.617][3] Caught signal 2, exiting
[13:51:22.617][4] Event processing interrupted
[13:51:22.617][3] usbmuxd shutting down
[13:51:22.617][5] device_kill_connections
[13:51:22.717][5] usb_shutdown
[13:51:22.717][3] Removed device 1 on location 0x1000a
[13:51:22.717][5] client_device_remove: id 1
[13:51:22.717][5] send_pkt fd 11 tag 0 msg 8 payload_length 282
[13:51:22.717][5] usb_disconnect: cancelling RX xfer 0x5567daa3bee0
[13:51:22.717][5] usb_disconnect: cancelling RX xfer 0x5567daa3ffd0
[13:51:22.718][5] usb_disconnect: cancelling RX xfer 0x5567daa440c0
[13:51:22.718][5] Device 1-10 RX transfer cancelled
[13:51:22.718][5] Device 1-10 RX transfer cancelled
[13:51:22.718][5] Device 1-10 RX transfer cancelled
[13:51:22.718][5] device_shutdown
[13:51:22.718][5] client_shutdown
[13:51:22.718][4] Disconnecting client fd 11
[13:51:22.718][3] Shutdown complete
```

The "payload" line you had me enter didn't get called, it looks like.

pauldotknopf, 2019-01-31:
Email me at pauldotknopf at gmail if you want to do a quick TeamViewer session.

nikias, 2019-01-31:
Ok that probably means that usbmuxd is using a different version of libusbmuxd. Can you do this:
```
$ ldd `which usbmuxd` |grep libusbmuxd
	libusbmuxd.so.4 => /usr/local/lib/libusbmuxd.so.4 (0x00007f8aed1fa000)
```
And then check libusbmuxd:
```
ls -l /usr/local/lib/libusbmuxd.so.4
```
To see if it matches the time you installed your self-compiled version.

closed 2019-01-31 by nikias