libimobiledevice/usbmuxd #98: Does not compile with Ubuntu 12.04

kamstrup, 2018-10-10:
I realize that Ubuntu 12.04 is ancient at this point; but I may have a friend of friend using this ;-P In any case...

When compiling I get:

```
make[2]: Entering directory `/root/chef/usbmuxd/b564c15818a90e31e356756b2a25beb1/src'
  CC     usbmuxd-client.o
  CC     usbmuxd-device.o
  CC     usbmuxd-preflight.o
  CC     usbmuxd-log.o
  CC     usbmuxd-usb.o
  CC     usbmuxd-utils.o
utils.c:331:12: error: static declaration of ‘clock_gettime’ follows non-static declaration
/usr/include/time.h:336:12: note: previous declaration of ‘clock_gettime’ was here
utils.c: In function ‘clock_gettime’:
utils.c:337:2: error: unknown type name ‘mach_timebase_info_data_t’
utils.c:339:2: warning: implicit declaration of function ‘mach_absolute_time’ [-Wimplicit-function-declaration]
utils.c:341:15: error: request for member ‘denom’ in something not a structure or union
utils.c:342:3: warning: implicit declaration of function ‘mach_timebase_info’ [-Wimplicit-function-declaration]
utils.c:345:15: error: request for member ‘numer’ in something not a structure or union
utils.c:345:39: error: request for member ‘denom’ in something not a structure or union
utils.c:348:35: error: request for member ‘numer’ in something not a structure or union
utils.c:348:53: error: request for member ‘denom’ in something not a structure or union
make[2]: *** [usbmuxd-utils.o] Error 1
make[2]: Leaving directory `/root/chef/usbmuxd/b564c15818a90e31e356756b2a25beb1/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/root/chef/usbmuxd/b564c15818a90e31e356756b2a25beb1'
make: *** [all] Error 2
```

kamstrup, 2018-10-10:
This seems to imply that config.h does not set HAVE_CLOCK_GETTIME. Although that function should be supported according to http://manpages.ubuntu.com/manpages/precise/man2/clock_getres.2.html

nikias, 2018-11-28:
Fixed with commit 96e4aabe0b9a46ea9da4955a10c774a8e58fe677.

closed 2018-11-28 by nikias