libimobiledevice/usbmuxd #101: Crash in libplist

Blefish, 2018-10-26:
Hi

I am struggling on figuring out where this crash might've come from.
`Oct 26 11:20:58 xx kernel: [84969.462172] traps: usbmuxd[12739] general protection ip:7f13437121fb sp:7fff68640080 error:0 in libplist.so.3.1.0[7f134370a000+c000]`

I traced it down to libplist.so line 527 (optimized), it actually seems to be from where plist_get_node_type returns data->type.

```
Oct 26 11:20:58 xx systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.030][3] usbmuxd v1.1.1 starting up
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.031][3] Successfully dropped privileges to 'usbmux'
Oct 26 11:20:58 xx upowerd[1300]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.3
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.037][3] Could not get old configuration descriptor for device 1-45: -5
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.039][3] Initialization complete
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.039][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.039][3] Connecting to new device on location 0x1002d as ID 1
Oct 26 11:20:58 xx colord[933]: CdMain: failed to emit DeviceAdded: failed to register object: An object is already exported for the interface org.freedesktop.ColorManager.Device at /org/freedesktop/ColorManager/devices/sysfs__null_
Oct 26 11:20:58 xx colord[933]: message repeated 2 times: [ CdMain: failed to emit DeviceAdded: failed to register object: An object is already exported for the interface org.freedesktop.ColorManager.Device at /org/freedesktop/ColorManager/devices/sysfs__null_]
Oct 26 11:20:58 xx upowerd[1300]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:4.1
Oct 26 11:20:58 xx usbmuxd[12739]: [11:20:58.140][3] Connected to v2.0 device 1 on location 0x1002d with serial number 9668141b228182495b153d94d0d42d16b0ac85d8
Oct 26 11:20:58 xx kernel: [84969.462172] traps: usbmuxd[12739] general protection ip:7f13437121fb sp:7fff68640080 error:0 in libplist.so.3.1.0[7f134370a000+c000]
Oct 26 11:20:58 xx systemd[1]: usbmuxd.service: Main process exited, code=killed, status=11/SEGV
Oct 26 11:20:58 xx systemd[1]: usbmuxd.service: Failed with result 'signal'.

00000000000081e0 <plist_dict_get_item>:
plist_dict_get_item():
libimobiledevice/libplist/src/plist.c:524
    81e0:       55                      push   %rbp
    81e1:       41 57                   push   %r15
    81e3:       41 56                   push   %r14
    81e5:       53                      push   %rbx
    81e6:       48 83 ec 18             sub    $0x18,%rsp
    81ea:       49 89 f6                mov    %rsi,%r14
libimobiledevice/libplist/src/plist.c:527
    81ed:       48 85 ff                test   %rdi,%rdi
    81f0:       74 38                   je     822a <plist_dict_get_item+0x4a>
plist_get_data():
libimobiledevice/libplist/src/plist.c:147
    81f2:       48 8b 47 20             mov    0x20(%rdi),%rax
plist_get_node_type():
libimobiledevice/libplist/src/plist.c:731
    81f6:       48 85 c0                test   %rax,%rax
    81f9:       74 2f                   je     822a <plist_dict_get_item+0x4a>
plist_dict_get_item():
libimobiledevice/libplist/src/plist.c:527
    81fb:       83 78 10 05             cmpl   $0x5,0x10(%rax)
    81ff:       75 29                   jne    822a <plist_dict_get_item+0x4a>
libimobiledevice/libplist/src/plist.c:530
    8201:       48 8b 18                mov    (%rax),%rbx
libimobiledevice/libplist/src/plist.c:531
    8204:       48 85 db                test   %rbx,%rbx
    8207:       74 32                   je     823b <plist_dict_get_item+0x5b>
libimobiledevice/libplist/src/plist.c:533
    8209:       4c 89 34 24             mov    %r14,(%rsp)

```

Built and running on Ubuntu 18.04. This is a VERY rare case but it causes usbmuxd  to shut down until a new device is plugged in.

Any ideas?

nikias, 2018-11-28:
I think this is what was the UaF here: #104, fixed with commit 9a93f8727414c33f9efa5448b468533a347cf519.

Blefish, 2018-11-29:
Great. Thanks!

closed 2018-11-29 by Blefish