libimobiledevice/usbmuxd #102: usbmuxd_get_device_by_udid function does not return CONNECTION_TYPE_NETWORK devices

t-yoshii, 2018-10-31:
Hi,
usbmuxd_get_device_by_udid function in libusbmuxd.c  in current master (9db5747c)  does not find devices with conn_type CONNECTION_TYPE_NETWORK. 

I think connection type check is not needed here.
https://github.com/libimobiledevice/libusbmuxd/blob/master/src/libusbmuxd.c#L1286

Older version of usbmuxd returns NETWORK devices.

usbmuxd comimt: 9db5747cd823b1f59794f81560a4af22a031f5c9

nikias, 2018-10-31:
This is done on purpose. For network devices, use the new API `usbmuxd_get_device` with the appropriate options.

nikias, 2018-10-31:
Yes I know. I am working on it.

dsrink, 2021-05-18:
Hi,

I am having the same problem, both "idevicesyslog -u DEVICE_ID" and "ideviceinfo -u DEVICE_ID" cannot find the device.

Is this issue still current? I would like to debug this in order to use backups over wifi. Please let me know where to touch/play.