libimobiledevice/usbmuxd #103: usbmuxd segfault on pair

pauldotknopf, 2018-11-14:
I'm using usbmuxd master branch ([this](https://github.com/libimobiledevice/usbmuxd/commit/1cc8b343a4bc9320758c78e187b13bb3c9029f12) commit).

I'm using libimobiledevice master branch ([this](https://github.com/libimobiledevice/libimobiledevice/commit/92c5462adef87b1e577b8557b6b9c64d5a089544) commit)

I can pair succesfully...

```bash
> idevicepair pair -u 963052f0100dfacf16f610ee9a9b8409d8311fbc
SUCCESS: Paired with device 963052f0100dfacf16f610ee9a9b8409d8311fbc
```

...but immediately after pairing, usbmuxd gets segfault.

```
Thread 1 "usbmuxd" received signal SIGSEGV, Segmentation fault.
plist_dict_get_item (node=0x6285d0, key=key@entry=0x40c239 "DeviceID") at plist.c:527
527         if (node && PLIST_DICT == plist_get_node_type(node))
(gdb) backtrace
#0  plist_dict_get_item (node=0x6285d0, key=key@entry=0x40c239 "DeviceID") at plist.c:527
#1  0x000000000040602d in client_command (hdr=0x671430, client=0x6155d0) at client.c:760
#2  process_recv (client=0x6155d0) at client.c:928
#3  client_process (fd=<optimized out>, events=<optimized out>) at client.c:954
#4  0x000000000040433b in main_loop (listenfd=4) at main.c:245
#5  main (argc=<optimized out>, argv=<optimized out>) at main.c:696
```

Could the problem be [this](https://github.com/libimobiledevice/usbmuxd/commit/d33402036d563667a76910dbfec50a37b622d3a5) commit?

mexmer, 2018-11-15:
did you also update libusbmuxd?

pauldotknopf, 2018-11-15:
Yes, I was running the latest tip of everything (plist included).

I'm mobile right now, but it was the latest commit as of today.

stychos, 2018-11-18:
Stumbled with the pairing on Ubuntu 18.10. Tried to build fresh binaries from the repos heads and got troubles with the `trying to connect to nonexistent device 0` error. Tried to walk trough sources a little and at some point thought that there really troubles with the libplist (all calling functions passing `dev_id = 1` to `device_start_connect()`, but I've seend `DeviceID = 0` in `client_command()` payloads (hacked it a little to dump plists to the screen). Then I replaced libplist3 with the distro-provided and now everything is ok.
Can someone provide me short guide or link on how to debug projects like this? I've tried to attach qtcreator's debugger but with no great luck :-) - at some point I've been able to stop flow and start stepping but I don't know how to make breakpoints in the attached process in needed place (i.e. at the start of the function that cause that errlog message). I'm not very good programmer, especially in C, and never debugged anything more than simple php scripts.. but I wish to learn and feel that this libimobiledevice code is very cool and clean.
Can someone point me?

mexmer, 2018-11-18:
@pauldotknopf if it crashes at plist functions, you might have compiled against different version of plist, than you have linked.

when compiling from source, and you are not sure, if there are still present distribution headers, it's best to manually add include path to source headers and also path to linked libraries.

i had similar issue in past with plist, because some other program required it, so i didn't uninstall distribution version, and my binaries linked against distribution version, but headers were used from source.

nikias, 2018-11-19:
@pauldotknopf certainly it is not a bug in libplist if you have device id 0 vs. device id 1, but what @mexmer says. The real question is why do you have device id 0? This should not happen since the first device id used will always be 1.