libimobiledevice/usbmuxd #106: (Question) Errors when setting usbmuxd as a service with systemd, but executes normally via terminal.

autor3iv, 2018-12-03:
I'm trying to set up usbmuxd to run as a service, however I keep running into the following error :

```
    Dec  3 10:29:19 asci-kubu-NUC7i3BNH systemd[1]: asci-usbmuxd.service: Main process exited, code=exited, status=1/FAILURE
Dec  3 10:29:19 asci-kubu-NUC7i3BNH systemd[1]: asci-usbmuxd.service: Failed with result 'exit-code'.
Dec  3 10:29:24 asci-kubu-NUC7i3BNH systemd[1]: asci-usbmuxd.service: Service hold-off time over, scheduling restart.
Dec  3 10:29:24 asci-kubu-NUC7i3BNH systemd[1]: asci-usbmuxd.service: Scheduled restart job, restart counter is at 46078.
Dec  3 10:29:24 asci-kubu-NUC7i3BNH systemd[1]: Stopped AssetScience usbmuxd service.
Dec  3 10:29:24 asci-kubu-NUC7i3BNH systemd[1]: Started AssetScience usbmuxd service.
Dec  3 10:29:24 asci-kubu-NUC7i3BNH usbmuxd[29950]: [0] Could not open lockfile

```

When I run it manually like so :

`sudo usbmuxd -v --user usbmux --foreground`

Everything is fine.

Here is my systemd .service file. I've read through the manual however I'm positive that I may have not set the correct capacities. I've verified the permissions/groups for usbmux user, usbmux is part of plugdev and sudo.

```
[Unit]
Description=usbmuxd service

[Service]
Type=simple
ExecStart=/usr/local/sbin/usbmuxd --user usbmux
Restart=always
CapabilityBoundingSet=CAP_SETGID CAP_SETUID CAP_CHOWN CAP_FOWNER CAP_NET_ADMIN CAP_NET_BIND_SERVICE
LimitNPROC=1
RestartSec=5
User=usbmux
Group=sudo

# Output to syslog
StandardOutput=syslog
StandardError=syslog
#Change this to find app logs in /var/log/syslog
SyslogIdentifier=usbmuxd

[Install]
WantedBy=multi-user.target
```
Any help would be appreciated, I've dug around as best I could trying to find a solution. Thanks!

Lessica, 2020-01-06:
User `usbmux` cannot access the lockfile, remove these lines:
```
User=usbmux
Group=sudo
```

bagbyte, 2020-05-21:
This [file](https://github.com/libimobiledevice/usbmuxd/blob/master/systemd/usbmuxd.service.in) is used as a template to create the `usbmuxd.service`. After you build, you should have the final `usbmuxd.service` file that you need.

You can copy the generated file to `/lib/systemd/system/usbmuxd.service` and restart the service:
```bash
sudo systemctl restart usbmuxd
```

In your version probably the `--systemd` flag was missing.