libimobiledevice/usbmuxd #108: Unable to install it on a MacOS

gabriellira, 2019-03-04:
I'm trying to install the _libimobiledevice_ , but I'm having some problems:

Every time I issue the command` brew install --HEAD usbmuxd`, I get the following error:

```
==> ./autogen.sh
==> ./configure --disable-silent-rules --prefix=/usr/local/Cellar/usbmuxd/HEAD-c75605d_1
==> make install
Last 15 lines from /Users/gabriel.lira/Library/Logs/Homebrew/usbmuxd/03.make:
In file included from /usr/local/include/string.h:25:
/usr/local/include/plist/Node.h:26:10: fatal error: 'cstddef' file not found
#include <cstddef>
         ^~~~~~~~~
In file included from socket.c:28:
In file included from /usr/local/include/string.h:25:
/usr/local/include/plist/Node.h:26:10: fatal error: 'cstddef' file not found
#include <cstddef>
         ^~~~~~~~~
1 error generated.
make[1]: *** [collection.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
1 error generated.
make[1]: *** [socket.lo] Error 1
make: *** [install-recursive] Error 1
```
I've cloned this repo and tried to install it from source, but when I issue the command `./autogen.sh` I get the error bellow:
```
...
checking for libplist >= 1.11... yes
checking for libimobiledevice-1.0 >= 1.2.1... no
checking for pthread_create, pthread_mutex_lock in -lpthread... yes
configure: error: preflight worker support requested but libimobiledevice could not be found
```
I've tried some solutions like using `brew uninstall --force usbmuxd`, `brew cleanup -s usbmuxd`, but none of that fixed the problem.

I'm using a MacOS 10.14.2

liaojinxing, 2019-03-05:
Similar issue here.
`liaojinxingdeMacBook-Pro :: ~/code/flutter » brew install --HEAD usbmuxd
==> Cloning https://git.sukimashita.com/libusbmuxd.git
Updating /Users/jinxing/Library/Caches/Homebrew/usbmuxd--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at c75605d Return meaningful error codes from usbmuxd_connect()
==> ./autogen.sh
==> ./configure --disable-silent-rules --prefix=/usr/local/Cellar/usbmuxd/HEAD-c75605d_1
==> make install
Last 15 lines from /Users/jinxing/Library/Logs/Homebrew/usbmuxd/03.make:
        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:213:2: error: unknown type name 'uint64_t'
        uint64_t ri_pkg_idle_wkups;
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:214:2: error: unknown type name 'uint64_t'
        uint64_t ri_interrupt_wkups;
        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [collection.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
20 errors generated.
make[1]: *** [socket.lo] Error 1
make: *** [install-recursive] Error 1`

craig006, 2019-04-20:
Did you find a solution to this @gabriellira ? I have it now as well

nikias, 2019-04-20:
You do realize that on macOS you don't have to install usbmuxd? MacOS has its own daemon that will work, and you would have to disable the original so you would be able to use the open source one.

liaojinxing, 2019-04-22:
Met this issue.
What I tried is reinstalling brew and then `brew install usbmuxd`, this issue gone.

gabriellira, 2019-04-22:
Hey guys, I've been using VS Code simulator launcher(maybe it's using the original `usbmuxd` as @nikias mentioned). I'll try reinstalling brew, again, to see if that issue is gone.

EinatK, 2019-07-16:
I have this issue too.. is there a solution to it? 
I've tried reinstalling brew and `brew install usbmuxd` -> it works without the HEAD flag but doesn't install what i need (for flutter), i do need the HEAD flag..
Anyone have any suggestions? I've tried anything and everything i could find, nothing helped :(

craig006, 2019-07-16:
@EinatK I ended up having to completely nuke brew and everything it had installed and start over. I think `brew doctor` or something like that pointed out that I had some serious issues with my brew setup. Sorry I cant think of the specifics, it was quite a while ago.

EinatK, 2019-07-16:
@craig006 do you remember how you went about it? Is there a command or series of commands to do it? Over time i've installed many things using brew, i'm not sure how to find them all.. I uninstall it using `ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/uninstall)"`

badhabitustc, 2019-09-24:
Wrong header file was placed at /usr/local/include .
run **_brew doctor_** and delete Unexpected header files

for this case just remove
/usr/local/include/string.h

It is a wrong header file

nikias, 2019-09-24:
@badhabitustc sorry but `string.h` is not coming from usbmuxd.

badhabitustc, 2019-09-24:
> @badhabitustc sorry but `string.h` is not coming from usbmuxd.

yes but usbmuxd included the wrong string.h

badhabitustc, 2019-09-24:
> @badhabitustc sorry but `string.h` is not coming from usbmuxd.

**brew doctor** ouput:
`
**Warning: Unbrewed header files were found in /usr/local/include.**
**If you didn't put them there on purpose they could cause problems when**
**building Homebrew formulae, and may need to be deleted.**

Unexpected header files:
 ...
  /usr/local/include/qsv.h
  /usr/local/include/random_seed.h
  /usr/local/include/rational.h
  /usr/local/include/rc4.h
  /usr/local/include/replaygain.h
  /usr/local/include/restore.h
  /usr/local/include/ripemd.h
  /usr/local/include/rtmp.h
  /usr/local/include/samplefmt.h
  /usr/local/include/sbservices.h
  /usr/local/include/screenshotr.h
  /usr/local/include/service.h
  /usr/local/include/sha.h
  /usr/local/include/sha512.h
  /usr/local/include/spherical.h
  /usr/local/include/stereo3d.h
  /usr/local/include/string.h <--------------------------------wrong string.h
  /usr/local/include/swresample.h
  /usr/local/include/swscale.h
  /usr/local/include/syslog_relay.h
  /usr/local/include/tea.h
  /usr/local/include/threadmessage.h
  /usr/local/include/time.h
  /usr/local/include/timecode.h
  /usr/local/include/timestamp.h
  /usr/local/include/tree.h
  /usr/local/include/twofish.h
  /usr/local/include/vaapi.h
  /usr/local/include/vda.h
  /usr/local/include/vdpau.h
  /usr/local/include/version.h
  /usr/local/include/videotoolbox.h
  /usr/local/include/vorbis_parser.h
  /usr/local/include/webinspector.h
  /usr/local/include/xtea.h
  /usr/local/include/xvmc.h
`
and gabriellira`s output indicated the error caused by **/usr/local/include/string.h**

nikias, 2019-09-24:
This is not a problem with usbmuxd. It's about your system being in an inconsistent state for brew.

badhabitustc, 2019-09-24:
> This is not a problem with usbmuxd. It's about your system being in an inconsistent state for brew.

Yes.  Not a problem of usbmuxd.
This page has a high rank when people search `'cstddef' file not found` so I post my solution

nanoscopic, 2026-03-05:
Uh... to my understanding modern macs lock down the USB driver and interactions so that you can't talk directly to the devices at the level needed for usbmuxd to work on MacOS ( this open source version )

Am I misunderstanding something?