libimobiledevice/usbmuxd #110: How to make usbmuxd hand off the device?

ashegoulding, 2019-04-27:
usbmuxd always hog on devices, restricting other process from accessing them. I ended up masking `usbmuxd.service` since Gnome Desktop starts the service every time a device is connected. This solved the issue for me, but how about the other users who want to use the util programs? Is there a way to command usbmuxd to hand off the device, or is it the way GDE is programmed?

I just needed to delete synced photos from my phone to get some space. I didn't know how to do it on Linux, so I tried to attach my device to my Windows QEMU vm. When there's usbmuxd processes running, I couldn't attach my phone to the vm.

Note: I always set `/org/gnome/desktop/media-handling/automount` to false so GDE doesn't mount devices until I click on them(this is a bad design, in my opinion). I tried `idevicepair unpair` command, but the process was still holding the device:

```
usbmuxd   5764                usbmuxd   14u      CHR             189,29         0t0     255291 /dev/bus/usb/001/030
usbmuxd   5764 5767 usbmuxd   usbmuxd   14u      CHR             189,29         0t0     255291 /dev/bus/usb/001/030
usbmuxd   5764 5818 usbmuxd   usbmuxd   14u      CHR             189,29         0t0     255291 /dev/bus/usb/001/030
usbmuxd   5764 5823 usbmuxd   usbmuxd   14u      CHR             189,29         0t0     255291 /dev/bus/usb/001/030
```

davimello28, 2020-08-02:
I know it has been an year since your issue is open, but since I'm on my third consecutive day searching for a solution I am creating this as a documentation for future requests. Here is a good list of that I've attempted to do:

- Prevent system to takeover the device, so I can use QEMU `usb-host` driver. Create a `usbmuxd` via an empty rule in `/etc/udev/rules.d/` (from [here](https://www.zeitgeist.se/2015/06/28/mount-an-iphone-inside-a-kvm-guest-by-disabling-usbmuxd/))

```
sudo echo "" >> /etc/udev/rules.d/39-usbmuxd.rules
```
Than refresh the rules:
```
 udevadm control -R
```

- Stop the service. It is possible that the `udev` rules are blocked, but somehow the service is kept in the background. Check it status:
```
sudo service usbmuxd status | grep "Active:" 
```
If you see `Active: inactive (dead) since ...`, the service is stopped.
- If somehow it is still running, you can `usbmuxd --exit` (from [here](https://manpages.debian.org/testing/usbmuxd/usbmuxd.1)). Also try the `--disable-hotplug` option. 

- Keep in mind that the usb 'takeover' can be happening by processes that uses root level. It would be a good idea to follow [this tutorial](https://www.linuxquestions.org/questions/blog/kingbeowulf-74138/qemu-usb-passthrough-with-specifics-for-iphone-37830/) to change the user level of the attached USB.

- QEMU 5.0.0 has an option to prevent the device from being reset by adding `quest-reset=false` to it. This can help if you get multiple plug/unplug events on your system when running a virtual machine. 

- Another possibility is to check `ipheth` driver. I did not test that yet, but `dmesg` was sending multiple requests for the driver when I was attempting to connect to QEMU. 

<details><summary>sudo dmesg | grep ipheth </summary>
<p>
<pre>[14740.983603] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14742.139124] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14742.142890] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14744.955519] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14842.111718] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14842.117170] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14881.753232] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14911.215486] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14911.250913] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14911.665422] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14912.443448] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14912.459920] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14914.141478] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14915.288593] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14915.292519] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14920.065474] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14920.854684] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14920.859951] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14922.557477] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14923.743449] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14923.749259] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[14928.473564] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[14929.285274] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[14929.289232] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[15002.556912] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[15318.225081] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[15318.229198] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[15319.101049] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[15320.241224] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[15320.244828] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[15324.984847] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[15325.782953] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[15325.788740] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[15327.484732] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[15328.634639] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[15328.639731] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
[15333.388653] ipheth 1-4:4.2: Apple iPhone USB Ethernet now disconnected
[15334.195037] ipheth 1-4:4.2: Apple iPhone USB Ethernet device attached
[15334.199069] ipheth 1-4:4.2 enx4abf6b73c4b0: renamed from eth1
</pre>
</p>
</details>

ashegoulding, 2020-08-02:
Here's a better solution: boycott Apple products!

davimello28, 2020-08-04:
In every situation that I stuck on this problem, it was `gvfs` that was holding the usb, not `usbmuxd`.

Here is the result of `lsof | grep "/dev/bus/usb/001/006"` before running qemu:

```bash
gvfsd-gph 59313                             davi   15u      CHR              189,5       0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59314 gmain                 davi   15u      CHR              189,5       0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59315 gdbus                 davi   15u      CHR              189,5       0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59318 dconf\x20             davi   15u      CHR              189,5       0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59322 pool                  davi   15u      CHR              189,5       0t0     207112 /dev/bus/usb/001/006
```
and this is the one after running qemu:

```
gvfsd-gph 59313                             davi   15u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59314 gmain                 davi   15u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59315 gdbus                 davi   15u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59318 dconf\x20             davi   15u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
gvfsd-gph 59313 59322 pool                  davi   15u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078                             davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078 61087 qemu-syst             davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078 61088 gmain                 davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078 61089 gdbus                 davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078 61093 qemu-syst             davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078 61095 threaded-             davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
qemu-syst 61078 61096 qemu-syst             davi   38u      CHR              189,5         0t0     207112 /dev/bus/usb/001/006
```

It seems to me that my situation was caused by `gvfs-gphoto`, not `usbmuxd`. The blacklist that I've pointed earlier worked as expected.

ronylee11, 2022-07-21:
I force uninstalled usbmuxd using sudo pacman -Rdd usbmuxd, its working for me!

AbdullahPrime, 2024-11-26:
> Here's a better solution: boycott Apple products!

No one likes apple but as devs we've got publish for iOS too to put food on the table

dxdxdt, 2024-11-26:
> > Here's a better solution: boycott Apple products!
> 
> No one likes apple but as devs we've got publish for iOS too to put food on the table

True.

It's almost 2025 and Apple is still refusing to add MTP to their products. It's their same tactic like Lightening port and USB-C genders. You do you, but know that we can't keep let them get away with their greedy business practices. I just feel sorry for devs who have to make this kind of libraries to forcefully bring their products to other platforms. It's not fun.