libimobiledevice/usbmuxd #114: Usbmuxd hangs when disconnecting device

Blefish, 2019-07-17:
On some usecases the usbmuxd service hangs when disconnecting the device. I am able to reproduce it quite easily in Virtualbox, but it has very rarely happened on real PC as well. When this happens, new devices cannot be detected.

It seems to me like the RX transfer never gets cancelled and the while loop is running forever: https://github.com/libimobiledevice/usbmuxd/blob/master/src/usb.c#L93

When I run the service manually from command line (so it does not exit via udev rules), it works just fine as I was unable to reproduce the problem.

What I find weird is that in the log, it shows that it cannot find device entry. Maybe it's removing it twice?

Running on Ubuntu 18.04. libusb is 1.0.21-2

```
juuli 17 11:37:51 nest9001 usbmuxd[16371]: [11:37:51.926][5] connection_teardown dev 1 sport 6 dport 62078
juuli 17 11:37:51 nest9001 usbmuxd[16371]: [11:37:51.926][5] [OUT] dev=1 sport=6 dport=62078 seq=3113 ack=8787 flags=0x4 window=131072[512] len=0
juuli 17 11:37:51 nest9001 usbmuxd[16371]: [11:37:51.926][4] Disconnecting client fd 14
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][3] Removed device 1 on location 0x1001c
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][5] client_device_remove: id 1
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][5] send_pkt fd 12 tag 0 msg 8 payload_length 282
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][2] Cannot find device entry while removing USB device 0x556a69de33b0 on location 0x1001c
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][5] usb_disconnect: cancelling RX xfer 0x556a69de4b70
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][5] usb_disconnect: cancelling RX xfer 0x556a69de4c50
juuli 17 11:37:52 nest9001 usbmuxd[16371]: [11:37:52.855][5] usb_disconnect: cancelling RX xfer 0x556a69de5ac0
```

nikias, 2021-01-09:
I think this was fixed with 63d1164e736d7419198d1b737d10a9aae85bef98.

Blefish, 2021-01-15:
Cool, thanks. I will close the ticket, I am no longer actively participating :)

closed 2021-01-15 by Blefish