libimobiledevice/usbmuxd #121: How to build Android version?

weng1250, 2019-11-11:
Could anyone figure out how to run usbmuxd on Android os?  I am trying to connect iOS device with Android device by usb and then communicate with each other. Thanks

seanhead, 2020-07-02:
I've been looking into this as well

XanaduAPI, 2020-12-05:
首先，我想我们需要解决 Android 设备里面 /dev/bus/usb 目录的权限问题

[image]

usbmuxd 需要依赖 静态编译的 其它库。然后 修改以下代码
main.c

[image]

[image]

[image]

再修改 libusbmuxd 里面的 相应连接
libusbmuxd.c 
[image]

seanhead, 2020-12-07:
I ended up skipping this path, since I didn't actually need to talk to "ios", just my app.  If you emulate a ethernet device on the USB port you end up with a full TCP/IP stack to do what ever you want with.

XanaduAPI, 2020-12-08:
@seanhead  
So isn't your goal to connect IOS devices ? 
Can you tell me how you simulated it ?

cczhr, 2021-02-04:
@XanaduAPI 
我编译好后实际运行会有错误提示 有什么方法解决么 我发现这两个路径都是无法访问
1|rk3288:/ # usbmuxd -v -f
[17:25:19.379][3] usbmuxd v1.1.2 starting up
[17:25:19.380][4] Creating socket
[17:25:19.380][4] socket port:27015
[17:25:19.380][4] Listening on /var/run/usbmuxd
[17:25:19.381][0] Failed to create required directory '/var/lib/lockdown': No such file or directory

XanaduAPI, 2021-02-04:
> @XanaduAPI
> 我编译好后实际运行会有错误提示 有什么方法解决么 我发现这两个路径都是无法访问
> 1|rk3288:/ # usbmuxd -v -f
> [17:25:19.379][3] usbmuxd v1.1.2 starting up
> [17:25:19.380][4] Creating socket
> [17:25:19.380][4] socket port:27015 [...]

你解决了这两个路径也是没用的。只有ROOT过才可以的。

cczhr, 2021-02-04:
@XanaduAPI 设备Root 过了也是不行

cczhr, 2021-02-04:
@XanaduAPI  已经ROOT的设备要怎么执行？

XanaduAPI, 2021-02-04:
> @XanaduAPI 已经ROOT的设备要怎么执行？

把 usbmuxd 的文件通信改为 socket 。在Linux中，usbmuxd 是通过文件和libusbmuxd 通信的。
你修改为 AF_INET 看看。
[image]

XanaduAPI, 2021-02-04:
> @XanaduAPI 已经ROOT的设备要怎么执行？

通过 127.0.0.1 进行通信。这样就可以避开 /var/run/usbmuxd 的权限问题。
或者，你可以把那些文件夹的权限都改掉。放假咯。

cczhr, 2021-02-04:
@XanaduAPI  我 main.c就是是这样改的，我都是用re管理器把这些目录创建和权限改成777也不行，通过 127.0.0.1 进行通信 是怎么操作？ 最近这一两天才开始研究 ，不太熟悉。是啊放假啦

XanaduAPI, 2021-02-04:
> @XanaduAPI 我 main.c就是是这样改的，我都是用re管理器把这些目录创建和权限改成777也不行，通过 127.0.0.1 进行通信 是怎么操作？ 最近这一两天才开始研究 ，不太熟悉。是啊放假啦

就是说，本来 usbmuxd 和 libusbmuxd 是通过 /var/run/usbmuxd 来通信的。现在改为 通过本地网络来通信。这样就可以避开读写 /var/run/usbmuxd 文件的问题。libusbmuxd 也要修改

cczhr, 2021-02-04:
@XanaduAPI 好的 谢谢 我晚上再看看

nikias, 2021-02-04:
Please try to post in English, even if it was just ran through a translator. Anyway, you don't need to change the source code. You can now pass `--socket 127.0.0.1:12345` to usbmuxd to specify a port to listen on instead of the default unix socket, and also `--pidfile NONE` to disable the writing (and locking) of `/var/run/usbmuxd.pid`. For anything using libusbmuxd, you can then set an environment variable to change the socket path, e.g. ```USBMUXD_SOCKET_ADDRESS=127.0.0.1:12345 ideviceinfo```. Hope that helps.

cczhr, 2021-02-04:
@nikias  Thank you for your help, I want to support libimobiledevice,must create a var/lib/lockdown path,creating this path requires the Android device to be rooted,can I easy customize this path?

cczhr, 2021-02-04:
@XanaduAPI 感谢 通过你的方法确实可以连接上了

cczhr, 2021-02-05:
@XanaduAPI  我在安卓上测试,发现有个问题,苹果的设备的线拔了能检测到断开，但是再插进安卓设备的话无任何反应，必须要重新启动usbmuxd ，这种情况你有遇到么？

cczhr, 2021-02-07:
On Android, I can change the var/lib/lockdown  to data/local/tmp  by modifying the base_config_dir variables of the [usbmuxd--src/conf.c]  and [libimobiledevice---userpref.c]  files  to avoid the need to give root permissions for each initialization

daomeixiong2547, 2021-03-31:
> 首先，我想我们需要解决 Android 设备里面 /dev/bus/usb 目录的权限问题
> 
> [image]
> 
> usbmuxd 需要依赖 静态编译的 其它库。然后 修改以下代码
> main.c
> 
> [image]
> 
> [image]
> 
> [image]
> 
> 再修改 libusbmuxd 里面的 相应连接
> libusbmuxd.c
> [image]

请问还需要修改其他地方吗？我按照修改后日志，我也该成通过127.0.0.1连接，但还是走了这个位置
Listening on /data/local/tmp/usbmuxd
Failed to create required directory '/var/lib/lockdown': No such file or directory

XanaduAPI, 2021-04-01:
@daomeixiong2547   你需要 搜索 /var/lib/lockdown ，然后替换为一个可读写的目录，这个位置是用来存储 iPhone 信任文件的。

daomeixiong2547, 2021-04-01:
> @daomeixiong2547 你需要 搜索 /var/lib/lockdown ，然后替换为一个可读写的目录，这个位置是用来存储 iPhone 信任文件的。

 我也搞好了 非常感谢！！！

XanaduAPI, 2021-04-01:
> > @daomeixiong2547 你需要 搜索 /var/lib/lockdown ，然后替换为一个可读写的目录，这个位置是用来存储 iPhone 信任文件的。
> 
> 我也搞好了 非常感谢！！！

你可以用root后的Android来连接手机了吗？

XanaduAPI, 2021-04-01:
> @XanaduAPI 感谢 通过你的方法确实可以连接上了

你用Root过的Android设备，可以通过  usbmuxd 和  libimobiledevice 来连接信任手机吗？

daomeixiong2547, 2021-04-13:
> > > @daomeixiong2547 你需要 搜索 /var/lib/lockdown ，然后替换为一个可读写的目录，这个位置是用来存储 iPhone 信任文件的。
> > 
> > 
> > 我也搞好了 非常感谢！！！
> 
> 你可以用root后的Android来连接手机了吗？

我手上没有root设备，但我用Genymoton和Virtual box模拟root设备连接usb的场景，替换了一下模拟器的cpu架构。
使用你描述的修改方法，打出来的包就可以可以通过 usbmuxd 和 libimobiledevice来信任手机并读取手机信息了。

XanaduAPI, 2021-04-19:
> > > > @daomeixiong2547 你需要 搜索 /var/lib/lockdown ，然后替换为一个可读写的目录，这个位置是用来存储 iPhone 信任文件的。
> > > 
> > > 
> > > 我也搞好了 非常感谢！！！
> > 
> > 
> > 你可以用root后的Android来连接手机了吗？
> 
> 我手上没有root设备，但我用Genymoton和Virtual box模拟root设备连接usb的场景，替换了一下模拟器的cpu架构。
> 使用你描述的修改方法，打出来的包就可以可以通过 usbmuxd 和 libimobiledevice来信任手机并读取手机信息了。

模拟器和真机还是不一样的，我用Root过后的真机试了是不行，通信不了。

daomeixiong2547, 2021-05-10:
> > > > > @daomeixiong2547 你需要 搜索 /var/lib/lockdown ，然后替换为一个可读写的目录，这个位置是用来存储 iPhone 信任文件的。
> > > > 
> > > > 
> > > > 我也搞好了 非常感谢！！！
> > > 
> > > 
> > > 你可以用root后的Android来连接手机了吗？
> > 
> > 
> > 我手上没有root设备，但我用Genymoton和Virtual box模拟root设备连接usb的场景，替换了一下模拟器的cpu架构。
> > 使用你描述的修改方法，打出来的包就可以可以通过 usbmuxd 和 libimobiledevice来信任手机并读取手机信息了。
> 
> 模拟器和真机还是不一样的，我用Root过后的真机试了是不行，通信不了。

加我微信daomeixiong2547，我发给你试试

XanaduAPI, 2021-05-10:
> 加我微信daomeixiong2547，我发给你试试

加了

aceai84, 2021-08-12:
> Please try to post in English, even if it was just ran through a translator. Anyway, you don't need to change the source code. You can now pass `--socket 127.0.0.1:12345` to usbmuxd to specify a [...]

@nikias Is there a way to make usbmuxd to run on a android device without rooted？thanks