libimobiledevice/usbmuxd #122: preflight.c:47:2: error: redeclaration of enumerator ‘CONNECTION_USBMUXD’

borisovg, 2019-11-11:
Getting this error trying to build latest master on Debian:

```make  all-recursive
make[1]: Entering directory '/home/borisov/src/usbmuxd'
Making all in src
make[2]: Entering directory '/home/borisov/src/usbmuxd/src'
  CC       usbmuxd-client.o
  CC       usbmuxd-device.o
  CC       usbmuxd-preflight.o
preflight.c:47:2: error: redeclaration of enumerator ‘CONNECTION_USBMUXD’
   47 |  CONNECTION_USBMUXD = 1
      |  ^~~~~~~~~~~~~~~~~~
In file included from preflight.c:34:
/usr/local/include/libimobiledevice/libimobiledevice.h:64:2: note: previous definition of ‘CONNECTION_USBMUXD’ was here
   64 |  CONNECTION_USBMUXD = 1,
      |  ^~~~~~~~~~~~~~~~~~
make[2]: *** [Makefile:497: usbmuxd-preflight.o] Error 1
make[2]: Leaving directory '/home/borisov/src/usbmuxd/src'
make[1]: *** [Makefile:423: all-recursive] Error 1
make[1]: Leaving directory '/home/borisov/src/usbmuxd'
make: *** [Makefile:355: all] Error 2```

nikias, 2019-11-11:
Hi, please pull latest and try again. I fixed this with commit 9af2b12552693a47601347e1eafc1e94132d727e.

borisovg, 2019-11-11:
Yes that fixed it - thank you for quick response.

closed 2019-11-11 by borisovg