libimobiledevice/usbmuxd #126: How can I make usbmuxd work on WSL?

OpalJellyfish, 2019-11-19:
I have installed usbmuxd on WSL and I could not establish a connection with my iPhone.

nikias, 2019-11-20:
I am not sure you even need usbmuxd there. Probably you can just talk to the windows one, given that iTunes is installed. Set environment variable `USBMUXD_SOCKET_ADDRESS=127.0.0.1:27015` and try to use the tools?

OpalJellyfish, 2019-11-21:
@nikias I have iTunes installed on Windows and I have usbmuxd installed on WSL. I tried adding `export USBMUXD_SOCKET_ADDRESS="127.0.0.1:27015"` to my .bashrc file and sourcing it before running my program.

Specifically, my program is for a Cozmo robot and while this worked perfectly in my Ubuntu virtual machine, my iPhone is not detected when I try running the program on WSL:
```
$ python3 ./02_drive_and_turn.py
2019-11-20 15:56:01,147 cozmo.general WARNING  No iOS device found running Cozmo: [Errno 2] No such file or directory
2019-11-20 15:56:01,151 cozmo.general WARNING  No Android device found running Cozmo: No connected Android devices running Cozmo in SDK mode
A connection error occurred: No devices connected running Cozmo in SDK mode
```

Is there something else I need to do?

haen7, 2020-06-15:
You can use `socat`. Enter this command in WSL:
`sudo socat UNIX-LISTEN:/var/run/usbmuxd,mode=777,reuseaddr,fork TCP:127.0.0.1:27015`
the you can use tools like idevicerestore or ideviceinstaller.

rchatterjee, 2022-03-14:
Any update on this? the `socat` solution does not work for me. 

```
$ sudo socat UNIX-LISTEN:/var/run/usbmuxd,mode=777,reuseaddr,fork TCP:127.0.0.1:27015
2022/03/14 10:30:48 socat[12201] E connect(5, AF=2 127.0.0.1:27015, 16): Connection refused
2022/03/14 10:30:48 socat[12202] E connect(5, AF=2 127.0.0.1:27015, 16): Connection refused
```

```cmd
> wsl --status
Default Distribution: Ubuntu
Default Version: 2

Windows Subsystem for Linux was last updated on 7/21/2021
WSL automatic updates are on.

Kernel version: 5.10.16```

eternalphane, 2023-01-22:
@rchatterjee `AppleMobileDeviceService` is listening `27015` on `127.0.0.1`, which means you can't simply access the port from within WSL. You should forward the port on the gateway of WSL vSwitch and add a firewall rule to allow accessing Windows ports from within WSL using commands below:

In Windows PowerShell:
```pwsh
# add firewall rule (this is a one-time setup)
New-NetFirewallRule -DisplayName "WSL" -Direction Inbound -InterfaceAlias "vEthernet (WSL)" -Action Allow

# port forwarding
$host = $(wsl -- ip route list default `| awk '{print \$3}')
# use `netsh interface portproxy` to forward port
netsh interface portproxy set v4tov4 listenport=27015 connectaddress=127.0.0.1 connectport=27015 listenaddress=$host
# or use `ncat` if nmap is installed (this will block the current terminal)
ncat -lk $host 27015 -c "ncat 127.0.0.1 27015"
```

In WSL bash:
```bash
export USBMUXD_SOCKET_ADDRESS="$(ip route list default | awk '{print $3}'):27015"
ideviceinfo
```

nikias, 2023-01-22:
May I ask why you want to go through WSL? You can compile the lib and tools for Windows natively, in fact I already have github actions that do that on every commit.

eternalphane, 2023-01-23:
> May I ask why you want to go through WSL?
Since I already built a macOS cross toolchain on WSL
> You can compile the lib and tools for Windows natively
May I ask how you make GNU autotools work on Windows?

mexmer, 2023-01-24:
> > May I ask why you want to go through WSL?
> 
> Since I already built a macOS cross toolchain on WSL
> 
> > You can compile the lib and tools for Windows natively
> 
> May I ask how you make GNU autotools work on Windows?

autotools are in cygwin enviroments for windows, you can use either cygwin, mingw or msys2, i prefer using msys2

nikias, 2023-01-24:
> > You can compile the lib and tools for Windows natively
> 
> May I ask how you make GNU autotools work on Windows?

Use msys2: https://www.msys2.org

Then install a few packages.

These packages are what I install for the github actions build action:
base-devel
git
mingw-w64-x86_64-gcc
make
libtool
autoconf
automake-wrapper

You should be able to install the packages with pacman -S PACKAGENAME

Then you go and compile and install the different sub-projects in the following order:
libplist
libimobiledevice-glue
libusbmuxd
libimobiledevice
libirecovery
idevicerestore

the last two are only required if you want to restore firmware.

"compile and install" in this case means basically doing
```
./autogen.sh && make && make install
```
libimobiledevice and libplist have python bindings via cython, and since that would require additional packages and a sane python environment, I suggest to not build them if you don't need them. So for those, run this instead:
```
./autogen.sh --without-cython
```

kjohnsen, 2023-04-09:
I was able to install this [old, pre-built version](https://github.com/libimobiledevice-win32/imobiledevice-net) on Windows super easily [using Scoop](https://scoop.sh/#/apps?q=libimobile&s=0&d=1&o=false). Looks like someone else maintains a nightly build as well.

Spencer-PW, 2024-02-01:
Just want to add:
In my case, I need `mingw-w64-x86_64-openssl` package for `libimobiledevice`'s `autogen.sh` to work.

nikias, 2024-02-01:
Why do you want to build it when you can just install the official one for Windows? Just install AppleMobileDeviceSupport64.msi

Spencer-PW, 2024-02-01:
Because the ultimate goal of mine is to make `libimobiledevice` working, not just `libusbmuxd`. That's why I was following the guide here:

> > > You can compile the lib and tools for Windows natively
> > 
> > 
> > May I ask how you make GNU autotools work on Windows?
> 
> Use msys2: https://www.msys2.org
> 
> Then install a few packages.
> 
> These packages are what I install for the github actions build action: base-devel git mingw-w64-x86_64-gcc make libtool autoconf automake-wrapper
> 
> You should be able to install the packages with pacman -S PACKAGENAME
> 
> Then you go and compile and install the different sub-projects in the following order: libplist libimobiledevice-glue libusbmuxd libimobiledevice libirecovery idevicerestore
> 
> the last two are only required if you want to restore firmware.
> 
> "compile and install" in this case means basically doing
> 
> ```
> ./autogen.sh && make && make install
> ```
> 
> libimobiledevice and libplist have python bindings via cython, and since that would require additional packages and a sane python environment, I suggest to not build them if you don't need them. So for those, run this instead:
> 
> ```
> ./autogen.sh --without-cython
> ```
> 
> Then obviously followed by
> 
> ```
> make && make install
> ```