libimobiledevice/usbmuxd #127: How to connect to AppleWatch 5

userlandkernel, 2019-12-03:
# Scenario

- I have an AppleWatch 5 paired with my iPhone.
- Recently I discovered that com.apple.companion_proxy can tunnel ports (gizmoPort) to the iPhone

# Question
Let's say I forwarded port 62078 to the iPhone, and the created port from the iPhone to the mac (ex: port 1337 on mac), how do I then make libimobiledevice work with that port?

Please be as kind to explain how this works or which code I need to change to make it work.

JCsplash, 2020-05-08:
I'm interested in this as well. If somehow Mac could get Apple Watch battery charge through iPhone, that'd be very helpful.

userlandkernel, 2020-05-14:
Hi @JCsplash do you mean forwarding actual current wirelessly from the iPhone to the Watch or do you mean getting battery statistics from the Watch via the iPhone to the mac?

JCsplash, 2020-05-14:
@userlandkernel  I meant getting the battery stats from the watch to the mac via iPhone. But it seems like there’s a way to directly pass the battery stats to mac (skipping the iPhone) - via com.apple.companion_proxy. 

I think Nikias the creator of libimobiledevice is working on implementing this for libimobiledevice.

https://github.com/libimobiledevice/libimobiledevice/issues/947