libimobiledevice/usbmuxd #137: [arm linux]Device is not paired with this host

yeahtoo, 2020-03-17:
i cross compile usbmuxd 1.1.1 on a arm linux chip with arm-himix100-linux-gcc, when run this application, connect failed,the log display this:

> [04:51:44.452][4] preflight_worker_handle_device_add: Device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b is not paired with this host.

but the phone have any trust tips,and the same version code i comlile on ubuntu16.04 with gcc and the other arm linux with aarch64-himix100-linux-gcc was work successful and no errors.

thanks for help!
this is the detail log when new connect the phone:

> [04:51:44.329][7] poll() returned 1
main_loop:232 FD_USB
[04:51:44.329][4] Found new device with v/p 05ac:12a8 at 1-19
[04:51:44.335][4] Setting configuration for device 1-19, from 1 to 4
[04:51:44.402][4] Found interface 1 with endpoints 04/85 for device 1-19
[04:51:44.406][4] Using wMaxPacketSize=512 for device 1-19
[04:51:44.407][4] USB Speed is 480 MBit/s for device 1-19
[04:51:44.407][7] fd count is 4
[04:51:44.408][7] poll() returned 2
main_loop:232 FD_USB
[04:51:44.408][4] Got lang ID 1033 for device 1-19
[04:51:44.409][7] fd count is 4
[04:51:44.409][7] poll() returned 1
main_loop:232 FD_USB
[04:51:44.417][4] Got serial '61b75e9f2e0c9f0c60e7221b6cec287820e7539b' for device 1-19
[04:51:44.419][3] Connecting to new device on location 0x10013 as ID 1
[04:51:44.419][6] send_packet(1, 0x0, 0xbedac97c, (nil), 0)
[04:51:44.420][5] All 3 RX loops started successfully
[04:51:44.421][6] TX callback dev 1-19 len 20 -> 20 status 0
[04:51:44.421][6] RX callback dev 1-19 len 20 status 0
[04:51:44.422][6] Mux data input for device 0x30088: 0x424c8 len 20
[04:51:44.422][6] send_packet(1, 0x2, (nil), 0x1c09c, 1)
[04:51:44.423][3] Connected to v2.0 device 1 on location 0x10013 with serial number 61b75e9f2e0c9f0c60e7221b6cec287820e7539b
[04:51:44.424][6] TX callback dev 1-19 len 17 -> 17 status 0
[04:51:44.425][7] fd count is 4
[04:51:44.425][4] preflight_worker_handle_device_add: Starting preflight on device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b...
[04:51:44.426][7] poll() returned 1
main_loop:241 FD_LISTEN
[04:51:44.426][4] New client on fd 13 (self)
[04:51:44.426][7] fd count is 5
[04:51:44.427][7] poll() returned 1
main_loop:249 FD_CLIENT
client_process:949 process_recv
[04:51:44.427][7] fd count is 5
[04:51:44.428][7] poll() returned 1
main_loop:249 FD_CLIENT
client_process:949 process_recv
[04:51:44.428][5] Client command in fd 13 len 512 ver 1 msg 8 tag 1
client_command:633 MESSAGE_PLIST
client_command:656 message:Connect
[04:51:44.429][5] Client 13 connection request to device 1 port 62078
[04:51:44.429][5] [OUT] dev=1 sport=1 dport=62078 seq=0 ack=0 flags=0x2 window=131072[512] len=0
[04:51:44.429][6] send_packet(1, 0x6, 0xbedaca7c, (nil), 0)
[04:51:44.430][7] fd count is 5
[04:51:44.430][7] poll() returned 1
main_loop:232 FD_USB
[04:51:44.431][6] TX callback dev 1-19 len 36 -> 36 status 0
[04:51:44.431][6] RX callback dev 1-19 len 36 status 0
[04:51:44.432][6] Mux data input for device 0x30088: 0x464d0 len 36
[04:51:44.432][5] [IN] dev=1 sport=62078 dport=1 seq=0 ack=1 flags=0x12 window=131072[512] len=0
[04:51:44.434][5] [OUT] dev=1 sport=1 dport=62078 seq=1 ack=1 flags=0x10 window=131072[512] len=0
[04:51:44.435][6] send_packet(1, 0x6, 0xbedac92c, (nil), 0)
[04:51:44.435][6] client_notify_connect fd 13 result 0
[04:51:44.436][5] send_pkt fd 13 tag 1 msg 8 payload_length 278
[04:51:44.437][6] update_connection: sendable 49116, events 1, flags 0
[04:51:44.437][6] TX callback dev 1-19 len 36 -> 36 status 0
[04:51:44.438][7] fd count is 5
[04:51:44.438][7] poll() returned 1
main_loop:249 FD_CLIENT
client_process:952 process_send
[04:51:44.439][5] Client 13 switching to CONNECTED state
[... 115 lines cut ...]
[04:51:44.488][5] [IN] dev=1 sport=62078 dport=1 seq=997 ack=1358 flags=0x10 window=131072[512] len=918
[04:51:44.490][6] update_connection: sendable 49116, events 5, flags 0
[04:51:44.490][5] [OUT] dev=1 sport=1 dport=62078 seq=1358 ack=997 flags=0x10 window=131072[512] len=0
[04:51:44.491][6] send_packet(1, 0x6, 0xbedac924, (nil), 0)
[04:51:44.492][6] update_connection: sendable 49116, events 5, flags 0
[04:51:44.492][6] TX callback dev 1-19 len 36 -> 36 status 0
[04:51:44.493][7] fd count is 5
[04:51:44.493][7] poll() returned 1
main_loop:249 FD_CLIENT
client_process:944 CLIENT_CONNECTED
[04:51:44.494][6] client_process in CONNECTED state
[04:51:44.494][6] device_client_process (4)
[04:51:44.494][6] client_write fd 13 buf 0xb5b5c008 len 918
[04:51:44.500][6] update_connection: sendable 49116, events 1, flags 1
[04:51:44.500][7] fd count is 5
[04:51:44.530][7] poll() returned 0
[04:51:44.530][5] Sending ACK due to expired timeout (20623262 -> 20623300)
[04:51:44.530][5] [OUT] dev=1 sport=1 dport=62078 seq=1358 ack=1915 flags=0x10 window=131072[512] len=0
[04:51:44.530][6] send_packet(1, 0x6, 0xbedacacc, (nil), 0)
[04:51:44.530][6] update_connection: sendable 49116, events 1, flags 0
[04:51:44.530][7] fd count is 5
[04:51:44.530][7] poll() returned 1
main_loop:232 FD_USB
[04:51:44.531][6] TX callback dev 1-19 len 36 -> 36 status 0
[04:51:44.531][7] fd count is 5

yeahtoo, 2020-03-17:
then i track the code and found that the dir "/var/lib/lockdown/" has no id.plist file.
on the  normal machine has this file,and if i delete this file, the phone will display trust tips,and will generate this file again.
but the not normal machine never generate this file ever and has any trust tips...

nikias, 2020-03-17:
Well `Device %s is not paired with this host` is not an error message. This is shown when the device is not paired yet. You can then see
```
[04:51:44.466][4] preflight_worker_handle_device_add: Found ProductVersion 13.3.1 device
```
And then it does 
```
[04:51:44.466][5] lockdownd_set_untrusted_host_buid: Setting UntrustedHostBUID to 32B5AAD2-16AE-7B85-DEBE-11DC865B9E32
```
followed by some packets (see [OUT] and [IN] log messages)

At some point it should call `lockdownd_pair` but it doesn't seem to complete this. Can you make sure libimobiledevice is configured and compiled with `--enable-debug-code` and then add a line
```
idevice_set_debug_level(1);
```
to `src/preflight.c` here: https://github.com/libimobiledevice/usbmuxd/blob/master/src/preflight.c#L228  
and then recompile & re-run usbmuxd so we can see the contents of the packets.

yeahtoo, 2020-03-17:
> Well `Device %s is not paired with this host` is not an error message. This is shown when the device is not paired yet. You can then see
>
> ```
> [04:51:44.466][4] preflight_worker_handle_device_add: [...]

thanks for your reply!

here is the debug info:

> main_loop:232 FD_USB
[04:02:51.849][4] Found new device with v/p 05ac:12a8 at 1-34
[04:02:51.856][4] Setting configuration for device 1-34, from 1 to 4
[04:02:51.922][4] Found interface 1 with endpoints 04/85 for device 1-34
[04:02:51.926][4] Using wMaxPacketSize=512 for device 1-34
[04:02:51.927][4] USB Speed is 480 MBit/s for device 1-34
main_loop:232 FD_USB
[04:02:51.928][4] Got lang ID 1033 for device 1-34
main_loop:232 FD_USB
[04:02:51.930][4] Got serial '61b75e9f2e0c9f0c60e7221b6cec287820e7539b' for device 1-34
[04:02:51.930][3] Connecting to new device on location 0x10022 as ID 1
[04:02:51.931][5] All 3 RX loops started successfully
[04:02:51.932][3] Connected to v2.0 device 1 on location 0x10022 with serial number 61b75e9f2e0c9f0c60e7221b6cec287820e7539b
[04:02:51.942][4] preflight_worker_handle_device_add: Starting preflight on device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b...
main_loop:241 FD_LISTEN
[04:02:51.943][4] New client on fd 13 (self)
main_loop:249 FD_CLIENT
main_loop:249 FD_CLIENT
[04:02:51.944][5] Client command in fd 13 len 512 ver 1 msg 8 tag 1
[04:02:51.945][5] Client 13 connection request to device 1 port 62078
[04:02:51.947][5] [OUT] dev=1 sport=1 dport=62078 seq=0 ack=0 flags=0x2 window=131072[512] len=0
main_loop:232 FD_USB
main_loop:232 FD_USB
[04:02:51.949][5] [IN] dev=1 sport=62078 dport=1 seq=0 ack=1 flags=0x12 window=131072[512] len=0
[04:02:51.950][5] [OUT] dev=1 sport=1 dport=62078 seq=1 ack=1 flags=0x10 window=131072[512] len=0
[04:02:51.951][5] send_pkt fd 13 tag 1 msg 8 payload_length 278
main_loop:249 FD_CLIENT
[04:02:51.952][5] Client 13 switching to CONNECTED state
main_loop:249 FD_CLIENT
[04:02:51.953][5] [OUT] dev=1 sport=1 dport=62078 seq=1 ack=1 flags=0x10 window=131072[512] len=284
main_loop:232 FD_USB
main_loop:232 FD_USB
[04:02:51.955][5] [IN] dev=1 sport=62078 dport=1 seq=1 ack=285 flags=0x10 window=131072[512] len=301
[04:02:51.956][5] [OUT] dev=1 sport=1 dport=62078 seq=285 ack=1 flags=0x10 window=131072[512] len=0
main_loop:232 FD_USB
main_loop:249 FD_CLIENT
[04:02:51.957][4] preflight_worker_handle_device_add: Device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b is not paired with this host.
main_loop:249 FD_CLIENT
[04:02:51.958][5] [OUT] dev=1 sport=1 dport=62078 seq=285 ack=302 flags=0x10 window=131072[512] len=332
main_loop:232 FD_USB
main_loop:232 FD_USB
[04:02:51.960][5] [IN] dev=1 sport=62078 dport=1 seq=302 ack=617 flags=0x10 window=131072[512] len=331
[04:02:51.961][5] [OUT] dev=1 sport=1 dport=62078 seq=617 ack=302 flags=0x10 window=131072[512] len=0
main_loop:249 FD_CLIENT
[04:02:51.963][4] preflight_worker_handle_device_add: Found ProductVersion 13.3.1 device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b
[04:02:51.963][5] reading key SystemBUID from config_file /var/lib/lockdown/SystemConfiguration.plist
[04:02:51.963][5] got SystemBUID 32B5AAD2-16AE-7B85-DEBE-11DC865B9E32
[04:02:51.963][5] using 32B5AAD2-16AE-7B85-DEBE-11DC865B9E32 as SystemBUID
[04:02:51.964][5] lockdownd_set_untrusted_host_buid: Setting UntrustedHostBUID to 32B5AAD2-16AE-7B85-DEBE-11DC865B9E32
04:02:51 property_list_service.c:132 internal_plist_send(): sending 404 bytes
04:02:51 service.c:144 service_send(): sending 4 bytes
04:02:51 service.c:144 service_send(): sending 404 bytes
04:02:51 property_list_service.c:137 internal_plist_send(): sent 404 bytes
04:02:51 property_list_service.c:138 internal_plist_send(): printing 404 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>usbmuxd</string>
        <key>Key</key>
[... 74 lines cut ...]
        cmxJTHBkd0QNCm16QlErZ2RUQmpjd0VqdmwrRTdNVm5WUUJvd3A4MWJjREtVL1RmU1dz
        OTI4RVA1dEJ3Vm50eWJaUW5LTEQreGsNCkdEOTNNRkYranlvNnhBN1RadHdMNlZ5dGk0
        TE9VcERwZXMvcEI3eVBIdGVXSnFRaVRGT2pTZEwvcXVvMGRMbGINCjVmRTUzdHlEQ3ox
        UlJ4MEk4NDBwL2poOGJaV2dvWEN2U0tGMStqNm9qekZOc285TCtwWXdLUVI5cGZlMkRG
        QVENCjltT3orNlgrRzBEZnZ3d1J0ZXJKWEhVK1l0Zmp6dS9wUndJREFRQUIKLS0tLS1F
        TkQgUlNBIFBVQkxJQyBLRVktLS0tLQo=
        </data>
</dict>
</plist>
04:02:51 lockdown.c:458 lockdownd_get_value(): success
04:02:51 lockdown.c:469 lockdownd_get_value(): has a value
04:02:51 lockdown.c:812 pair_record_generate(): device public key follows:
-----BEGIN RSA PUBLIC KEY-----
MIIBCgKCAQEAtwZFpStds7+OpdhArvKp8SDHzYOwcjlxRHNLvmx9Vm0qukqCvp71
IkV6fg18DC0Qn0L8htidagH0x9fKIJMU883BBaKpvz4osSsATyrLS7ALrlILpdwD
mzBQ+gdTBjcwEjvl+E7MVnVQBowp81bcDKU/TfSWs928EP5tBwVntybZQnKLD+xk
GD93MFF+jyo6xA7TZtwL6Vyti4LOUpDpes/pB7yPHteWJqQiTFOjSdL/quo0dLlb
5fE53tyDCz1RRx0I840p/jh8bZWgoXCvSKF1+j6ojzFNso9L+pYwKQR9pfe2DFAQ
9mOz+6X+G0DfvwwRterJXHU+Ytfjzu/pRwIDAQAB
-----END RSA PUBLIC KEY-----

04:02:51 userpref.c:391 pair_record_generate_keys_and_certs(): Generating keys and certificates...
[04:02:52.010][5] Sending ACK due to expired timeout (104090749 -> 104090780)
[04:02:52.010][5] [OUT] dev=1 sport=1 dport=62078 seq=1358 ack=1915 flags=0x10 window=131072[512] len=0
main_loop:232 FD_USB

yeahtoo, 2020-03-17:
[image]

i compare the work normal machine (left) and the not successed machine (right) debug info, found after "userpref.c:391 pair_record_generate_keys_and_certs(): Generating keys and certificates...", then have no more info, i don't understand why this? the phone is the same phone

nikias, 2020-03-17:
Hmm what hardware is this on? It might be slow to generate keys for it. Can you try this patch for libimobiledevice and see if it will succeed with pairing then?
https://gist.github.com/nikias/bc855639fed21f6db3800e32da82d8b2

yeahtoo, 2020-03-18:
> Hmm what hardware is this on? It might be slow to generate keys for it. Can you try this patch for libimobiledevice and see if it will succeed with pairing then?
> https://gist.github.com/nikias/bc855639fed21f6db3800e32da82d8b2

hello, the normal machine chip is HI3559AV100, and the not normal is HI3518EV300, and i patched the code, debug info is this:

`18:59:24 lockdown.c:847 pair_record_generate(): key generation is slow, making sure the lockdown connection is kept alive
18:59:24 property_list_service.c:132 internal_plist_send(): sending 325 bytes
18:59:24 service.c:144 service_send(): sending 4 bytes
18:59:24 service.c:144 service_send(): sending 325 bytes
18:59:24 property_list_service.c:137 internal_plist_send(): sent 325 bytes
18:59:24 property_list_service.c:138 internal_plist_send(): printing 325 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>usbmuxd</string>
        <key>Key</key>
        <string>WiFiAddress</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
main_loop:249 FD_CLIENT
[18:59:24.071][5] [OUT] dev=1 sport=1 dport=62078 seq=1358 ack=1915 flags=0x10 window=131072[512] len=329
main_loop:232 FD_USB
main_loop:232 FD_USB
[18:59:24.073][5] [IN] dev=1 sport=62078 dport=1 seq=1915 ack=1687 flags=0x10 window=131072[512] len=339
[18:59:24.073][5] [OUT] dev=1 sport=1 dport=62078 seq=1687 ack=1915 flags=0x10 window=131072[512] len=0
main_loop:249 FD_CLIENT
18:59:24 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
18:59:24 property_list_service.c:211 internal_plist_receive_timeout(): 335 bytes following
18:59:24 property_list_service.c:224 internal_plist_receive_timeout(): received 335 bytes
18:59:24 property_list_service.c:250 internal_plist_receive_timeout(): printing 335 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Key</key>
        <string>WiFiAddress</string>
        <key>Request</key>
        <string>GetValue</string>
        <key>Value</key>
        <string>5c:1d:d9:6e:de:c5</string>
</dict>
</plist>
18:59:24 lockdown.c:459 lockdownd_get_value(): success
18:59:24 lockdown.c:470 lockdownd_get_value(): has a value
[18:59:24.110][5] Sending ACK due to expired timeout (157882843 -> 157882880)
[18:59:24.110][5] [OUT] dev=1 sport=1 dport=62078 seq=1687 ack=2254 flags=0x10 window=131072[512] len=0
main_loop:232 FD_USB

18:59:30 lockdown.c:847 pair_record_generate(): key generation is slow, making sure the lockdown connection is kept alive
18:59:30 property_list_service.c:132 internal_plist_send(): sending 325 bytes
18:59:30 service.c:144 service_send(): sending 4 bytes
18:59:30 service.c:144 service_send(): sending 325 bytes
18:59:30 property_list_service.c:137 internal_plist_send(): sent 325 bytes
18:59:30 property_list_service.c:138 internal_plist_send(): printing 325 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>usbmuxd</string>
        <key>Key</key>
        <string>WiFiAddress</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
main_loop:249 FD_CLIENT
[18:59:30.011][5] [OUT] dev=1 sport=1 dport=62078 seq=1687 ack=2254 flags=0x10 window=131072[512] len=329
main_loop:232 FD_USB
main_loop:232 FD_USB
[18:59:30.013][5] [IN] dev=1 sport=62078 dport=1 seq=2254 ack=2016 flags=0x10 window=131072[512] len=339
[18:59:30.013][5] [OUT] dev=1 sport=1 dport=62078 seq=2016 ack=2254 flags=0x10 window=131072[512] len=0
main_loop:249 FD_CLIENT
18:59:30 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
18:59:30 property_list_service.c:211 internal_plist_receive_timeout(): 335 bytes following
18:59:30 property_list_service.c:224 internal_plist_receive_timeout(): received 335 bytes
18:59:30 property_list_service.c:250 internal_plist_receive_timeout(): printing 335 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Key</key>
        <string>WiFiAddress</string>
        <key>Request</key>
        <string>GetValue</string>
        <key>Value</key>
        <string>5c:1d:d9:6e:de:c5</string>
</dict>
</plist>
18:59:30 lockdown.c:459 lockdownd_get_value(): success
18:59:30 lockdown.c:470 lockdownd_get_value(): has a value
[18:59:30.050][5] Sending ACK due to expired timeout (157888784 -> 157888820)
[18:59:30.050][5] [OUT] dev=1 sport=1 dport=62078 seq=2016 ack=2593 flags=0x10 window=131072[512] len=0
main_loop:232 FD_USB`

thanks for your help!

nikias, 2020-03-18:
Oh I see, well the HI3518EV300 doesn't really seem a slow CPU. That output will repeat every 5 seconds, does it just stop?

yeahtoo, 2020-03-18:
> Oh I see, well the HI3518EV300 doesn't really seem a slow CPU. That output will repeat every 5 seconds, does it just stop?

no, this never stoped,and the phone has no turst tips.
how can i do next now? please
i test the usb via a usb disk wirte&read aboot 8.5Mb/s&12.5Mb/s, and the usb net phy also worked normally

yeahtoo, 2020-03-18:
this is the cross complie project, can you look for a while？3q
https://github.com/yeahtoo/openlib

yeahtoo, 2020-03-19:
> Oh I see, well the HI3518EV300 doesn't really seem a slow CPU. That output will repeat every 5 seconds, does it just stop?

hello, now i use a other chip HI3519AV100 for test and still failed, the debug info said "preflight_worker_handle_device_add: Waiting for user to trust this computer on device xx", but the phone has no trust tips, here is the detail debug info, thank your help!

```
[00:03:31.112][3] usbmuxd v1.1.1 starting up
[00:03:31.114][4] Creating socket
[00:03:31.116][5] initialized config_dir to /var/lib/lockdown
[00:03:31.117][5] client_init
[00:03:31.117][5] device_init
[00:03:31.117][4] Initializing USB
[00:03:31.117][5] usb_init for linux / libusb 1.0
[00:03:31.118][4] Registering for libusb hotplug events
[00:03:31.118][4] 0 devices detected
[00:03:31.118][3] Initialization complete
main_loop:232 FD_USB

main_loop:232 FD_USB
[00:03:39.242][4] Found new device with v/p 05ac:12a8 at 1-2
[00:03:39.251][4] Setting configuration for device 1-2, from 1 to 4
[00:03:39.321][4] Found interface 1 with endpoints 04/85 for device 1-2
[00:03:39.321][4] Using wMaxPacketSize=512 for device 1-2
[00:03:39.321][4] USB Speed is 480 MBit/s for device 1-2
main_loop:232 FD_USB
main_loop:232 FD_USB
[00:03:39.321][4] Got lang ID 1033 for device 1-2
main_loop:232 FD_USB
[00:03:39.321][4] Got serial '61b75e9f2e0c9f0c60e7221b6cec287820e7539b' for device 1-2
[00:03:39.322][3] Connecting to new device on location 0x10002 as ID 1
[00:03:39.322][5] All 3 RX loops started successfully
[00:03:39.322][3] Connected to v2.0 device 1 on location 0x10002 with serial number 61b75e9f2e0c9f0c60e7221b6cec287820e7539b
[00:03:39.322][4] preflight_worker_handle_device_add: Starting preflight on device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b...
main_loop:241 FD_LISTEN
[00:03:39.322][4] New client on fd 13 (self)
main_loop:249 FD_CLIENT
main_loop:249 FD_CLIENT
[00:03:39.322][5] Client command in fd 13 len 512 ver 1 msg 8 tag 1
[00:03:39.323][5] Client 13 connection request to device 1 port 62078
[00:03:39.323][5] [OUT] dev=1 sport=1 dport=62078 seq=0 ack=0 flags=0x2 window=131072[512] len=0
main_loop:232 FD_USB
main_loop:232 FD_USB
[00:03:39.323][5] [IN] dev=1 sport=62078 dport=1 seq=0 ack=1 flags=0x12 window=131072[512] len=0
[00:03:39.323][5] [OUT] dev=1 sport=1 dport=62078 seq=1 ack=1 flags=0x10 window=131072[512] len=0
[00:03:39.323][5] send_pkt fd 13 tag 1 msg 8 payload_length 278
main_loop:249 FD_CLIENT
[00:03:39.323][5] Client 13 switching to CONNECTED state
main_loop:249 FD_CLIENT
[00:03:39.324][5] [OUT] dev=1 sport=1 dport=62078 seq=1 ack=1 flags=0x10 window=131072[512] len=284
main_loop:232 FD_USB
main_loop:232 FD_USB
[00:03:39.324][5] [IN] dev=1 sport=62078 dport=1 seq=1 ack=285 flags=0x10 window=131072[512] len=301
[00:03:39.324][5] [OUT] dev=1 sport=1 dport=62078 seq=285 ack=1 flags=0x10 window=131072[512] len=0
main_loop:249 FD_CLIENT
[00:03:39.324][4] preflight_worker_handle_device_add: Device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b is not paired with this host.
main_loop:249 FD_CLIENT
[00:03:39.325][5] [OUT] dev=1 sport=1 dport=62078 seq=285 ack=302 flags=0x10 window=131072[512] len=332
main_loop:232 FD_USB
main_loop:232 FD_USB
[00:03:39.325][5] [IN] dev=1 sport=62078 dport=1 seq=302 ack=617 flags=0x10 window=131072[512] len=331
[00:03:39.325][5] [OUT] dev=1 sport=1 dport=62078 seq=617 ack=302 flags=0x10 window=131072[512] len=0
main_loop:249 FD_CLIENT
[00:03:39.325][4] preflight_worker_handle_device_add: Found ProductVersion 13.3.1 device 61b75e9f2e0c9f0c60e7221b6cec287820e7539b
[00:03:39.326][5] reading key SystemBUID from config_file /var/lib/lockdown/SystemConfiguration.plist
[00:03:39.326][5] got SystemBUID B38DFBCA-438D-B3BF-58C7-E586FFE1A91B
[00:03:39.326][5] using B38DFBCA-438D-B3BF-58C7-E586FFE1A91B as SystemBUID
[00:03:39.326][5] lockdownd_set_untrusted_host_buid: Setting UntrustedHostBUID to B38DFBCA-438D-B3BF-58C7-E586FFE1A91B
00:03:39 property_list_service.c:132 internal_plist_send(): sending 404 bytes
00:03:39 service.c:144 service_send(): sending 4 bytes
main_loop:249 FD_CLIENT
00:03:39 service.c:144 service_send(): sending 404 bytes
[... 299 lines cut ...]
^C[00:04:05.437][3] Caught signal 2, exiting
[00:04:05.438][4] Event processing interrupted
[00:04:05.438][3] usbmuxd shutting down
[00:04:05.438][5] device_kill_connections
[00:04:05.438][5] connection_teardown dev 1 sport 2 dport 49168
[00:04:05.438][5] [OUT] dev=1 sport=2 dport=49168 seq=654 ack=1 flags=0x4 window=131072[512] len=0
[00:04:05.438][4] Disconnecting client fd 15
00:04:05 idevice.c:552 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -104 (Connection reset by peer)
00:04:05 service.c:167 service_receive_with_timeout(): could not read data
00:04:05 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
00:04:05 notification_proxy.c:273 np_get_notification(): NotificationProxy: error -256 occurred!
00:04:05 notification_proxy.c:62 np_unlock(): Unlocked
[00:04:05.539][5] usb_shutdown
[00:04:05.540][3] Removed device 1 on location 0x10002
[00:04:05.540][5] client_device_remove: id 1
[00:04:05.540][5] usb_disconnect: cancelling RX xfer 0x30bc0
[00:04:05.540][5] usb_disconnect: cancelling RX xfer 0x34c50
[00:04:05.540][5] usb_disconnect: cancelling RX xfer 0x34cd8
[00:04:05.540][5] Device 1-2 RX transfer cancelled
[00:04:05.540][5] Device 1-2 RX transfer cancelled
[00:04:05.540][5] Device 1-2 RX transfer cancelled
[00:04:05.540][5] device_shutdown
[00:04:05.540][5] client_shutdown
[00:04:05.540][3] Shutdown complete
```

nikias, 2020-03-19:
Ok this time we have an actual error message:
```00:03:39 userpref.c:570 pair_record_generate_keys_and_certs(): ERROR: Signing device certificate with root private key failed!
```
I wonder if something is wrong with OpenSSL here...

yeahtoo, 2020-03-20:
hello, after add debug info, the firse error was happen here:

```
userpref.c:pair_record_generate_keys_and_certs-755 root_cert_pem data:-----BEGIN CERTIFICATE-----
MIGCMG6gAwIBAgIBADANBgkqhkiG9w0BAQUFADAAMB4XDTcwMDEwMTAzMjE0NFoX
DTc5MTIzMDAzMjE0NFowADAcMA0GCSqGSIb3DQEBAQUAAwsAMAgCAQACAwEAAaMT
MBEwDwYDVR0TAQH/BAUwAwEB/zANBgkqhkiG9w0BAQUFAAMBAA==
-----END CERTIFICATE-----
<Sk
    size:237
userpref.c:pair_record_generate_keys_and_certs-756 root_key_pem data:-----BEGIN RSA PRIVATE KEY-----
MIGZAgEAAgEAAgMBAAECAQACfQCgq/epAAAAALxpYwgAAAAA8a5qPQAAAACoEF8L
AAAAAFiYOHsAAAAAu6rF3gAAAAAhDwrrAAAAAPDrjYIAAAAAYLnBDQAAAACIEFOB
AAAAAPmHoygAAAAAcwIuLQAAAACM72pcAAAAAAtCa6sAAAAAfC5NaAAAAABp9yIb
AgEAAgEAAgEAAgEA
-----END RSA PRIVATE KEY-----
 size:274
userpref.c:pair_record_generate_keys_and_certs-757 host_cert_pem data:-----BEGIN CERTIFICATE-----
MIGPMHugAwIBAgIBADANBgkqhkiG9w0BAQUFADAAMB4XDTcwMDEwMTAzMjE0NFoX
DTc5MTIzMDAzMjE0NFowADAcMA0GCSqGSIb3DQEBAQUAAwsAMAgCAQACAwEAAaMg
MB4wDAYDVR0TAQH/BAIwADAOBgNVHQ8BAf8EBAMCBaAwDQYJKoZIhvcNAQEFBQAD
AQA=
-----END CERTIFICATE-----
9Yj|size:254
userpref.c:pair_record_generate_keys_and_certs-758 host_key_pem data:-----BEGIN RSA PRIVATE KEY-----
MIGYAgEAAgEAAgMBAAECAQACfHU6ZIAAAAAAbNefwQAAAABNQZ/zAAAAAPB7+voA
AAAA5wuQpAAAAACyo8NKAAAAAEu1DMEAAAAApULBWwAAAADL8Do5AAAAAInjZXUA
AAAAWIoqTAAAAABUasLXAAAAANj6AdoAAAAAmSMm9QAAAACfHhxnAAAAANWwn58C
AQACAQACAQACAQA=
-----END RSA PRIVATE KEY-----
size:274
userpref.c:pair_record_generate_keys_and_certs-759 dev_cert_pem data:(null) size:0
lockdown.c:pair_record_generate-819 uret:-3
```

the dev_cert_pem data is null and size 0

yeahtoo, 2020-03-20:
> Ok this time we have an actual error message:
> 
> ```
> 00:03:39 userpref.c:570 pair_record_generate_keys_and_certs(): ERROR: Signing device certificate with root private key failed!
> ```
> 
> I wonder if something is wrong with OpenSSL here...

ohh, the error was happen here "X509_sign(dev_cert, root_pkey, EVP_sha1()", 
userpref.c:pair_record_generate_keys_and_certs-556,

yeahtoo, 2020-03-20:
finally,the problem had been solved,the reason is the arm cpu is 32bit, the HI3559AV100 is a 64bit chip,but when cross complile openssl don't use i386, so when use openssl to pair phone always not success.
thanks for your help!

closed 2020-03-20 by yeahtoo