libimobiledevice/usbmuxd #138: usbmux breaks Touch Bar, ALS, and camera on MacBookPro's

roadrunner2, 2020-04-07:
When running on a MacBookPro, commits ec5ff91cfabd30637f8af8f5c79baf4d7818ce57 and 8a69f1a78a58476f77b66916091d2405d0cd815f cause usbmux to misconfigure (configuration 0 is not a valid config) the iBridge device (T1 chip), breaking Touch Bar, light sensor, and camera functionality. The problem was originally reported at roadrunner2/macbook12-spi-driver#42.

I don't really understand the usbmux in any detail, so not sure how to fix this exactly, but it seems like you need to distinguish between internal USB devices and external ones plugged into the system, and avoid touching the internal ones.

mexmer, 2020-04-14:
usbmuxd is not supposed to be run on OSX, it has own mux server

nikias, 2020-04-14:
@mexmer I think he is running it on Linux on the MacBook Pro.

mexmer, 2020-04-14:
oh i see.

now question is, how to limit usbmux only to ios devices, without need to explicitly list them.

nikias, 2020-04-14:
@mexmer I am working on it with @rickmark.

nikias, 2022-04-12:
On Linux, configuration 0 basically means the device is "unconfigured". Not sure why this breaks though; the kernel driver should know about this. Anyway, I think maybe the 8600 device might just need a different rule. Can you guys tell me what the "normal" config of the T2 is? Does it work after modifying the udev rule to set it to 1 on the 8600 (T2)? Try this patch:

```diff
diff --git a/udev/39-usbmuxd.rules.in b/udev/39-usbmuxd.rules.in
index ab20517..7f9bf10 100644
--- a/udev/39-usbmuxd.rules.in
+++ b/udev/39-usbmuxd.rules.in
@@ -4,7 +4,8 @@
 SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*|5ac/8600/*", TAG+="systemd"

 # Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
-SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*|5ac/8600/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
+SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
+SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/8600/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="1", OWNER="usbmux", @udev_activation_rule@

 # Make sure properties don't get lost when bind action is called
 SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*|5ac/8600/*", ACTION=="bind", ENV{USBMUX_SUPPORTED}="1", OWNER="usbmux", @udev_activation_rule@
```

/cc @davidjo

davidjo, 2022-04-13:
For quick fix just copy the 39-udevadm.rules file from Ubuntu 20.04 (which is missing all |5ac/8600 additions) and touchbar is restored.

davidjo, 2022-06-23:
I just tried this - but get a syslog error
systemd-udevd[353]: /usr/lib/udev/rules.d/39-usbmuxd.rules:7 Invalid key/value pair, ignoring.
systemd-udevd[353]: /usr/lib/udev/rules.d/39-usbmuxd.rules:8 Invalid key/value pair, ignoring.

lines 7 8 are the green lines in your patch

of course the message doesnt say which key is the problem

(so touchbar works because udev doesnt understand the file so ignores it)

night199uk, 2023-02-02:
> On Linux, configuration 0 basically means the device is "unconfigured". Not sure why this breaks though; the kernel driver should know about this. Anyway, I think maybe the 8600 device might just need a different rule. Can you guys tell me what the "normal" config of the T2 is? Does it work after modifying the udev rule to set it to 1 on the 8600 (T2)? Try this patch:
> 
> ```diff
> diff --git a/udev/39-usbmuxd.rules.in b/udev/39-usbmuxd.rules.in
> index ab20517..7f9bf10 100644
> --- a/udev/39-usbmuxd.rules.in
> +++ b/udev/39-usbmuxd.rules.in
> @@ -4,7 +4,8 @@
>  SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*|5ac/8600/*", TAG+="systemd"
> 
>  # Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
> -SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*|5ac/8600/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
> +SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
> +SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/8600/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="1", OWNER="usbmux", @udev_activation_rule@
> 
>  # Make sure properties don't get lost when bind action is called
>  SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*|5ac/8600/*", ACTION=="bind", ENV{USBMUX_SUPPORTED}="1", OWNER="usbmux", @udev_activation_rule@
> ```
> 
> /cc @davidjo
> 
> Don't forget to rebuild and reinstall so the udev rule gets updated.

This patch actually seems to work great. I see it's been a while since this bug was updated. Any chance this patch could get integrated, as with Touch Bar support now in the mainline kernels this would fix the OOTB Touch Bar experience on most T1/T2 Macs.

davidjo, 2023-02-09:
could you post the actual 39-usbmuxd.rules here - hadnt realized this is a pre-cursor file which is filled in and setting up the makefile seems complicated as it just replaces the udev_activation_rule parameter with other parameters which is not clear where they are defined.
(This explains my symbol issues).

KiLLeRRaT, 2023-07-19:
I've also run the patch @nikias posted, and worked for me too.

It'd be create to get this merged in.

Cheers,

davidjo, 2023-07-20:
could you post the value of the string @udev_activation_rule@ in your  actual udev file please

KiLLeRRaT, 2023-07-20:
Sure thing, here you go:

`cat /usr/lib/udev/rules.d/39-usbmuxd.rules`:
```
# usbmuxd (Apple Mobile Device Muxer listening on /var/run/usbmuxd)

# systemd should receive all events relating to device
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/8600/*", TAG+="systemd"

# Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-4]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", @udev_activation_rule@
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/8600/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="1", OWNER="usbmux", @udev_activation_rule@

# Make sure properties don't get lost when bind action is called
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/8600/*", ACTION=="bind", ENV{USBMUX_SUPPORTED}="1", OWNER="usbmux", ENV{SYSTEMD_WANTS}="usbmuxd.service"

# Exit usbmuxd when the last device is removed
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/8600/*", ACTION=="remove", RUN+="/usr/bin/usbmuxd -x"
```

nikias, 2023-07-20:
Fixed with 61b99ab5c25609c11369733a0df97c03a0581a56.

closed 2023-07-20 by nikias

hanscnelson, 2026-11-04:
A current distro package still ships the pre-61b99ab rule, and on this T1 the working configuration is 2, not 1.

Machine: MacBookPro14,2 (2017 13-inch, T1). Internal iBridge is USB `05ac:8600`. Kernel 7.2.15. Arch `usbmuxd` 1.1.1-4 installs `/usr/lib/udev/rules.d/39-usbmuxd.rules` with this add rule:

```
ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/8600/*", ACTION=="add", ... ATTR{bConfigurationValue}="0"
```

That is the old behavior from before commit 61b99ab. Upstream master now sets `8600` to configuration 1. This package does not.

Boot at 12:51, before that package was installed. The Touch Bar and the T1 network interface came up in configuration 2:

```
[drm] Initialized appletbdrm 1.0.0 for 1-3:2.3 on minor 0
apple_t1_ncm 1-3:2.4 eth0: register 'apple_t1_ncm' ... Apple T1 private NCM
input: T1Bridge Touch Bar
```

Boot at 14:24, after installing `usbmuxd` 1.1.1-4 and rebooting. `appletbdrm` registered and then never bound. `bConfigurationValue` for `1-3` was empty, `eth0` did not exist, and systemd logged:

```
dev-dri-touchbar.device: Job dev-dri-touchbar.device/start timed out.
Timed out waiting for device /dev/dri/touchbar.
```

Plymouth stayed at the end of its progress bar for that 90 second device timeout.

So configuration 0 still breaks a T1 MacBookPro14,2, and the configuration that actually enumerated the Touch Bar here was 2 (`1-3:2.3`), not the configuration 1 chosen in 61b99ab. Configuration 1 was not tested. Leaving `8600` out of the configuration assignment lets `t1bridge-cfgselector` bring the chip up in configuration 2. iPhone IDs (`12[9a]*`) can keep the configuration 0 assignment.