libimobiledevice/usbmuxd #139: Build fails if libimobiledevice is installed with custom prefix

exioReed, 2020-04-28:
The build fails if `libimobiledevice` is installed with custom prefix for the following reasons:

1. Typo in [`src/Makefile.am`](https://github.com/libimobiledevice/usbmuxd/blob/ab51bbbe6d78678db1bdd04be1eeec6bb1d59c34/src/Makefile.am) that results in missing CFLAGS for `libimobiledevice`.

```
$ make
make  all-recursive
make[1]: Entering directory '/opt/libimobiledevice/usbmuxd'
Making all in src
make[2]: Entering directory '/opt/libimobiledevice/usbmuxd/src'
  CC       usbmuxd-client.o
  CC       usbmuxd-device.o
  CC       usbmuxd-preflight.o
preflight.c:34:10: fatal error: libimobiledevice/libimobiledevice.h: No such file or directory
 #include <libimobiledevice/libimobiledevice.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

2. Check for enum `idevice_connection_type` in [`configure.ac`](https://github.com/libimobiledevice/usbmuxd/blob/9af2b12552693a47601347e1eafc1e94132d727e/configure.ac#L39) yields a false negative. As a result, `HAVE_ENUM_IDEVICE_CONNECTION_TYPE` is undefined and `idevice_connection_type` and will be redeclared in [`src/preflight.c:47`](https://github.com/libimobiledevice/usbmuxd/blob/8a69f1a78a58476f77b66916091d2405d0cd815f/src/preflight.c#L47).

```
$ make
make  all-recursive
make[1]: Entering directory '/opt/libimobiledevice/usbmuxd'
Making all in src
make[2]: Entering directory '/opt/libimobiledevice/usbmuxd/src'
  CC       usbmuxd-preflight.o
preflight.c:47:6: error: nested redefinition of ‘enum idevice_connection_type’
 enum idevice_connection_type {
      ^~~~~~~~~~~~~~~~~~~~~~~
preflight.c:47:6: error: redeclaration of ‘enum idevice_connection_type’
In file included from preflight.c:34:
/opt/libimobiledevice-1.2.1/include/libimobiledevice/libimobiledevice.h:63:6: note: originally defined here
 enum idevice_connection_type {
      ^~~~~~~~~~~~~~~~~~~~~~~
preflight.c:48:2: error: redeclaration of enumerator ‘CONNECTION_USBMUXD’
  CONNECTION_USBMUXD = 1,
  ^~~~~~~~~~~~~~~~~~
In file included from preflight.c:34:
/opt/libimobiledevice-1.2.1/include/libimobiledevice/libimobiledevice.h:64:2: note: previous definition of ‘CONNECTION_USBMUXD’ was here
  CONNECTION_USBMUXD = 1,
  ^~~~~~~~~~~~~~~~~~
preflight.c:49:2: error: redeclaration of enumerator ‘CONNECTION_NETWORK’
  CONNECTION_NETWORK
  ^~~~~~~~~~~~~~~~~~
In file included from preflight.c:34:
/opt/libimobiledevice-1.2.1/include/libimobiledevice/libimobiledevice.h:65:2: note: previous definition of ‘CONNECTION_NETWORK’ was here
  CONNECTION_NETWORK
  ^~~~~~~~~~~~~~~~~~
make[2]: *** [Makefile:497: usbmuxd-preflight.o] Error 1
make[2]: Leaving directory '/opt/libimobiledevice/usbmuxd/src'
make[1]: *** [Makefile:423: all-recursive] Error 1
make[1]: Leaving directory '/opt/libimobiledevice/usbmuxd'
make: *** [Makefile:355: all] Error 2
```

exioReed, 2020-04-28:
@nikias Could you provide directions on how to fix the second part. Essentially, is has to be ensured that `AC_TRY_COMPILE` succeeds with custom include directories for `libimobiledevice` ([`configure.ac:39`](https://github.com/libimobiledevice/usbmuxd/blob/8a69f1a78a58476f77b66916091d2405d0cd815f/configure.ac#L39)).

exioReed, 2020-04-29:
Regarding part 2.

The documentation of [`AC_TRY_COMPILE`](https://ftp.gnu.org/old-gnu/Manuals/autoconf-2.53/html_node/Examining-Syntax.html) states

> This macro also uses CFLAGS or CXXFLAGS if either C or C++ is the currently selected language, as well as CPPFLAGS, when compiling.

For `AC_TRY_COMPILE` to be successful even if `libimobiledevice` is installed with custom prefix, the necessary include directories could be added as follows:

```diff
diff --git a/configure.ac b/configure.ac
index fb59203..6bac57c 100644
--- a/configure.ac
+++ b/configure.ac
@@ -35,6 +35,9 @@ if test "x$have_limd" = "xyes"; then
     AC_DEFINE(HAVE_LIBIMOBILEDEVICE, 1, [Define if you have libimobiledevice support])
     AC_SUBST(libimobiledevice_CFLAGS)
     AC_SUBST(libimobiledevice_LIBS)
+    # Cache CFLAGS
+    CACHE_CFLAGS_CHECK_ENUM_IDEVICE_CONNECTION_TYPE="$CFLAGS"
+    CFLAGS+=" $libimobiledevice_CFLAGS"
     AC_CACHE_CHECK(for enum idevice_connection_type, ac_cv_enum_idevice_connection_type,
       AC_TRY_COMPILE([
         #include <libimobiledevice/libimobiledevice.h>
@@ -44,6 +47,8 @@ if test "x$have_limd" = "xyes"; then
     if (test "$ac_cv_enum_idevice_connection_type" = "yes"); then
       AC_DEFINE(HAVE_ENUM_IDEVICE_CONNECTION_TYPE, 1, [Define if enum idevice_connection_type is available])
     fi
+    # Restore CFLAGS
+    CFLAGS="$CACHE_CFLAGS_CHECK_ENUM_IDEVICE_CONNECTION_TYPE"
   fi
 else
   if test "x$with_preflight" == "xyes"; then
```

Would this be an acceptable solution?

nikias, 2020-04-29:
@exioReed well in this case you should be able to achieve the same thing by passing `CFLAGS="-I/your/non-standard-prefix"` to `./autogen.sh`

nikias, 2020-04-29:
But hmm. You are getting the non-standard prefix from `pkg-config`?

exioReed, 2020-04-29:
Sure. The `PKG_CHECK_MODULES` for `libimobiledevice` in [`configure.ac:21`](https://github.com/libimobiledevice/usbmuxd/blob/52a24c5a0e0dc80e2ced8d20fc3f14e82938e8e1/configure.ac#L21) succeeds and `pkg-config` yields

```
$ pkg-config --cflags libimobiledevice-1.0
-I/opt/libplist-2.1.0/include -I/opt/libusbmuxd-2.0.2/include -I/opt/imobiledevice-1.2.1/include
```

nikias, 2020-04-30:
I pushed a fix.

closed 2020-04-30 by nikias

voltagex, 2021-05-14:
I am still hitting this with the latest code from master.

nikias, 2021-05-15:
What exactly happens?

voltagex, 2021-05-16:
@nikias apologies, it was an issue with the include path - I got it working.