libimobiledevice/usbmuxd #142: idevicebackup2 failure with usbmuxd 1.1.1

corsac-s, 2020-06-17:
Hi,
updating usbmuxd to 1.1.1 (after upgrading libplist, libusbmuxd, libimobiledevice) leads to failure during idevicebackup2 operations.

Here is the log from idevicebackup2:
```
[====================================              ]  71% Finished
Receiving files
[==================================================] 100% (38.5 MB/38.5 MB)       
[======================================            ]  74% Finished
Receiving files
[=======================================           ]  77% Finished/32.7 MB)      
ERROR: Could not receive from mobilebackup2 (-4)
Received 1846 files from device.
Backup Aborted.
```
Here's the journalctl output with logs from usbmuxd:
```
juin 17 23:25:31 scapa usbmuxd[50440]: [23:25:31.467][1] ERROR: client_write: sending to fd 13 failed: Broken pipe
juin 17 23:25:31 scapa usbmuxd[50440]: [23:25:31.467][1] ERROR: client_write: sending to fd 13 failed: Broken pipe
juin 17 23:25:31 scapa usbmuxd[50440]: [23:25:31.467][1] connection_teardown: aborting buffer flush to client after error.
```
And here's the log from idevicesyslog:
```
Jun 17 23:35:32 BackupAgent2(MobileDeviceLink)[1138] <Notice>: 1138:1876258816|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketSend (data): error 32: Broken pipe
Jun 17 23:35:32 BackupAgent2(MobileDeviceLink)[1138] <Notice>: 1138:1876258816|StreamHandler.c:_WriteFunc| ERROR: Error sending data: Can't send data
Jun 17 23:35:32 BackupAgent2(MobileDeviceLink)[1138] <Notice>: 1138:1876258816|DeviceLinkConnection.c:_DLSendBytesAndDisconnectOnError| ERROR: Could not send data: 'Error writing data'
Jun 17 23:35:32 BackupAgent2(liblockdown.dylib)[1138] <Notice>: lockconn_disable_ssl: <private>
Jun 17 23:35:32 BackupAgent2[1138] <Error>: Computer disconnected: Lost connection

Downgrading to usbmuxd from 20191130 (9af2b12) fixed the issue.

nikias, 2020-06-17:
Since usbmuxd didn't really get any internal changes after 9af2b12 it seems more like an issue with one of the libraries which is either libusbmuxd or libimobiledevice. So far it worked for me, but let me try this again on a Linux box.

nikias, 2020-06-18:
I just tested on a Linux system with all the latest and it does not fail...

corsac-s, 2020-06-18:
I only downgraded usbmuxd to successfully backup, so I think it's really in usbmuxd. Unfortunately, after managing a successful backup (which was quite large) the later backups (which are lower in size I guess) all succeeds, whether from 9af2b12 or from 1.1.1.

I'll try to find some space to make a complete backup with 1.1.1 and see if I can reproduce (and then maybe bisect).

mexmer, 2020-06-18:
juin 17 23:25:31 scapa usbmuxd[50440]: [23:25:31.467][1] ERROR: client_write: sending to fd 13 failed: Broken pipe

this suggest issue with communication. so might have been just hicup on usb.
unless it was repeatable situation.

from my experience, sometimes when backup fails (even using itunes on pc/mac), it's best to either replug phone, or restart phone.

corsac-s, 2020-06-18:
It was definitely repeatable back then, but now it isn't. I've tried a full backup with 1.1.1 which went fine. I'm assuming this was a (reproducible) hiccup, so I'm closing this and will reopen if I manage to reproduce again.

closed 2020-06-18 by corsac-s

nikias, 2020-06-18:
Actually I remembered something. Is libimobiledevice linked against OpenSSL or GnuTLS? There was a change in libusbmuxd (non-blocking sockets) and a few OpenSSL-specific adaptions in libimobiledevice, so it might  be that some cases are not correctly handled in case GnuTLS is being used. I performed a full ~160GB backup with OpenSSL-linked libimobiledevice successfully.

corsac-s, 2020-06-19:
Indeed, due to the OpenSSL licensing exception we build everything against GnuTLS in Debian, so it might be related. The full back went fine (on a 128G device) but I'll let you know if it happens again.