libimobiledevice/usbmuxd #146: if no network device, first power on, connect very slow

yeahtoo, 2020-07-09:
Hi, my application use "usbmuxd_get_device_list" to get the device counts, if counts > 0 will connect the device. And i found the first power on(has no avliable network device), it almous wait 20~80 seconds to be connect successful,but the phone is always pluged on the hardware, and if i use plug a usb wifi or usb phy device, first power on, the connection will be very faster. and after first power on connect successful, i plug out and plug in the iphone, the connection is normal(not slow), if has network device or not.
here is my application log:
```
Jul  7 12:00:04 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:05 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:06 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:07 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:08 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:09 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:10 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:11 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:12 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:13 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:14 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:15 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:16 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:17 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:18 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:19 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:20 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:21 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:22 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :0
Jul  7 12:00:23 (coco) user.debug syslog: [phone.c main_loop 244] usbmuxd_get_device_list :1
Jul  7 12:00:23 (coco) user.info syslog: [phone.c main_loop 276] connect port[62222] success!
Jul  7 12:00:23 (coco) user.info syslog: [phone.c main_loop 279] iphone connect success
```

yeahtoo, 2020-07-09:
here is usbmuxd log:
[12:00:03.061][3] usbmuxd v1.1.1 starting up
[12:00:03.064][4] Creating socket
[12:00:03.066][5] Initialized config_dir to /var/lib/lockdown
[12:00:03.067][5] client_init
[12:00:03.068][5] device_init
[12:00:03.068][4] Initializing USB
[12:00:03.069][3] Using libusb 1.0.23
[12:00:03.077][4] Registering for libusb hotplug events
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.008232] [000003d4] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x31388 with handle 1
[ 0.008299] [000003d4] libusb: debug [libusb_get_device_list] 
[ 0.008379] [000003d4] libusb: debug [libusb_get_device_descriptor] 
[12:00:03.078][4] Found new device with v/p 05ac:12a8 at 1-3
[ 0.008435] [000003d4] libusb: debug [libusb_open] open 1.3
[ 0.008561] [000003d4] libusb: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.008593] [000003d4] libusb: debug [libusb_get_configuration] 
[ 0.009264] [000003d4] libusb: debug [libusb_get_configuration] active config 1
[ 0.009524] [000003d4] libusb: debug [libusb_kernel_driver_active] interface 0
[12:00:03.079][4] Setting configuration for device 1-3, from 1 to 4
[ 0.009618] [000003d4] libusb: debug [libusb_set_configuration] configuration 4
[ 0.012591] [000003dd] libusb: debug [linux_netlink_event_thread_main] netlink event thread entering
[12:00:03.154][4] Found interface 1 with endpoints 04/85 for device 1-3
[ 0.084413] [000003d4] libusb: debug [libusb_claim_interface] interface 1
[ 0.084564] [000003d4] libusb: debug [libusb_alloc_transfer] transfer 0x31808
[12:00:03.154][4] Using wMaxPacketSize=512 for device 1-3
[12:00:03.154][4] USB Speed is 480 MBit/s for device 1-3
[ 0.085459] [000003d4] libusb: debug [libusb_submit_transfer] transfer 0x31808
[ 0.085501] [000003d4] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[12:00:03.155][4] 1 device detected
[12:00:03.155][3] Initialization complete
[ 0.086685] [000003d4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.087083] [000003d4] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.087144] [000003d4] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[ 0.087171] [000003d4] libusb: debug [handle_events] poll() returned 1
[ 0.087200] [000003d4] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.087221] [000003d4] libusb: debug [handle_control_completion] handling completion status 0
[ 0.087242] [000003d4] libusb: debug [disarm_timerfd] 
[ 0.087270] [000003d4] libusb: debug [usbi_handle_transfer_completion] transfer 0x31808 has callback 0x183e8
[12:00:03.157][4] Got lang ID 1033 for device 1-3
[ 0.087333] [000003d4] libusb: debug [libusb_submit_transfer] transfer 0x31808
[ 0.087352] [000003d4] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.087910] [000003d4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.087948] [000003d4] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[ 0.087970] [000003d4] libusb: debug [handle_events] poll() returned 1
[ 0.087999] [000003d4] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=82
[ 0.088018] [000003d4] libusb: debug [handle_control_completion] handling completion status 0
[ 0.088038] [000003d4] libusb: debug [disarm_timerfd] 
[ 0.088063] [000003d4] libusb: debug [usbi_handle_transfer_completion] transfer 0x31808 has callback 0x19274
[12:00:03.157][4] Got serial '00008020000551CE1A9A002E' for device 1-3
[ 0.088130] [000003d4] libusb: debug [libusb_free_transfer] transfer 0x31808
[12:00:03.157][3] Connecting to new device on location 0x10003 as ID 1
[12:00:03.158][6] send_packet(1, 0x0, 0xbede5994, (nil), 0)
[ 0.088309] [000003d4] libusb: debug [libusb_alloc_transfer] transfer 0x31530
[ 0.088329] [000003d4] libusb: debug [libusb_submit_transfer] transfer 0x31530
[ 0.088349] [000003d4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 20
[ 0.088452] [000003d4] libusb: debug [libusb_alloc_transfer] transfer 0x315e8
[ 0.088602] [000003d4] libusb: debug [libusb_submit_transfer] transfer 0x315e8
[ 0.088633] [000003d4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 16384
[ 0.088730] [000003d4] libusb: debug [libusb_alloc_transfer] transfer 0x31808
[ 0.088772] [000003d4] libusb: debug [libusb_submit_transfer] transfer 0x31808
[ 0.088793] [000003d4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 16384
[ 0.088844] [000003d4] libusb: debug [libusb_alloc_transfer] transfer 0x31890
[ 0.088880] [000003d4] libusb: debug [libusb_submit_transfer] transfer 0x31890
[ 0.088900] [000003d4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 16384
[12:00:03.158][5] All 3 RX loops started successfully
[ 0.088990] [000003d4] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=20
[ 0.089012] [000003d4] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.089032] [000003d4] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[... 604 lines cut ...]
[12:00:23.677][5] [OUT] dev=1 sport=2 dport=62222 seq=1 ack=1 flags=0x10 window=131072[512] len=0
[12:00:23.677][6] send_packet(1, 0x6, 0xbede594c, (nil), 0)
[20.607396] [000003d4] libusb: debug [libusb_alloc_transfer] transfer 0x321b8
[... cut ...]
[43.433126] [000003dd] libusb: debug [linux_netlink_read_message] netlink hotplug found device busnum: 1, devaddr: 4, sys_name: 1-1.1, removed: no
[43.433175] [000003dd] libusb: debug [linux_enumerate_device] busnum 1 devaddr 4 session_id 260
[43.433198] [000003dd] libusb: debug [linux_enumerate_device] allocating new device for 1/4 (session 260)
[43.433496] [000003dd] libusb: debug [linux_get_parent_info] Dev 0x313a8 (1-1.1) has parent 0x32768 (1-1) port 1
[43.435262] [000003d4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling

yeahtoo, 2020-07-09:
the usbmuxd service is always running after boot up, not use the udev hotplug, before my application,the startup scripts is this:
```
usbmuxd &
cocoapp &
monitor &
```
Please help, thanks!

yeahtoo, 2020-07-16:
ok, at last i found the problem is in openssl, if no mac address the system entropy is generate very slow, so the program is block in openssl initial,so change the /dev/random to /dev/urandom,here is my modify in openssl:
diff --git a/crypto/rand/rand_unix.c b/crypto/rand/rand_unix.c
```
index fe457ca..7059e94 100644
--- a/crypto/rand/rand_unix.c
+++ b/crypto/rand/rand_unix.c
@@ -633,7 +633,9 @@ size_t rand_pool_acquire_entropy(RAND_POOL *pool)
         bytes_needed = rand_pool_bytes_needed(pool, 1 /*entropy_factor*/);
         while (bytes_needed != 0 && attempts-- > 0) {
             buffer = rand_pool_add_begin(pool, bytes_needed);
-            bytes = syscall_random(buffer, bytes_needed);
+            memset(buffer, 0x26, bytes_needed);
+            bytes = bytes_needed;
+            // bytes = syscall_random(buffer, bytes_needed);
             if (bytes > 0) {
                 rand_pool_add_end(pool, bytes, 8 * bytes);
                 bytes_needed -= bytes;
diff --git a/e_os.h b/e_os.h
index 34223a0..94df2af 100644
--- a/e_os.h
+++ b/e_os.h
@@ -30,7 +30,7 @@
 #  define DEVRANDOM "/dev/urandom", "/dev/random", "/dev/hwrng", "/dev/srandom"
 #  if defined(__linux) && !defined(__ANDROID__)
 #   ifndef DEVRANDOM_WAIT
-#    define DEVRANDOM_WAIT   "/dev/random"
+#    define DEVRANDOM_WAIT   "/dev/urandom"
 #   endif
 /*
  * Linux kernels 4.8 and later changes how their random device works and there

```

mexmer, 2020-07-16:
what you did is removing entropy from ssl generation, so it will have same initialization everytime, which is bad, because you need to catch communication only once, and then you will be able to decrypt it always.

first of, since this issue is with openssl, send it to openssl authors, second ... you really should not use openssl library patched with your "fix" if you care about your system security.

yeahtoo, 2020-07-17:
> what you did is removing entropy from ssl generation, so it will have same initialization everytime, which is bad, because you need to catch communication only once, and then you will be able to [...]

ok, thanks for your advise