libimobiledevice/usbmuxd #147: iphone times out during restore. usbmuxd shows timeout err.

TheNegativeZero, 2020-07-10:
This is related to issue #338 reported on idevicerestore but the source of the problem seems to be on usbmuxd or libusbmuxd.

After reviewing all compiles, links, libs, dependencies, etc., idevicerestore gets a timeout after  successfully sending NORData: 

idevice.c:642 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -104 (Connection reset by peer).

After tracking down the issue in libimobiledevice programs and running usbmuxd -f -vv, I see that usbmuxd 1.1.2 (libusb.1.0.23) shows error -71 as a result of a timeout and disconnects.  This happens after a long conversation and transfers between host and device:

[88.808281] [0000cff5] libusb: debug [usbi_handle_transfer_completion] transfer 0x55c0401e9a10 has callback 0x55c03f5a4760
[88.808286] [0000cff5] libusb: debug [libusb_free_transfer] transfer 0x55c0401e9a10
[88.990091] [0000cff5] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[88.990110] [0000cff5] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[88.990119] [0000cff5] libusb: debug [handle_events] poll() returned 1
[88.990129] [0000cff5] libusb: debug [reap_for_handle] urb type=3 status=-71 transferred=0
[88.990135] [0000cff5] libusb: debug [handle_bulk_completion] handling completion status -71 of bulk urb 1/1
[88.990143] [0000cff5] libusb: debug [handle_bulk_completion] low level error -71
[88.990149] [0000cff5] libusb: debug [usbi_handle_transfer_completion] transfer 0x55c0401fc670 has callback 0x55c03f5a48b0
[13:38:08.103][4] Device 1-37 RX aborted due to error or disconnect

I have tried using different systems (Ubuntu 20.04 and 18.04) and cables and I still cannot find the reason why the host drops the device.  After reading libusbmuxd and libimobiledevice programs it is unclear whether the host reconnects to the device before sending data at some point.

I am also attaching a couple of  txt files with the console output of "idevicerestore -l -d" (libimobiledevice compiled with --enable-debug) and "usbmuxd -f -vv" which runs on a separate console terminal.

BTW, all ilibmobiledevice's supporting libraries are git-cloned, compiled, and properly linked.

Any hint is highly appreciated.  Best rgds.

[usbmuxd.txt](https://github.com/libimobiledevice/usbmuxd/files/4905378/usbmuxd.txt)
[idevicerestore.txt](https://github.com/libimobiledevice/usbmuxd/files/4905404/idevicerestore.txt)

TheNegativeZero, 2020-08-07:
I have noticed something odd.  I hope someone might add some thoughts about this.
I am posting on this forum as the problem is not related to idevicestore but to the handling of usbmuxd (or libusb) of the process.

Three terminal windows:
Term 1: sudo usbmuxd -f -vv
Term 2: lsusb -d 05ac:1281.  Here I can see the Apple Iphone device registered on Bus 001 Device 0XX.  Manually update frequently during execution of Term 3.
Term 3: sudo idevicerestore -l -d

During idevicerestore run, I can see that the program sets the Iphone in Recovery mode.  Iphone desconnects, lsusb loses track of it; and in a few seconds reconnects with a different Bus/Device ID.  That is fine; Iphone just rebooted into Recovery mode.  Idevicerestore sees the iPhone after rebooting and continues execution.  LSUSB gets the iPhone with a new Device ID.  No problem here.

Further down the idevicerestore execution, the program sets the iPhone into Restore mode.  Phone reboots and enters the screen with the progress bar which does not move at all.  LSUSB in Term 2 does record that iphone reboots (disappears) but never registers back.  After reboot the iPhone disappears to never appear again.

As Iphone never registers back, usbmuxd/libusb processes TIMEOUT and eventually idevicerestore does as well with all subsequent program exceptions related to the timeout handling.

So question is: although iPhone gets the "reboot in restore mode" instruction and does so, why it fails to reconnect to the system after reboot?  (UBUNTU 20.04, all libimobiledevice and related libs updated to date).  It has already rebooted once into Recovery mode and processed all data transfers correctly.  However it fails to reconnect the second time.

Interesting that the iPhone does enter Restore mode with the progress bar halted.  Everything times out as LSUSB never gets the rebooted phone.  After idevicerestore finished unsuccessful execution, iPhone also processes timeout, reboots, goes back to the iTunes+usb cable image and reappears in LSUSB command (term 2).

Any thoughts are much appreciated.  Thanks.

nikias, 2020-08-07:
So the device doesn't show up when in restore mode? If lsusb doesn't show the device then I don't know what's going on. Does the kernel print any usb errors?

TheNegativeZero, 2020-08-08:
Thanks @nikias , I do not mean to be repetitive:

Before starting idevicerestore -l -d:
- lsusb shows iphone: Bus 001 Device 011: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]

During idevicerestore execution: iPhone enters in Recovery mode:
- iPhone no longer shown in lsusb command
- after a few seconds, iPhone is shown again
- Bus 001 Device 012: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]
- idevicerestore execution continues.

During idevicerestore execution: iPhone enters in Restore mode:
- iPhone not shown in lsusb command.  Never shows up again
- iPhone does enter in Restore mode and shows static Progress Bar.
- usbmuxd reports timeout.  idevicerestore reports timeout.  iPhone is not reported as connected to the system.
- after idevicerestore finishes execution (unsuccessfully), iPhone reboots and it shown again in lsusb command:
  Bus 001 Device 014: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]

dmesg | grep -i USB does not show anything abnormal.  I guess this is because at this point the iPhone is connected and registered.  Same for usb-devices.

Thanks again.

TheNegativeZero, 2020-08-09:
I just found out that when idevicerestore and support programs set the iPhone to Restore Mode, the ID in LSUSB changes to:
Bus 001 Device 0YY: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6, 
from
Bus 001 Device 0XX: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode].

Anyhow, USBMUXD/LIBUSB times out after this iphone reboot to Restore Mode and execution finishes.
I wouldnt expect that this change on iPhone ID has any effect on the execution of the program, unless it never enters in Restore mode.  Will take a look at the code.

Thanks.

nikias, 2020-08-09:
The change of the USB ID is expected and has always been like that. Recovery mode is handled by libirecovery and restore mode by usbmuxd.

TheNegativeZero, 2020-08-12:
Thanks @nikias , I reached the same conclusion.  

Everything works well up to the point when the program flow reaches the "waiting for the device to enter Restore mode".  Iphone reboots to Restore mode, changes USB ID (and the actual lsusb device ID), then usbmuxd drops it.  Program flow processes Time Out; idevicerestore terminates.

I tried downgrading to usbmuxd (1.1.1) and libusbmuxd (2.0.1).  No luck.

Same as before, iPhone gets to Restore Mode with the Progress Bar remaining static, after a while the phone also times out and reboots to the iTunes Restore screen (cable and iTunes logo).

What bugs me is that apparently USBMUXD never really connects while the previous step, using the libirecovery library does interact with the phone fine.

I want to rule this out as a hardware issue but the fact that irecovery and the first steps of idevicerestore both work well make me believe that there is an issue with usbmuxd / libusbmuxd.

Any hint is appreciated.  Thanks.

nikias, 2020-08-12:
Well, does usbmuxd work with a device in normal mode?

TheNegativeZero, 2020-08-13:
I don't have more iOS devices available.  All tests are done with this single broken iphone.

nikias, 2020-08-13:
Well then it could very well be the "broken" attribute of your iPhone, or your USB controller, or USB hub (some iPhones don't like some USB hubs!). With millions of other people using this software without the issues you mention it is unlikely that there's something wrong with usbmuxd.

TheNegativeZero, 2020-08-13:
Thanks for your timely feedback @nikias,  Appreciated.

I will put this resurrection project on the shelf for the time being and will give it another shot when time permits.

I am not using any external USB hub, and when tested, there are no other USB devices connected.  My system is an Intel-i7, 2.6-GHz,16GB, with 3 USB3 ports.  I shall try with a different system.

Also, there are a number of open tickets in the idevicerestore forum that present issues with timeouts after Sending NORData, failing to recover FDR data, unable to connect to FDR client, unable to receive message from FDR, or simple FDR read timeouts which all relate to the iPhone disappearing when USBMUXD tries to connect.  It could also all be broken phones beyond repair as well since usbmuxd is used by millions of other people.

Thanks.

nikias, 2020-08-13:
FYI most of the "FDR timeout" is just incorrectly reported as an error while it's just a connection reset (i.e. end of connection). There are many reasons why things can fail; on my mac for example I had abort issues because the usb driver was messed up (because I did something unrelated that just messes things up by design) but then after reboot it worked fine. So it is really hard to tell what is going on...

TheNegativeZero, 2020-08-29:
OK, time to close this issue unless anyone has something to say.

Tried using another iPhone, same cable.  Everything seems to connect fine.
However, the iPhone6 in recovery bootloop (iTunes screen) cannot be seen by usbmuxd.  Played around with udev rules and some programming to no avail.

Usbmuxd -f -vv spits the following when the bootlooped iPhone6 is connected:

[341.532597] [00034875] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[341.532684] [00034875] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[341.532724] [00034875] libusb: debug [handle_events] poll() returned 0
[341.618343] [00034876] libusb: debug [linux_get_device_address] getting address for device: 1-3 detached: 0
[341.618362] [00034876] libusb: debug [linux_get_device_address] scan 1-3
[341.618416] [00034876] libusb: debug [linux_get_device_address] bus=1 dev=29
[341.618422] [00034876] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[341.618427] [00034876] libusb: debug [linux_enumerate_device] busnum 1 devaddr 29 session_id 285
[341.618435] [00034876] libusb: debug [linux_enumerate_device] allocating new device for 1/29 (session 285)
[341.618476] [00034876] libusb: debug [linux_get_parent_info] Dev 0x7f0e980034b0 (1-3) has parent 0x55d658d54a80 (usb1) port 3
[341.618522] [00034875] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[341.618529] [00034875] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[341.618536] [00034875] libusb: debug [handle_events] poll() returned 1
[341.618541] [00034875] libusb: debug [handle_events] caught a fish on the event pipe
[341.618545] [00034875] libusb: debug [handle_events] hotplug message received
[341.618554] [00034875] libusb: debug [libusb_get_device_descriptor] 
[341.641371] [00034876] libusb: debug [linux_get_device_address] getting address for device: 1-3 detached: 0
[341.641384] [00034876] libusb: debug [linux_get_device_address] scan 1-3
[341.641435] [00034876] libusb: debug [linux_get_device_address] bus=1 dev=29
**[341.641443] [00034876] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.**
**[341.641447] [00034876] libusb: error [udev_hotplug_event] ignoring udev action bind**

USBMUXD kind of detects the iPhone plugged-in but cannot "bind" it.  Iphones 4, 5 and iPod 3rd Gen are detected fine.  IPhone 5 using same cable.  Ubuntu 20.04, latest compiles of libimobiledevice and related and pre-requisite libs.

Any hints are much appreciated before I shelf this.

Thank.

closed 2020-09-06 by TheNegativeZero