libimobiledevice/usbmuxd #150: Could it be used as localabstract in ADB

airhandsome, 2020-08-05:
We usually use adb forward  localabstract in Android Devices. In some case, we cannot get the accurate port in iphone. There I want to known is there any method like localabstract used in iproxy?

zhoujunhe, 2021-05-20:
iproxy  5000   10000

airhandsome, 2021-05-20:
> iproxy 5000 10000

thanks for your reply. This command means map port 5000 in ios device  to port 10000 in the windows or mac. It is ok. I want to know that is there any method to map the accurate port to unix socket?  like adb forward localabstract:<unix domain socket name>?

zhoujunhe, 2021-05-26:
端口映射成unix socket  可以通过 nginx 反向代理， iproxy 5000 10000  是把设备10000端口映射成本地电脑5000，然后在通过nginx 把 5000端口映射成 unix socket

airhandsome, 2021-05-26:
> 端口映射成unix socket 可以通过 nginx 反向代理， iproxy 5000 10000 是把设备10000端口映射成本地电脑5000，然后在通过nginx 把 5000端口映射成 unix socket

Emmm谢谢，是这个意思，就是方向反了，我在ios里的应用开启了一个unix socket， 但是我想直接用iproxy 把PC的Port 映射到 Ios内部的unix socket。

zhoujunhe, 2021-05-26:
目前这个库不支持，如果支持，我看了一下，改动量还是麻大的。

airhandsome, 2021-05-26:
> 目前这个库不支持，如果支持，我看了一下，改动量还是麻大的。

了解了。谢谢。

closed 2021-05-26 by airhandsome