libimobiledevice/usbmuxd #155: configure: error: libpthread is required to build usbmuxd

XanaduAPI, 2020-11-30:
In the process of compiling Android version, I encountered the following error. How can I solve it?

Compilation parameters:./autogen.sh --host=arm-linux-androideabi --prefix=$ANDROID_ARM_32

ANDROID_ARM_32=$ANDROID_ARM_32/ARM_32
$NDK_HOME/build/tools/make-standalone-toolchain.sh --platform=android-24 --arch=arm --install-dir=$ANDROID_ARM_32/ARM_32

XanaduAPI, 2020-11-30:
[image]

nikias, 2020-11-30:
I think on Android, you don't need to pass -lpthread so you can disable this check in configure.ac:
https://github.com/libimobiledevice/usbmuxd/blob/63d1164e736d7419198d1b737d10a9aae85bef98/configure.ac#L22

XanaduAPI, 2020-11-30:
@nikias Thank you very much. I just compiled it successfully.

XanaduAPI, 2020-11-30:
@nikias  
[image]

The above error occurred later.

I'm blocking the

//#define malloc rpl_ malloc

//#define realloc rpl_ realloc

And it worked

closed 2020-12-01 by XanaduAPI