libimobiledevice/usbmuxd #156: free(): invalid pointer

ekristen, 2020-12-07:
I am occasionally seeing `free(): invalid pointer` on my usbmuxd instances. I only have a single device per usbmuxd.

Possibly related https://github.com/libimobiledevice/usbmuxd/commit/a1963ec3717e1e652de9d10311077a454d9bb8e9

It appears to happen in the client_close call chain and it seems to only happen when the client died mid-connect and the connection gets aborted.

I think that while writing this issue up and looking at the source it might have already been fixed? I just realized my version is from Jun 15, and looks like there was some fixes around client_close in the last commit.

@nikias perhaps you can comment/confirm? 

Example 1
```
[05:08:09.253][4] Disconnecting client fd 15
[05:08:27.089][4] New client on fd 15 (pid 0)
[05:08:34.830][4] Client 15 connection closed
[05:08:34.970][4] Disconnecting client fd 15
[05:08:34.970][4] Client died mid-connect, aborting device 8 connection
[05:08:34.970][4] Disconnecting client fd 15
free(): invalid pointer
```

Example 2
```
[05:58:11.762][4] Client 21 is going to be disconnected: /proc/0/cmdline[0]
[06:10:34.244][4] Client 15 accepted: /proc/0/cmdline[0]
[06:11:02.018][4] Client 15 connection closed
[06:11:02.717][4] Client 15 is going to be disconnected: /proc/0/cmdline[0]
[06:11:13.212][4] Client 15 accepted: /proc/0/cmdline[0]
[06:11:17.839][4] Client connected to device 2 (14105->62078)
[06:11:20.459][1] Sending to client fd 15 failed: -1 Broken pipe
[06:11:20.464][4] Client 15 is going to be disconnected: /proc/0/cmdline[0]
[06:11:20.465][4] Client died mid-connect, aborting device 2 connection
[06:11:20.465][4] Client 15 is going to be disconnected: /proc/0/cmdline[0]
free(): invalid pointer
```

Example 3
```
[09:20:58.661][4] Client 15 is going to be disconnected: /proc/0/cmdline[0]
[09:21:05.987][4] Client 15 accepted: /proc/0/cmdline[0]
[09:21:58.412][4] Client connected to device 1 (25686->62078)
[09:22:26.405][4] Client 15 connection closed
[09:22:53.791][4] Client 15 is going to be disconnected: /proc/0/cmdline[0]
[09:22:58.049][4] Client died mid-connect, aborting device 1 connection
[09:23:12.256][4] Client 15 is going to be disconnected: /proc/0/cmdline[0]
free(): invalid pointer
```

nikias, 2020-12-07:
Yes this is what should have been fixed with this commit:
https://github.com/libimobiledevice/usbmuxd/commit/63d1164e736d7419198d1b737d10a9aae85bef98
At least I had exactly the same scenario where it crashed after "Client died mid-connect" and then I came up with the above commit.

ekristen, 2020-12-07:
Awesome thanks for the confirmation.

closed 2020-12-07 by ekristen