libimobiledevice/usbmuxd #158: Device not recognised while in DFU

jenkshields, 2021-01-07:
Kia ora!

I’m working with an old iPod Touch, following a set of instructions that require me to run some commands while the device is in DFU mode. Unfortunately, attempting to ssh to the device while it’s in DFU gives me an error message saying no device has been found. Is this functionality supported, or is there something specific I need to be doing?

thank you!

demhademha, 2021-01-07:
What is the output of "irecovery -v -q"?

nikias, 2021-01-07:
You can't SSH into a device in DFU or Recovery mode. Sounds to me like you are trying to boot special ram disk with an ssh daemon on it. But without any details (what instructions?) nobody really knows how to help..