libimobiledevice/usbmuxd #162: 1.1.1: preflight_worker_handle_device_add: ERROR

adrianaryaputra, 2021-03-11:
hello, i've got following error when connecting my phone to PC.

```
sudo usbmuxd -fv                                                         
[13:37:46.097][3] usbmuxd v1.1.1 starting up
[13:37:46.097][4] Creating socket
[13:37:46.097][4] Initializing USB
[13:37:46.106][4] Registering for libusb hotplug events
[13:37:46.106][4] 0 devices detected
[13:37:46.106][3] Initialization complete
[13:38:13.918][4] Found new device with v/p 05ac:12a8 at 1-56
[13:38:13.918][3] Could not get old configuration descriptor for device 1-56: -5
[13:38:13.918][4] Setting configuration for device 1-56, from 0 to 4
[13:38:13.957][4] Found interface 1 with endpoints 04/85 for device 1-56
[13:38:13.958][4] Using wMaxPacketSize=512 for device 1-56
[13:38:13.958][4] USB Speed is 480 MBit/s for device 1-56
[13:38:13.958][4] Got lang ID 1033 for device 1-56
[13:38:13.959][4] Got serial 'd4545c5642b1baf4f75c69057fc4ad64c8e83ad6' for device 1-56
[13:38:13.959][3] Connecting to new device on location 0x10038 as ID 1
[13:38:13.959][3] Connected to v2.0 device 1 on location 0x10038 with serial number d4545c5642b1baf4f75c69057fc4ad64c8e83ad6
[13:38:13.959][4] preflight_worker_handle_device_add: Starting preflight on device d4545c5642b1baf4f75c69057fc4ad64c8e83ad6...
[13:38:13.959][4] New client on fd 13 (self)
[13:38:23.969][1] preflight_worker_handle_device_add: ERROR: Could not get lockdownd type from device d4545c5642b1baf4f75c69057fc4ad64c8e83ad6, lockdown error -7
[13:38:23.970][4] Disconnecting client fd 13
```

am i missing something?

nikias, 2021-03-11:
hmm it times out after 10 seconds, does it happen consistently when you replug the device a few times?

mexmer, 2021-03-11:
looks like usb communication issue, can be hardware problem (usb cable), or might be some hardlock inside phone USB stack (in that case it will require phone restart). first i would try to connect device to different computer with same cable, or connect to same computer but with different cable.

closed 2021-03-12 by adrianaryaputra