libimobiledevice/usbmuxd #163: Unable to connect device (iPhone 7)

cooperbang, 2021-03-19:
Sometimes my iPhone 7 is not recognized by usbmuxd.

```
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.120][3] Using libusb 1.0.24
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: libusb: warning [op_get_configuration] device unconfigured
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: libusb: error [op_get_active_config_descriptor] device unconfigured
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.126][3] Could not get old configuration descriptor for device 1-4: LIBUSB_ERROR_NOT_FOUND
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.128][3] Initialization complete
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.128][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.128][3] Connecting to new device on location 0x10004 as ID 1
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.214][3] Removed device 1 on location 0x10004
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: [08:11:50.214][2] Cannot find device entry while removing USB device 0x55dxxxxxxxxx on location 0x10004
Mär 19 08:11:50 localhost.localdomain usbmuxd[4297]: libusb: error [udev_hotplug_event] ignoring udev action unbind
```
After replugging the device I get:
```
Mär 19 08:17:39 localhost.localdomain usbmuxd[4297]: libusb: warning [op_get_configuration] device unconfigured
Mär 19 08:17:39 localhost.localdomain usbmuxd[4297]: libusb: error [op_get_active_config_descriptor] device unconfigured
Mär 19 08:17:39 localhost.localdomain usbmuxd[4297]: [08:17:39.342][3] Could not get old configuration descriptor for device 1-5: LIBUSB_ERROR_NOT_FOUND
Mär 19 08:17:39 localhost.localdomain usbmuxd[4297]: [08:17:39.358][3] Connecting to new device on location 0x10005 as ID 2
Mär 19 08:17:39 localhost.localdomain usbmuxd[4297]: [08:17:39.394][3] Connected to v2.0 device 2 on location 0x10005 with serial number 5c000xxxxxxxxxxxxxxxxxxxxxxxxxxxx>
Mär 19 08:17:39 localhost.localdomain usbmuxd[4297]: [08:17:39.395][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device 5c000xxxxxxxxx>
```
Workaround:  `systemctl stop usbmuxd` and then `systemctl start usbmuxd` - after some time it will connect...

cooperbang, 2021-03-26:
Additional info: The problem seems to occur always after installing a new kernel update.

After a kernel update the device isn´t recognized by usbmuxd.
Some seconds afterh replugging the usb cable the ""Trust this computer?" messages pops up on the phone.

It doesn help to click "Trust" on the phone - you additionally need to restart usbmuxd to get the phone recognized by usbmuxd (see workaround above).

maravtdm, 2021-12-17:
FYI
I have the same issue, whatever the kernel
I have to kill usbmuxd and restart it

To solve this, I compile my own kernel with :
`CONFIG_APPLE_MFI_FASTCHARGE=y `

in : Device Drivers / USB Support / Fast charge control for iOS devices