libimobiledevice/usbmuxd #168: Create new release

SuperSandro2000, 2021-06-14:
It has been a while since the last one. I think a 1.1.2 wouldn't hurt :)

mieszko4, 2022-06-12:
@nikias @FunkyM Any plans to cut a release? :pray:  I think it would resolve some of the issues that people have with newer iOS.
I would try to do that myself but I am not sure where to start :)

muvaf, 2024-10-16:
Any plans to cut a new release? Looks like the last release was 4 years ago and there have been quite a few changes since then that forces us to build from tip of the main branch.

WhyNotHugo, 2024-12-22:
@nikias can we release a new version?

Most devices and iOS versions from the last few years need changes from master, but downstream won't package unreleased versions. As a result, the only way to use this package is for end-users to build from source.