libimobiledevice/usbmuxd #171: build on mac using cmake Mention Error:Conflicting types for '__builtin___stpcpy_chk'

ShiverZm, 2021-06-23:
[env] macos 14.1 cmake 3.10 xcode 10.1 

[image]

nikias, 2021-06-23:
1. usbmuxd is supposed to be built with autotools, not cmake or Xcode, see README.md
2. The error you are seeing is because of missing configure step
3. macOS ships with its own official usbmuxd, so unless you know what you're doing you don't need to built the open source one

ShiverZm, 2021-06-23:
Thanks For Your Reply Very Much!
first, i was sorry i ignored the configure,and it is pity i do know little about autotool,and i just need the project for cross platform,because  i used usbmuxd-vs for windows,and now i need used  it on Mac.
Thank you again For your Patience.
And after i watch the configure again, 

> # Checks for header files.
>AC_HEADER_STDC
>AC_CHECK_HEADERS([stdint.h stdlib.h string.h])
>
># Checks for typedefs, structures, and compiler characteristics.
>AC_C_CONST
>AC_TYPE_SIZE_T
>AC_TYPE_SSIZE_T
>AC_TYPE_UINT16_T
>AC_TYPE_UINT32_T
>AC_TYPE_UINT8_T
>
># Check if clock_gettime requires -lrt (old GNU systems)
>AC_SEARCH_LIBS([clock_gettime],[rt posix4])
>
># Checks for library functions.
>AC_CHECK_FUNCS([strcasecmp strdup strerror strndup stpcpy malloc realloc])
>AC_CHECK_FUNCS([ppoll clock_gettime localtime_r])

you just use string.h which is from c++ std? and no AC_define HAVE_STPCPY?

nikias, 2021-06-23:
Just follow my advice. You do NOT need to build usbmuxd on macOS, it is already there. So please just don't try to build it. In fact, if you build your own, you will have problems later on since the official usbmuxd is already running.
Also, Windows has usbmuxd too, just install the AppleMobileDevice drivers. No need to build it.

ShiverZm, 2021-06-24:
i have to complie it and modify this ,because i need receive data from IPhone by Usb similarly like PeerTalk and on Windows I made it and modify usbmuxd-vs.

ShiverZm, 2021-06-24:
i compile finished! the real problem is that i am not Familiar with macos ,there is so many rename problem.