libimobiledevice/usbmuxd #172: Usbmuxd does not work straight away

AryToNeX, 2021-07-08:
In my machine, I have to start usbmuxd twice before it actually works. This is the log I have of the first time it runs (at each reboot this happens).

```
[14:34:07.557][3] usbmuxd v1.1.1 starting up
[14:34:07.557][4] Creating socket
[14:34:07.557][3] Successfully dropped privileges to 'usbmux'
[14:34:07.557][4] Initializing USB
[14:34:07.557][3] Using libusb 1.0.24
[14:34:07.558][4] Registering for libusb hotplug events
[14:34:07.558][4] 0 devices detected
[14:34:07.558][3] Initialization complete
[14:34:07.558][4] Client 9 accepted: /usr/lib/org_kde_powerdevil[3848]
[14:34:07.558][4] Client 10 accepted: /usr/bin/latte-dock[3952]
[14:34:07.558][4] Client 11 accepted: /usr/bin/kded5[3764]
[14:34:16.907][4] Client 12 accepted: /usr/lib/udev/iphone-set-info[20377]
[14:34:16.907][4] Client 12 connection closed
[14:34:16.907][4] Client 12 is going to be disconnected: /usr/lib/udev/iphone-set-info[20377]
[14:34:16.908][4] Caught SIGUSR1/2 but this instance was not started with "--enable-exit", ignoring.
[14:34:16.910][4] Found new device with v/p 05ac:12a8 at 1-3
libusb: error [op_get_configuration] device unconfigured
libusb: error [op_get_active_config_descriptor] device unconfigured
[14:34:16.910][3] Could not get old configuration descriptor for device 1-3: LIBUSB_ERROR_NOT_FOUND
[14:34:16.910][4] Setting configuration for device 1-3, from 0 to 4
[14:34:16.960][4] Found interface 1 with endpoints 04/85 for device 1-3
[14:34:16.960][4] Using wMaxPacketSize=512 for device 1-3
[14:34:16.960][4] USB Speed is 480 MBit/s for device 1-3
[14:34:16.960][4] Got lang ID 1033 for device 1-3
[14:34:16.961][4] Got serial '00008030000579A03EF9402F' for device 1-3
[14:34:16.961][3] Connecting to new device on location 0x10003 as ID 1
[14:34:16.961][3] Connected to v2.0 device 1 on location 0x10003 with serial number 00008030-000579A03EF9402F
[14:34:16.961][4] preflight_worker_handle_device_add: Starting preflight on device 00008030-000579A03EF9402F...
[14:34:16.961][4] Client 14 accepted: usbmuxd[20356]
[14:34:16.961][4] Client connected to device 1 (1->62078)
[14:34:17.050][3] Removed device 1 on location 0x10003
[14:34:17.050][4] Client 14 is going to be disconnected: usbmuxd[20356]
[14:34:17.050][1] preflight_worker_handle_device_add: ERROR: Could not get lockdownd type from device 00008030-000579A03EF9402F, lockdown error -256
[14:34:17.050][2] Cannot find device entry while removing USB device 0x559eb1801e30 on location 0x10003
libusb: error [udev_hotplug_event] ignoring udev action unbind
[14:37:28.889][4] Found new device with v/p 05ac:12a8 at 1-4
libusb: error [op_get_configuration] device unconfigured
libusb: error [op_get_active_config_descriptor] device unconfigured
[14:37:28.889][3] Could not get old configuration descriptor for device 1-4: LIBUSB_ERROR_NOT_FOUND
[14:37:28.889][4] Setting configuration for device 1-4, from 0 to 4
[14:37:28.951][4] Found interface 1 with endpoints 04/85 for device 1-4
[14:37:28.951][4] Using wMaxPacketSize=512 for device 1-4
[14:37:28.951][4] USB Speed is 480 MBit/s for device 1-4
[14:37:28.951][4] Got lang ID 1033 for device 1-4
[14:37:28.952][4] Got serial '00008030000579A03EF9402F' for device 1-4
[14:37:28.952][3] Connecting to new device on location 0x10004 as ID 2
[14:37:28.952][3] Connected to v2.0 device 2 on location 0x10004 with serial number 00008030-000579A03EF9402F
[14:37:28.952][4] preflight_worker_handle_device_add: Starting preflight on device 00008030-000579A03EF9402F...
[14:37:28.952][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device 00008030-000579A03EF9402F, lockdown error -8
```

I run Artix Linux (Arch derivative with the OpenRC init system)

mexmer, 2021-07-11:
you need to compile usbmuxd from source, 1.1.1 is old and will not work with recent iOS. since your device has new UDID format, then it should not be older that 2 or 3 years (i think first model with this format was iPhone XR/XS) you really need latest version of usbmuxd and libimobile, since you most likely running iOS 14.

AryToNeX, 2021-07-12:
> you need to compile usbmuxd from source, 1.1.1 is old and will not work with recent iOS. since your device has new UDID format, then it should not be older that 2 or 3 years (i think first model [...]

Thing is, there aren't newer releases other than 1.1.1 so evereything newer is literally 'compile it from git' kinda attitude

mexmer, 2021-07-14:
@AryToNeX i think you misunderstanding people role here. source code developer is not your distribution package maintainer.
it's up to decision of your distribution maintainer to decide which version will be included in distribution, they might go with stable (yet sometimes old version) or they might decide to get with source code snaphost from any date (although that is unusualy done in experimental branches of linux distribution, some do that even on stable ones).

if you want newer version of package for your distribution, you need to ask package maintainer for distribution you use.

KokoseiJ, 2021-08-15:
Not necessariilly talking about package distribution, the last release in https://github.com/libimobiledevice/usbmuxd/releases also stays in 1.1.1 since june last year. If there was a major change which is now stabilized enough to a point where compiling from git is encouraged, Shouldn't that be tagged and released as a new version?

FedX-sudo, 2021-10-06:
I have run into the same issue, but I am on NixOS, and it is difficult to compile from source (and very much less than ideal) due to the way NixOS handles system services (see [this](https://stackoverflow.com/questions/58243712/how-to-install-systemd-service-on-nixos) for a little more info). Are there plans to release v1.1.2 in the near future?

mr-who2012, 2023-03-05:
It's march 2023, meanwhile and there is still no release for usbmuxd 1.1.2... 

@KokoseiJ: Could you please tell me where you got this version? Pulling from github and compiling usbmuxd gives me version 1.1.1-46-gd0cda19. It just won't detect my ipad6. I'm using newest libusb1.0.26 There is also a usbmuxd2 on github. But this fork doesn't work as well, unfortunately, though it's a bit newer than usbmuxd 1.1.1.

erzakiev, 2024-02-07:
Yes, verbosely specifying this clandestine 1.1.2 version while cloning `git clone https://github.com/libimobiledevice/usbmuxd.git --branch 1.1.2` returns
`fatal: Remote branch 1.1.2 not found in upstream origin` so i'm not sure what's going on

mexmer, 2024-02-07:
> Yes, verbosely specifying this clandestine 1.1.2 version while cloning `git clone https://github.com/libimobiledevice/usbmuxd.git --branch 1.1.2` returns `fatal: Remote branch 1.1.2 not found in upstream origin` so i'm not sure what's going on

did you even check the repo? there is no such branch/tag, you need to checkout HEAD, of if you want specific commit, then pick that commit.

erzakiev, 2024-02-07:
> there is no such branch/tag
exactly, the 1.1.1 is the latest

but you wrote 
> 1.1.1 is old

this doesn't compute

mexmer, 2024-02-07:
tag 1.1.1 is last tag, but was released Jun 15, 2020, therefore is **OLD**.
since there was no newer release/tag, you need to use **HEAD**, eg. current source code (latest commit)

erzakiev, 2024-02-09:
@mexmer good point, it works now, it turns out i was just connecting a wrong device

WhyNotHugo, 2024-12-22:
Indeed, the latest release won't work with recent iPhone or with recent versions of iOS. See: https://github.com/libimobiledevice/usbmuxd/issues/168