libimobiledevice/usbmuxd #175: log info: socket read() return error code:104,  error description: Connection reset by peer. What are the possible reasons for this error?

GavinYu, 2021-08-10:
sender client：iPhone se , ios 14.6
receiver client：SAMSUNG Glalaxy A71 5G, Android 10
iPhone use Replaykit framework implement record the screen mirroring to Android phone, which use USB data line for data communication. after run one hour，it occur this error that error code:104 and 
error description: Connection reset by peer.

GavinYu, 2021-08-10:
@nikias , help me，please. thanks!

nikias, 2021-08-10:
Can you provide some details how you get the screen record data? Is it tunneled over a usbmux connection? Is it a specific service that you connect to?

GavinYu, 2021-11-09:
@nikias ，I upload the log info for the question，could you help me to analyse reason of the question，and give me some opinion to resolve the question？Thanks！

-------------------------------------------------- log info --------------------------------------------
2021-11-09 16:01:35.607 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: send_packet(1, 0x6, 0x7aa85fee10, 0x0, 0)
2021-11-09 16:01:35.607 30594-30984/com.cidana.mirror.receiver D/CiUSBMuxD: usb_send: fd:47, length:36
2021-11-09 16:01:35.607 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: update_connection: sendable 49116, events 5, flags 0
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### poll() returned 1
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### parse fd: 53 event: 0x1 owner: 2
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### poll process out
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### USB Add fd 53
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### fd count is 4
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### poll() returned 1
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### parse fd: 97 event: 0x4 owner: 1
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: client_process in CONNECTED state
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: device_client_process (4)
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: client_write fd 97 buf 0x7a9ac0df40 len 711
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: update_connection: sendable 49116, events 1, flags 1
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: FD_CLIENT out
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### poll process out
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### USB Add fd 53
2021-11-09 16:01:35.607 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### fd count is 4
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/usbhost: USBDEVFS_REAPURB returned 0
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/usbhost: [ urb @0x7aa8645340 status = 0, actual = 512 ]
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/CiUSBMuxD: enter usb_recv_data
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver D/CiUSBMuxD: usb_recv_data: fd:47, index:11 length:512
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: DEVICE device_data_input
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Mux data input for device 0x7abf278740: 0x73423bb0 len 512
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Copied mux data to buffer (size: 512)
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/usbhost: USBDEVFS_REAPURB returned 0
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/usbhost: [ urb @0x7aa86453c0 status = 0, actual = 512 ]
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/CiUSBMuxD: enter usb_recv_data
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver D/CiUSBMuxD: usb_recv_data: fd:47, index:12 length:512
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: DEVICE device_data_input
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Mux data input for device 0x7abf278740: 0x73423dd0 len 512
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Appended mux data to buffer (total size: 1024)
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/usbhost: USBDEVFS_REAPURB returned 0
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/usbhost: [ urb @0x7aa8645440 status = 0, actual = 512 ]
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver I/CiUSBMuxD: enter usb_recv_data
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver D/CiUSBMuxD: usb_recv_data: fd:47, index:13 length:512
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: DEVICE device_data_input
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Mux data input for device 0x7abf278740: 0x7340ff80 len 512
2021-11-09 16:01:35.625 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Appended mux data to buffer (total size: 1536)
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver I/usbhost: USBDEVFS_REAPURB returned 0
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver I/usbhost: [ urb @0x7aa86454c0 status = 0, actual = 512 ]
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver I/CiUSBMuxD: enter usb_recv_data
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver D/CiUSBMuxD: usb_recv_data: fd:47, index:14 length:512
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: DEVICE device_data_input
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Mux data input for device 0x7abf278740: 0x734101a0 len 512
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver V/CiUSBMuxD: Appended mux data to buffer (total size: 2048)
2021-11-09 16:01:35.626 30594-30984/com.cidana.mirror.receiver I/usbhost: USBDEVFS_REAPURB returned 0
[... cut ...]
2021-11-09 16:01:46.807 30594-30991/com.cidana.mirror.receiver V/CiUSBMuxD: ### fd count is 3
2021-11-09 16:01:46.808 1151-1151/? I/TrafficMonitor: expired arrive. level:-1
2021-11-09 16:01:46.808 1151-1151/? I/TrafficMonitor: during_ms:2001,period_ms:2000
2021-11-09 16:01:46.808 1151-1151/? I/TrafficMonitor: count:155,rx_sum:52,tx_sum:26,rxBytes:3590,during_ms:2001,rx_speed:1794.1029485257372,tx_speed:3991.0044977511247,rto:2.0
2021-11-09 16:01:46.809 1151-1151/? D/HuaweiWifiWatchdogStateMachine: speed: rxpkt:52, rxSpd:1794B/s, txpkt:26, txSpd:3991

SourceAdFlt, 2024-01-08:
I got the same issue