libimobiledevice/usbmuxd #176: Can't Connect to any Devices

JBlocklove, 2021-09-07:
I have two iPods (7th Gen Classic and 6th Gen Nano) that I'm trying to use and `usbmuxd` doesn't seem to ever acknowledge either. It doesn't start automatically on plugin like the `udev` rule seems to indicate that it should. Even if I start it myself it doesn't seem to acknowledge any devices.

I'm using Arch, with `usbmuxd` and `libimobiledvice` both installed through `pacman`.

I can't seem to find anywhere where my configuration is going wrong and I'm not really sure where to look to figure it out.

JBlocklove, 2021-09-11:
I just did a fresh install of Arch and this problem still persists.

hockeymikey, 2021-11-20:
Can you tell more about your environment? I am on Arch too and my iPhones connect alright.

JBlocklove, 2021-11-21:
What else would you like to know? I'm happy to share whatever, but there's a lot of environment info I could share for Arch.