libimobiledevice/usbmuxd #184: [Windows] usbmuxd: Could not open device

bialasjaroslaw, 2022-03-07:
I have looked for precompiled binaries for native (MSVC) and MSYS2. Each time I am trying to do anything (run any of tools provided) it ends up with 
`ERROR: No device found!`

I ended up with MSVC 2019 and compilation from source. Luckyly after some time, I was able to run some tools. I am getting the same results as before. 

I have noticed that if I run `idevicesyslog -d` it is showing me that
> 16:04:15 C:\Users\ADMIN\libimobiledevice-vs\libimobiledevice\src\idevice.c:243 idevice_get_device_list_extended(): ERROR: usbmuxd is not running!

So this is a step forward, I thought. And tried to make `usbmuxd` run. However after few tries with or without admin rights, it boils down to 

> [16:06:41.1646665][3] enabling libimobiledevice logging
[16:06:41.1646665][3] usbmuxd v1.1.1 starting up
[16:06:41.1646665][4] Creating socket
[16:06:41.1646665][4] Starting WSA
[16:06:41.1646665][4] Opening socket
[16:06:41.1646665][4] Setting socket to non-blocking
[16:06:41.1646665][4] Binding to socket
[16:06:41.1646665][4] Starting to listen on socket
[16:06:41.1646665][4] Initializing USB
[16:06:41.1646665][3] Using libusb 1.0.24
[16:06:41.1646665][1] libusb does not support hotplug events
[16:06:41.1646665][4] Found new device with v/p 05ac:12a8 at 1-18
[16:06:41.1646665][2] Could not open device 1-18: -799101688
[16:06:41.1646665][4] 0 devices detected
[16:06:41.1646665][3] Initialization complete

After that it is either crash or it repeats over and over one or two lines (depends if I have used `-v` or not)
> [16:14:37.1646666][4] Found new device with v/p 05ac:12a8 at 1-18
[16:14:37.1646666][2] Could not open device 1-18: -834818808

Postmortem debug is presenting so many different places where it crashes (usually it is right after calling some Windows API - and exception is just `The heap has been corrupted`), so it is hard to tell what is going on. I have noticed that it is rarely happening if I run `usbmuxd` without `-v` option, so it is probably something more subtle and not connected directly with my issue.

I can try and build debug version to provide more information, but maybe there is something that can be done or should not be done?
**At the end** what I am trying to do is provide a simple way to view IOS application logs on Windows OS for non technical person. I don't want to install too much of extra stuff, so copy/paste of couple apps with bunch of dll's would be enough, but if there is something that works OOTB I would be happy as well.

frankmarco2000, 2023-04-12:
Hello, Have you had any luck with this?
I am having the same issues. I would prefer not to have to install iTunes - but may have no choice.

ngn999, 2023-05-02:
Meet the same problem.

luc-git, 2023-07-25:
> I found the solution, just install iTunes for Windows.

you are right as you can see here

https://github.com/libimobiledevice/libusbmuxd

> The usbmuxd daemon is running upon installing iTunes on Windows and Mac OS X.