libimobiledevice/usbmuxd #186: ERROR: Could not get ProductName from device

TipzRickyCheung, 2022-04-10:
**Environment**
OS: elementaryOS 6.1 (based on Ubuntu 20.04)
CPU: Intel Xeon E3-1275 v2
Device: iPhone 4S (iPhone4,1)

**Error**
An iPhone 4 on 4.3.3 can be detected, but an iPhone 4S on 6.1.3 can't.

**Logs**
```
Apr 10 15:34:03 TipzLab systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.539][3] usbmuxd v1.1.2 starting up
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.540][3] Successfully dropped privileges to 'usbmux'
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.540][3] Using libusb 1.0.23
Apr 10 15:34:03 TipzLab usbmuxd[609832]: libusb: error [op_get_configuration] device unconfigured
Apr 10 15:34:03 TipzLab usbmuxd[609832]: libusb: error [udev_hotplug_event] ignoring udev action bind
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.864][3] Initialization complete
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.864][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.865][3] Connecting to new device on location 0x10008 as ID 1
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.865][3] Connected to v2.0 device 1 on location 0x10008 with serial number 5d5d321b420bb905c2b6ee9e01ca94a78e4fabd3
Apr 10 15:34:03 TipzLab usbmuxd[609832]: [15:34:03.878][1] preflight_worker_handle_device_add: ERROR: Could not get ProductName from device 5d5d321b420bb905c2b6ee9e01c
```

WilsontheWolf, 2022-07-03:
I'm getting the same error with my iPhone 4 on ios 7

mexmer, 2022-09-21:
this happens to me with iphone 3GS (ios 3.1.2) and iPad (ios 5.1.2)
i posted fix, that is based on DeviceClass instead of ProductName, that seems to be not universaly available.

nikias, 2022-09-21:
This should be fixed with 7f24e9a9c9e862c4d65c3a460fc2315d606d9c7d.

closed 2022-09-21 by nikias