libimobiledevice/usbmuxd #189: [macOS 12.4+] failure to set configuration and claim interface for devices

omerporze, 2022-06-17:
Hi, after debugging and digging through code and seeing that the error stems from libusb I tried contacting them here https://github.com/libusb/libusb/issues/1153 but it seems that they can not help with the issue and suggested trying here. 
Anyways this is the same info as I put there:
darwin_claim_interface seems to fail on macOS 12.4+ on some devices. 
I tried connection iOS devices and I get a `LIBUSB_ERROR_NOT_FOUND` as a result of that function.
On macOS version 12.3.1 the function works without issues.

For more info:
Attached here is output from the command `sudo usbmuxd -v -f` when connecting an iOS device to the computer.
macOS 12.3.1 (working as intended):
```[17:15:23.476][3] usbmuxd v1.1.2 starting up
[17:15:23.477][4] Creating socket
[17:15:23.477][4] Listening on /var/run/usbmuxd
[17:15:23.477][4] Initializing USB
[17:15:23.477][3] Using libusb 1.0.25
[17:15:23.482][4] Registering for libusb hotplug events
[17:15:23.482][4] Found new device with v/p 05ac:12a8 at 3-37
[17:15:23.482][2] Device 3-37: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[17:15:23.482][4] Detaching kernel driver for device 3-37, interface 1
[17:15:23.486][4] Setting configuration for device 3-37, from 5 to 4
[17:15:23.555][4] Found interface 1 with endpoints 04/85 for device 3-37
[17:15:23.562][4] Using wMaxPacketSize=512 for device 3-37
[17:15:23.562][4] USB Speed is 480 MBit/s for device 3-37
[17:15:23.562][4] 1 device detected
[17:15:23.562][3] Initialization complete
[17:15:23.562][4] Got lang ID 1033 for device 3-37
 ```

macOS 12.4 (failing to work):
```
[17:58:39.477][3] usbmuxd v1.1.2 starting up
[17:58:39.477][4] Creating socket
[17:58:39.478][4] Listening on /var/run/usbmuxd
[17:58:39.479][2] chmod(/var/db/lockdown, 02775) failed: Operation not permitted
[17:58:39.479][4] Initializing USB
[17:58:39.479][3] Using libusb 1.0.26
[17:58:39.495][4] Registering for libusb hotplug events
[17:58:39.495][4] Found new device with v/p 05ac:12a8 at 2-52
[17:58:39.495][2] Device 2-52: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[17:58:39.495][4] Detaching kernel driver for device 2-52, interface 1
[17:58:39.496][4] Detaching kernel driver for device 2-52, interface 2
[17:58:39.496][4] Detaching kernel driver for device 2-52, interface 3
[17:58:39.497][4] Detaching kernel driver for device 2-52, interface 4
[17:58:39.497][4] Detaching kernel driver for device 2-52, interface 5
[17:58:39.497][4] Setting configuration for device 2-52, from 5 to 4
[17:58:39.568][4] Found interface 1 with endpoints 04/85 for device 2-52
[17:58:39.570][2] Could not claim interface 1 for device 2-52: error code: -5 LIBUSB_ERROR_NOT_FOUND
[17:58:39.570][4] 0 devices detected
[17:58:39.570][3] Initialization complete
```
It appears that the failure stems from libusb not being able to claim the interface for some reason.
The attempt to claim the interface is done using: `libusb_claim_interface`
which repeatedly fails when calling `darwin_get_interface` or `darwin_set_configuration`.
I would greatly appreciate a fix and will supply more logs/info if needed.
Thanks

mcuee, 2022-06-18:
If Apple changes the kernel side to claim the device exclusively, then libusb and usbmuxd can not be used as of now (just like the situation for UVC camera) as mentioned in libusb #1014. Nothing can be done as of now.
* https://github.com/libusb/libusb/issues/1014.

mcuee, 2022-06-18:
@omerporze 
On the other hand, I just tested our iPhone X and it seems to work under macOS 12.4 (mac Mini M1 -- but I use the usbmuxd x86_64 version as I have some building issues with the M1 version).

Let's see if others have similar issues or not.

```
mcuee@mcuees-Mac-mini usbmuxd % sudo usbmuxd -v -f
[09:56:19.488][3] usbmuxd v1.1.2 starting up
[09:56:19.489][4] Creating socket
[09:56:19.489][4] Listening on /var/run/usbmuxd
[09:56:19.489][4] Initializing USB
[09:56:19.489][3] Using libusb 1.0.26
[09:56:19.494][4] Registering for libusb hotplug events
[09:56:19.494][4] 0 devices detected
[09:56:19.494][3] Initialization complete
[09:56:22.958][4] Found new device with v/p 05ac:12a8 at 2-21
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
[09:56:22.959][4] Setting configuration for device 2-21, from 0 to 4
[09:56:22.959][2] Could not set configuration 4 for device 2-21: LIBUSB_ERROR_NO_DEVICE
[09:56:23.216][4] Found new device with v/p 05ac:12a8 at 2-22
[09:56:23.216][2] Device 2-22: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[09:56:23.216][4] Setting configuration for device 2-22, from 0 to 4
[09:56:23.219][4] Found interface 1 with endpoints 04/85 for device 2-22
[09:56:23.236][4] Using wMaxPacketSize=512 for device 2-22
[09:56:23.236][4] USB Speed is 480 MBit/s for device 2-22
[09:56:23.236][4] Got lang ID 1033 for device 2-22
[09:56:23.236][4] Got serial '222dxxxxxxxxxxx' for device 2-22
[09:56:23.236][3] Connecting to new device on location 0x20016 as ID 1
[09:56:23.236][3] Connected to v2.0 device 1 on location 0x20016 with serial number 222d2xxxxxx
^C
[09:56:26.392][3] Caught signal 2, exiting
[09:56:26.393][4] Event processing interrupted
[09:56:26.393][3] usbmuxd shutting down
[09:56:26.494][3] Removed device 1 on location 0x20016
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
[09:56:26.500][3] Shutdown complete
```

omerporze, 2022-07-25:
Following the discussion here: https://github.com/libusb/libusb/issues/1153
It looks like updating to macOS 12.5 fixes the issue.

closed 2022-07-25 by omerporze