libimobiledevice/usbmuxd #197: Accidentally created wiki on repo

ArrowAced, 2022-09-29:
I accidentally created a wiki for this repo and can't delete it (which shouldn't be possible in the first place.) Can it be deleted from settings?

nikias, 2022-10-05:
I *would* do it, but unfortunately @FunkyM didn't give me ownership permissions for the project yet. So he has to take care of that himself, or even better, @FunkyM, could you just give me the necessary owner/maintainer permissions? Thanks.

rickmark, 2022-10-05:
I think it's critical for this project to stay relevant (there are alternates cropping up like python based versions) that the org have sufficient collaborators, and @nikias is one of the most prolific contributors here...

rickmark, 2022-10-05:
Erm I think gsub collaborators with administrators

sbingner, 2022-10-05:
This seems like a problem, not having more than one person with this access is asking for a disaster or major headache. @nikias has been doing enough work on this for long enough that he should be trusted with full permissions.

iH8sn0w, 2022-10-05:
It makes sense for the active maintainer to have the necessary project permissions to rectify issues like this. Given @nikias has been leading and maintaining this project alongside the countless others, it would help improve resolution time on such issues and grant a well-deserved status too.

jeppojeps, 2022-10-06:
I mean, this is open source software, @nikias has been the heaviest committer on this project since a lot of time. Instead of fostering yet another fork, it'd be nice to see some fairness.

pekn, 2022-10-11:
I think @nikias should have maintainer rights - he is the author of most commits after all.

sywesk, 2022-10-11:
Wait, isn't @nikias the owner of this organization? It's the only person I talked with and the only one that merges PRs so I'm a bit surprised actually 😅

But he should have admin rights on the org, that would enable so many things like releasing a new version of libmobiledevice 👌

P5-2005, 2022-10-11:
> I *would* do it, but unfortunately @FunkyM didn't give me ownership permissions for the project yet. So he has to take care of that himself, or even better, @FunkyM, could you just give me the necessary owner/maintainer permissions? Thanks.

lmao, thought it was a joke 😆, but seriously you earn this ownership/maintainer role 100%, @nikias  your contribution on all of limbd projects is an asset for the community :-)!, hope we see you with this role soon.

Cryptiiiic, 2022-10-11:
From all the good nikias has done for our community, not having maintainer would be a step back in our journey. He has contributed far to much to be left on the side of the road.

TheRealClarity, 2022-10-12:
I really thought and I am very surprised that @nikias isn’t the owner of the organization, since he’s been pretty much maintaining this whole project for the last few years, and he’s always the first person we think about when we find ourselves with some issue related to libimobiledevice.

nikias, 2022-11-06:
@FunkyM it's literally just changing my role from `Member` to `Owner`...