libimobiledevice/usbmuxd #198: Preflight check fails with `lockdown error -8`

aluedeke, 2022-10-05:
I am trying to build usbmuxd myself in docker:

```
FROM docker.io/ubuntu:22.04 AS BUILD

RUN apt-get update && apt-get install -y git build-essential pkg-config checkinstall libtool-bin autoconf automake libssl-dev libusb-1.0-0-dev

RUN git clone https://github.com/libimobiledevice/libplist.git
RUN cd libplist && \
    ./autogen.sh --prefix=/usr/ --without-cython && \
    make && \
    make install

RUN git clone https://github.com/libimobiledevice/libimobiledevice-glue.git
RUN cd libimobiledevice-glue && \
    ./autogen.sh --prefix=/usr/ && \
    make && \
    make install

RUN git clone https://github.com/libimobiledevice/libusbmuxd.git
RUN cd libusbmuxd && \
    ./autogen.sh --prefix=/usr/ && \
    make && \
    make install

RUN git clone https://github.com/libimobiledevice/libimobiledevice.git
RUN cd libimobiledevice && \
    ./autogen.sh --prefix=/usr --without-cython && \
    make && \
    make install

RUN git clone https://github.com/libimobiledevice/usbmuxd.git
RUN cd usbmuxd && \
    ./autogen.sh --prefix=/usr/ --without-systemd && \
    make  && \
    make install

RUN apt-get update && apt-get install -y libssl3 libusb-1.0-0

EXPOSE 5037

CMD ["usbmuxd", "-v", "-f", "--socket", "0.0.0.0:5037", "--disable-hotplug", "-U", "root"]
```

when I run the container the preflight check always fails:
```
[09:08:20.783][3] usbmuxd v1.1.2 starting up
[09:08:20.784][4] Creating socket
[09:08:20.784][4] Listening on 0.0.0.0:5037
[09:08:20.785][4] Not dropping privileges to root
[09:08:20.785][4] Initializing USB
[09:08:20.785][3] Using libusb 1.0.25
[09:08:21.084][4] Registering for libusb hotplug events
[09:08:21.084][4] Found new device with v/p 05ac:12a8 at 1-17
[09:08:21.085][4] Setting configuration for device 1-17, from 1 to 4
[09:08:21.382][4] Found interface 1 with endpoints 04/85 for device 1-17
[09:08:21.382][4] Using wMaxPacketSize=512 for device 1-17
[09:08:21.382][4] USB Speed is 480 MBit/s for device 1-17
[09:08:21.383][4] 1 device detected
[09:08:21.383][3] Initialization complete
[09:08:21.383][3] Automatic device discovery on hotplug disabled.
[09:08:21.383][4] Got lang ID 1033 for device 1-17
[09:08:21.383][4] Got serial '00008101000E20623638001E' for device 1-17
[09:08:21.384][3] Connecting to new device on location 0x10011 as ID 1
[09:08:21.384][3] Connected to v2.0 device 1 on location 0x10011 with serial number 00008101-000E20623638001E
[09:08:21.384][4] preflight_worker_handle_device_add: Starting preflight on device 00008101-000E20623638001E...
[09:08:21.385][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device 00008101-000E20623638001E, lockdown error -8
```

Does anyone has a hint what i am doing wrong?

mexmer, 2022-10-05:
what device with what version of ios you are trying to connect?

aluedeke, 2022-10-05:
iphone 12 pro max with ios 16.0. Btw Ubuntu 22.14 as docker base image and `apt-get install usbmuxd` works perfectly. Unfortunately this version doesn't support the socket parameter yet.

mexmer, 2022-10-05:
since you have compiled libimobiledevice yourself, make sure, there is no preinstalled libimobiledevice from ubuntu. otherwise they will be in conflict.

aluedeke, 2022-10-05:
I am compiling inside a Docker container, so I am 100% sure there is no preinstalled usbmuxd version

mexmer, 2022-10-05:
lockdown error -8 is usbmux error, probly it can't connect to lockdown service for some reason.

if you add 
```
idevice_set_debug_level(1);
```
to https://github.com/libimobiledevice/usbmuxd/blob/12e24e4bd161c76631c77244f63ff02ef18f251d/src/preflight.c#L154
and recompile usbmuxd, we might get some output from libimobiledevice, that could help

aluedeke, 2022-10-05:
I figured out what's going wrong. As i am setting `--socket 0.0.0.0:5037` libimobiledevice needs to know where to find the usbmuxd socket for the preflight check. Thus i have to set `USBMUXD_SOCKET_ADDRESS=127.0.0.1:5037` as well. Feels a bit inconvenient tbh.

closed 2022-10-07 by aluedeke

mexmer, 2022-10-10:
hmm, didn't thought about that. maybe we can ask @nikias if there is something, that can be done, i thought preflight always connects to localhost.

nikias, 2022-10-10:
I am looking into it. It should indeed just connect internally.