libimobiledevice/usbmuxd #200: Fresh build fails with "cannot open shared object"

bricechart, 2022-10-08:
I'm trying to build the latest 1.1.2 of usbmuxd but I'm running into an error with libimobiledevice-glue. I'm getting the following error.

`error while loading shared libraries: libimobiledevice-glue-1.0.so.0: cannot open share object`

I built libimobiledevice-glue the from latest git repo before building usbmuxd since it has a dependency on usbmuxd. The build of usbmuxd appears to build without errors. Any thoughts on what I might be doing wrong or is this a bug? I'm running this all on a Raspberry Pi model B.

nikias, 2022-10-08:
Have you tried to run
```
sudo ldconfig
```
?

bricechart, 2022-10-08:
> Have you tried to run
> 
> ```
> sudo ldconfig
> ```
> 
> ?

That worked! Thanks! Now the next step is get my iPhone to appear as a network adapter. I think this may be a bug with iOS 16, because I'm having the same issue in Windows 11.

vincejv, 2024-06-16:
An explanation why `ldconfig` needs to be ran

>  ldconfig creates the necessary links and cache to the most recent
       shared libraries found in the directories specified on the
       command line, in the file /etc/ld.so.conf, and in the trusted
       directories, /lib and /usr/lib.  On some 64-bit architectures
       such as x86-64, /lib and /usr/lib are the trusted directories for
       32-bit libraries, while /lib64 and /usr/lib64 are used for 64-bit
       libraries.

since you are compiling and installing all libraries from source, the shared libraries cache must be refreshed

more details: https://man7.org/linux/man-pages/man8/ldconfig.8.html