libimobiledevice/usbmuxd #207: lockdown error -5 when attempting to connect iPod Touch

BezPowell, 2023-01-06:
I'm trying to backup all the data from my Partner's old iPod Touch. All of the libimobiledevice commands return a message saying `No Device Found`.

The device shows up on `lsusb`
```
Bus 001 Device 006: ID 05ac:129e Apple, Inc. iPod Touch 4.Gen
```

Running `usbmuxd -f` gives the following output.
```
[15:49:50.234][3] usbmuxd v1.1.1 starting up
[15:49:50.235][3] Using libusb 1.0.25
[15:49:50.241][3] Initialization complete
[15:49:50.249][3] Connecting to new device on location 0x10006 as ID 1
[15:49:50.250][3] Connected to v2.0 device 1 on location 0x10006 with serial number 608da3e8a75e39c5cd99953626f8c6bf2e36eca6
[15:49:50.367][1] preflight_worker_handle_device_add: The stored pair record for device 608da3e8a75e39c5cd99953626f8c6bf2e36eca6 is invalid. Removing.
[15:49:51.061][1] preflight_worker_handle_device_add: ERROR StartSession failed on device 608da3e8a75e39c5cd99953626f8c6bf2e36eca6, lockdown error -5
```

I'm running Fedora 36 and using the latest builds of everything from the repositories. I seem to remember seeing another bug report that suggested it could be an issue with openssl, but was wondering whether there war any other steps I could try and take?

Running `openssl ciphers -s SSLv3` returns the following output:
```
TLS_AES_256_GCM_SHA384:TLS_CHACHA20_POLY1305_SHA256:TLS_AES_128_GCM_SHA256:TLS_AES_128_CCM_SHA256:DHE-RSA-AES256-SHA:DHE-RSA-CAMELLIA256-SHA:DHE-RSA-AES128-SHA:DHE-RSA-CAMELLIA128-SHA:AES256-SHA:CAMELLIA256-SHA:AES128-SHA:CAMELLIA128-SHA
```

I don't need to regularly mount this to interact with it, just getting it to work once and take a complete backup of all the data is more than sufficient.

bigkahuna1986, 2023-08-24:
In the same boat as you (3rd gen iPod touch). I tried the steps from [this article](https://lkubuntu.wordpress.com/2016/04/19/how-to-fix-lockdown-error-5-on-usbmuxd/) to no avail. Please update if you have any more information!

mexmer, 2023-08-24:
remove libimobiledevice and usbmuxd and all related packages distributed by your system, compile all from sources and install, it should work.

if you leave any distribution provided packages installed, it might cause issues.

jackrr, 2025-09-01:
To fix on Fedora 41 I found it sufficient to run `update-crypto-policies --set LEGACY` as root to fix