libimobiledevice/usbmuxd #220: Can not Bind local TCP port 8527 and forward to port 6661 of the first device connected via USB:

nyz110, 2023-07-03:
`lsof -n -i4TCP:9527`
print nothing
but try exec command
`iproxy 9527:6661  `
print
```
Creating listening port 9527 for device port 6661
waiting for connection
```
just wait....
help

mexmer, 2023-07-03:
not sure what exactly you asking, iproxy forwards port from device to your computer, then you can connect to that port.

let's say, you have some custom http server running on your phone on port 8080
so you run
iproxy 12345:8080
then use browser to open http://localhost:12345 and it will connect to your http server inside device

nyz110, 2023-07-10:
sorry，iproxy has no problem。
  My script run iproxy 9527:6661  in terminal successfully。I think the iproxy service has while logic to observer connection and print logs。so，other commands after iproxy command in my script file will not run。 I solve it with  `iproxy 9527:6661  &`.  this iusse can be closed。

closed 2023-07-10 by nyz110

mexmer, 2023-07-10:
correct, iproxy runs in foreground so if you want to run it inside script, you should use background operator, also i would recommned redirecting stdout/stderr to /dev/null or log file.